OpenCPN/OpenCPN #1584: Sometimes waypoints become invisible when O is restarted

transmitterdan, 2019-12-25:
1) Draw a route using existing way points or draw a route by adding way points automatically.
2) Use route manager to make route invisible.
3) When asked if shared way points should also be made invisible click NO. Then close route manager. Route points are visible as they should be.
4) Close O and restart.
5)  Route points are now invisible even though that is not what the user wanted. Way points not in a route respect the visibility flag of the way point but way points used in a route do not.

We fixed the bigger bug of the shared way points disappearing right away.  Now they disappear on next invocation.

Hakansv, 2019-12-25:
@transmitterdan 's note confirmed on Win10

transmitterdan, 2021-04-14:
This appears to be fixed at least in Win10

closed 2021-04-14 by transmitterdan