OpenCPN/OpenCPN #1585: PI Installer:  Weather_routing, Flatpak failure due to very old commit and bad commit date

rgleason, 2019-12-25:
Please see  https://github.com/seandepagnier/weather_routing_pi/issues/258

I am hoping someone will have some experience with this so that I can get FlatPak building.
I am very nervous about trying to change the commit date using git rebase, as this error is about 3 years ago and many many commits ago. It is too many commits away to be a sure thing and too much rides on top of it.

Your thoughts would be appreciated.
1. Is there a way to automatically create a completely current "flat" version with no history, that I can use?   even if I have to make a completely new repository? Just so I can get it to compile?
2. Is there some other way that will preserve history, and not screw up all the dependent forks and branches?

rgleason, 2019-12-25:
Jon, just got the MacOS building.  One more to go... how to fix the bad date to get flatpak to build?

leamas, 2019-12-26:
Well... it could be tricky. But if you just tag your HEAD it's easy to revert a failed rebase. It's also possible to retrieve the old head after a failed rebase even without a tag pointing to it.

So, the way is to try.... after making a tag as backup

jongough, 2019-12-26:
https://github.com/OpenCPN/OpenCPN/issues/1585#issuecomment-568937787
This is now fixed. Needed to provide the non-master branch id so that flatpak would download and build the correct version. This is controlled by the `set(GIT_REPOSITORY_BRANCH "api-1.16")` in CMakeLists.txt, part of the general 'configuration' part of the new cmake process.

rgleason, 2019-12-26:
Yes, Jon does have branch "api-1.1.16" working now, with the correct non-master branch!
I need to remember that setting is essential!

The branch "master-fix-time" is all ready to try fixing the time problem:
1. Tagging the commit with the bad time with "bad-year" tag.
2. Using the Tag to change the commit time (just the year) using git rebase -i  HEAD~<tag>  ?
3. Then see how it works.
4. Then learn how to add (is that "cherry pick", just Jon's two additional commits from the  "api-1.16" branch.
5. Then see how it looks.

leamas, 2019-12-26:
Time to close?

rgleason, 2019-12-26:
Good Morning. Almost. I want to make the change to the bad time entry, in branch master-fix-time. and see what happens. Then decide what to do.

rgleason, 2019-12-26:
I tried this:

git tag -a year 23c07cc5d2be7ce68349f4b3719b6fa6fe90e0bf

git show year

    Tagger: Rick Gleason <fcgleason@tgp-architects.com>
    Date:   Thu Dec 26 08:40:52 2019 -0600
    Fix year to June 19, 2017

    commit 23c07cc5d2be7ce68349f4b3719b6fa6fe90e0bf (tag: year, tag: show, stelian42/crossover, crossover)
    Author: Sean D'Epagnier <seandepagnier@gmail.com>
    Date:   Sun Feb 7 06:28:56 2106 -40643156

        ui updates and other changes

Previous commit is "improve UI and add polar editing"  c3d90e861de93721ba1e2a1a42f72e118deb3db1

so use

git rebase -i  c3d90e861de93721ba1e2a1a42f72e118deb3db1

in text editor put  "edit" by  23c07cc5d2be7ce68349f4b3719b6fa6fe90e0bf   ui updates and other changes

"edit 23c07cc ui updates and other changes"

git commit --amend --date="Sat, Jun 19 2017 06:28:56 +0000"

I then ran into all kinds of conflicts that had to be resolved and just decided to bail out with

git rebase --abort

My take is even if the commit date is totally wrong, don't mess with it.

closed 2019-12-26 by rgleason