OpenCPN/OpenCPN #1586: Plugin Manager - xml control file generation/maintenance

jongough, 2019-12-25:
The xml file used for the presentation of plugins needs to be genrated/maintained on a very regular basis, i.e. each time a plugin is 'generated' in the cloudsmith repository. Requiring this to be done by hand could be an onerous task as, currently, only one person can make and publish the changes. If this becomes multiple people there are issues with collisions in changes, etc.

How is the xml file or files (if there is a file per catalog) going to be maintained in a meaningful way?

leamas, 2019-12-26:
In two ways:

  - The generation of XML metadata should be made part of the regular build process. See
     https://github.com/leamas/OpenCPN/wiki/Plugin-adaptation.

  - The coordination of different plugins (that is, maintenance of the common plugin catalog)
     is the purpose of the [plugins project](https://github.com/opencpn/plugins)

In particular, a plugin developer have two ways:

  - Create a private catalog for testing  using tools in the plugin project.
  - Submit a PR to the plugins project, adding or updating XML metadata files(s), which when 
     merged incorporates the changes to the "official" catalog.

rgleason, 2019-12-26:
Note that I and others are not going to be able to test all OS versions for a given plugin. We will be depending on others for that. 

Therefore we need a "beta_plugin.xml for untested plugins and a means to confirm that the plugin's OS version works properly, so that it can be moved to the regular catalog.

leamas, 2019-12-26:
This is basically something for a PR against the plugins project. A beta test is a good example how to use the channel mechanism. The workflow:

  - Dave (maintainer) creates a new branch call beta-1 or whatever.
  - Interested parties submits PRs with XML metadata for plugins to test.
  - Dave applies these PRs to the beta-1 branch.
  - As soon as the branch beta-1 exists it's visible as a channel in the plugin catalog interface.
  - Testing takes place using  catalogs from the beta-1 channel, new PR:s with bugfixes for Dave
     to merge.
  - Eventually, Dave merges beta-1 onto master, making the plugins visble in the master channel

jongough, 2019-12-26:
This is not really a workable solution. I have tried the python script and it is harder than copy/paste of the required xml into the ocpn-plugins.xml file. The process is only useful when doing one or two tests on a couple of instances of a plugin, but it is not useful when doing proper testing of multiple versions of a single plugin let alone when doing this for multiple plugins.

The process needs to be automated, needs to not rely on one or two people (that may be OK for final/release versions, but....) and needs to be simple. The current process is anything but. The plugins project currently does not address the issues, it is purely a hack to get something working.

Remember there are 40 plugins currently. Some of these plugins have periods of intense work where new versions/build occur very regularly. The testing/user base have vastly different skill levels and knowledge and should be able to work the way they have in the past. The whole process needs to be distributed and possibly under user control from a simple gui extension to Plugin Manager (the current 'advanced' download is not suitable).

This issue needs to be taken seriously and given serious thought. It appears to be random thought bubbles at the moment.

leamas, 2019-12-26:
Well, this is what we have for now. 

Patches welcome!

jongough, 2019-12-26:
You are quite welcome to make your own patches and hopefully the developer/designer of the Plugin Manager/Installer changes will accept them.

jongough, 2019-12-26:
If I were to attempt to make changes I would, initially, move away from the tarball process and back to the platform installation process. I would then focus on the UI and the user experience, then on the ability to handle versioning of plugins as well as dynamic update of the xml (may not even need this, just the location of the download store and take it from there). When all of that is working well and bedded in I would look at possibly using tarballs if the platform installation process will not work. And I would then focus on ensuring all non-user generated data was cleanly removed when uninstalling OCPN or the plugins (there should be no difference). I could go on, but you have a fundamental issue with my approach and will not accept it. So there is little point in me trying to do this work.

leamas, 2019-12-27:
> You are quite welcome to make your own patches and hopefully the developer/designer of the Plugin Manager/Installer changes will accept them.

This is OpenCPN code. Dave and Pavel are the only maintainers of it, it's their decision to accept code or not.

jongough, 2019-12-28:
So, are you going to fix this up or not?

leamas, 2019-12-28:
Personally, I have no such plans.

jongough, 2019-12-28:
So the developer is leaving a half done job and does not intend to fix the issue? You may need to contact the developer, as I don't know who that is, and suggest strongly that they should consider this urgently as this will become a much bigger issue when more plugins move to this process. If they are not going to sort this out who should? It is rather fundamental to the whole process.

leamas, 2019-12-28:
I still think the basics described in https://github.com/OpenCPN/OpenCPN/issues/1586#issuecomment-569028801 are a working, reasonable well-defined process. 

In particular I see no basic problems automating this flow:
  
  - Clone plugins project
  - Let the build(s) drop XML files into the plugins project metadata folder
  - Run the python script which creates ocpn_plugins.xml
  - Commit ocpn-plugins.xml into the cloned project

This flow creates an URL which can be used as Custom URL by testers which then can update their catalog with a single button click.

If you have other ideas you need to be more specific.

rgleason, 2019-12-28:
This process appears to be surprizingly simple, as presented. 

We need to actually be able to achieve these steps first, with 3 or 4 examples, that all work, install, operate properly, and uninstall reliably for all OS.

Jon has given us his thoughts after much programming and testing. Thank you. I need to work more on this to understand what the issues are, and will do so. I will say that I find the new location for plugins for windows, more difficult to get too and it is a third location for opencpn files and after defining the two ways to get to the standard plugin location for windows users, at the beginning, I have wondered why we end up with this third location, which has potential to confuse the confusible windows users, but I think the technical reason given was "permission" problems. However the current pi executibles seem to work ok, but that is a kind of manual installation, whereas the PI Installer is executing from Opencpn.  Other programs do this without issues, so they must provide credentials for themselves somehow. Notepad++ is an example.

rgleason, 2019-12-28:
Perhaps the problem stems from the way credentials are handled in different OS. Linux is different than Windows, and we need a talented windows programmer to take a look at this?

leamas, 2019-12-28:
There is no basic difference between windows and Linux in this respect. Programs are installed in areas where regular users only have read access like \\Program Files\\ on windows and /usr/{bin,lib,share} on linux. Since the design criteria is to run as a regular user these areas cannot be used by the installer.

When it comes to user-writable areas they reside under $HOME/%HOME% in both cases.  The user-writable configuration directory on windows  is C:\\ProgramData\\opencpn, this is nothing new. It corresponds to ~/.opencpn on linux. 

However, the windows user-writable data directory is %HOME%\\Appdata\\Local. This nothing invented here, it's an established and documented location. If you look into it you will find multiple programs storing their data there.

So, what we are doing on windows is the same as for example many windows programs and the browsers. It is certainly not strange.

jongough, 2019-12-28:
The 'new' installation process is reinventing the wheel. There are already robust, tested and supported installers/uninstallers for all the platforms that are used by OCPN. Creating an OCPN installer/uninstaller is asking for issues. I am sure the platform installers did not work first time out of the box. Why does OCPN need to get into 'dynamic locations', as it is supposed to be navigation software not package management? Is it really that much of a problem to click OK on a dialog box, or provide a password when installing software? That is far less work that 'finding' bits of tarballs around the system when the controlling software crashes. Most users know nothing of 'hidden' directories and 'AppData' is a hidden directory in Windows,  'ProgramData' is not. Thought needs to be given to general users who know nothing about how systems work and really don't want to, they want to point and shoot, not stuff around.

The idea of using a control file to help users find, install and uninstall plugins is good. This process could have used the platform install processes to do the install/uninstall work. This would ensure that the original install process and this one would work hand in hand. It would be up to the user which one to choose, but the new process would not mask the old but would work with it.

The idea of having 'repositories' for each plugin is not new, OCPN and the plugins have been using Launchpad and git for these functions. The use of Cloudsmith is just a different take on it. The issues that need to be sorted out are:
* Standardised naming and use of repositories
* Standardised repository functions:
  * Release - Released versions of packages
  * Beta - Next version of package which is open for testing by all users
  * Alpha - Next version of package which is intended for experienced users/developers
* Standardised search/find for packages
* Automatic update of the control file - don't require the OCPN maintainers to maintain the control file for all plugins as doing this will make delivery of non-Release versions difficult (think patches for user found issues).

So one possible solution is to have distributed control files.
* A master control file pointing to the different repositories but not the files within them. This is maintained by OCPN and would only need changing if repositories are added/changed/moved/deleted. This 'may' need to be three, Release, Beta, Alpha to keep things clean, but naming standards could help here.
* Each repository will have their own control files (standardised). Use could be made of the current ones as they all have 'xml' as a suffix and 'should' have a standardised initial part of the name.

When a user wants to install a plugin they go to the Plugin Manager and use the 'current' catalog for selection, or they 'update' the catalog (similar to an `apt update`) to get the current lists. This is dynamically built and then cached for use later.The user could also, if required, select which repository type to look for (Release, Beta, Alpha).

This next bit will need some trial and error to get the right feel. The user should be able to select the version of the plugin they wish to install, i.e. Latest, Previous or 'select from list'. This would enable the user to go back to a version of the plugin that works for them if the 'Latest' one doesn't (this does happen!).

All of the above would be accomplished with control files (a modified version of the current ocpn-plugins.xml file perhaps), with no need, or very little, for centralised manual control, construction and maintenance of the actual files.

leamas, 2019-12-29:
The current design is based on a few criterias, one of which that the project should have  control over the set of approved plugins -- this is the basic reason why the catalog is a single, centrally maintained file. 

So the lack of distributed repo support is not a problem, it's a feature.

jongough, 2019-12-29:
Interesting that you call a failure a feature. This is the worst of political speak.

With this feature how are patches/updates to be made to programs for users? If there is a requirement that someone update the catalog for each change to make the change available, do they know the work they are letting themselves in for and the time frames that users will be expecting?

To try out your new version of squiddio required me to go to git hub, download the catalog you had built, then copy/paste the contents into the catalog I had on the machine I was working with before I could try out the change. This is not something a user would do.

If you want the production/release version of plugins to be controlled by the project (which project is this?), then it will slow down changes. But if you want all changes to be controlled by the project it will bring all work on plugins to a halt. Or users will have revert to using the current process and ignore the Plugin Manager. Either the Plugin Manager is supposed to be used and be usable, or it needs to be dropped and the current method maintained. 

I have given you a possible solution, but for some reason you don't want to consider it.

leamas, 2019-12-29:
correctamunto. I will not start investigating this path (early this spring I actually started there, but reconsidered after discussing with Dave).

When it comes to me, it  also  too late  for these kind of drastic design changes. I have made mockups, discussed and refined. I will not start from the beginning again.

But that is only me. There are other devs out there, and perhaps someone picks your ideas and actually builds something. Or perhaps you do.

Bottom line:  I'm not interested to discuss the fundamental design at this point. Bugfixes, enhancements etc are of course another thing and always welcome.

jongough, 2019-12-30:
Basically you have tossed a partial thought bubble on to OCPN and are now not interested in supporting it. I was trying to get you to look at the bigger picture before this got accepted into the mainline, but was unsuccessful. Now it is in the mainline it needs proper support and fixes (these seem rather slow in coming). You need to accept all issues and address them, not by providing political double speak.

The changes I am proposing are all small, and would enhance the capability. This change would reduce the workload on those that currently have to manually manage the repositories. Again, this is not a big bit of work.

Accept that this implementation is not perfect and try and improve it. Don't rely on others to clean up after you.

bdbcat, 2019-12-31:
I can see some value in @jongough suggestions.
Not as direction, but for thought:
1.  Having three named repos or catalogs will definitely assist us in Alpha and Beta testing.
2.  Having a "meta-meta" structure in the catalog could be a nice way to implement this.  By this I mean a tag which indirectly refers to another XML file maintained by the plugin author, and used only for Alpha and Beta testing.
3.  With this scheme, the user selects the "Beta" track if desired.  Plugins that are being dynamically maintained by the author during Beta testing are then available for installation.

I'll need to read some code, and think about this some more.
And yes, this is indeed another "thought bubble".  And that is how we will make progress.

rgleason, 2019-12-31:
I like these ideas, with the beta.xml's being created by the pi authors and used by testers, without to much involvement by Dave. Then working up to a release version which is a more curated and  controlled procedure.
I like the idea of user selection of a beta channel that somehow shows the beta.xml track for various PI using the appropriate cloudsmith beta repository.
This allows flexibility initially and ease of testing for beta tester, then more control for the release version.

leamas, 2020-01-01:
H
@  bdbcat wrote:
>
> I can see some value in @jongough suggestions.
> Not as direction, but for thought:
>
>     Having three named repos or catalogs will definitely assist us in Alpha and Beta testing.
>     Having a "meta-meta" structure in the catalog could be a nice way to implement this. By this I mean a tag which indirectly refers to another XML file maintained by the plugin author, and used only for Alpha and Beta testing.
>     With this scheme, the user selects the "Beta" track if desired. Plugins that are being dynamically maintained by the author during Beta testing are then available for installation.

There are some support for this in current code. Expanding on above, let's assume Alice controls the plugins project, Bob want's to organize some sort of beta testing and Eve actually run tests.

  - From the beginning there is a master branch where Alice publishes the plugin catalog.
  - When Eve invokes *Update Catalog* she will download the latest catalog from the master branch.
  - If Alice wants to create a beta test channel she just creates a git branch called beta. This becomes visible in the UI as a "channel" Eve can select. If she selects this  her catalog updates comes from the 'beta' rather than the master branch
 -  If Alice trusts Bob she can give him administrator privileges in the plugins project after making a deal  that Bob only commits to a new branch called experimental.  For Eve, this works as above.
  - If Alice don't trust Bob he can still publish his ocpn-plugins.xml on an alternative url, and instruct  Eve to set this url as Custom URL in the UI. Eve will then get her catalog updates from Bob.

Note that the available channels in the UI reflects available branches in the git repo, it's not hardcoded.

leamas, 2020-01-01:
@rgleason wrote:
> Is there a way to show a "beta" version somehow or would that be. Indicated in the version number somehow?

The new plugin API (117)  defines a  *getPlugInVersionPre()* function intended for beta tags etc.; that is, pre-release info. This is part of the semantic version spec.

leamas, 2020-01-01:
> Having a "meta-meta" structure in the catalog could be a nice way to implement this

Perhaps. My idea has been to use git branches instead.  This basically means that we  treat  a branch called beta like a PR which eventually is merged into the master branch using regular git tools. Less code (no meta-catalog stuff) and probably more flexible.

rgleason, 2020-01-01:
Useful, but in this schema, the beta branch requires action by Alice, rather than PI authors just publishing to a common place for beta testing.

leamas, 2020-01-01:
Not necessarily. Assuming Bob is admiinstrating the common place for beta testting, Alice (that is, Dave) could give Bob adminstrative privileges in the plugins project (fourth bullet).

And if not, Bob can still publish the catalog on an url (last bullet). This means that he clones the plugins project and commits updates to the catalog, typically on the clone's master branch. Bob  informs Eve about the URL which she uses as Custom URL.

In both these scenarios Bob can update the catalog without intervention from Alice. The difference is that in first case there is some delegated trust given by Alice to Bob which makes Bob's changes available as a channel. 

For Eve selecting a channel is of course easier than updating the Custom URL.

rgleason, 2020-01-01:
OK here is a beta branch
https://github.com/rgleason/plugins/tree/beta

Here is the link that works using "RAW" view in the upper right.
https://raw.githubusercontent.com/rgleason/plugins/beta/ocpn-plugins-beta.xml

It is not possible to make a PR to 
https://github.com/OpenCPN/plugins/beta
because it does not exist.

leamas, 2020-01-01:
No,  and that is not the proposed workflow.  Assuming you are the one administrating the plugin testing (which, BTW, I think is a great idea):

  - Either Dave give you commit rights in the plugins project so you can make commits. These
    powers should then only be used on for example the beta branch, gentlemens agreement.

  - Or you inform your users about the RAW url (so to speak)  which they uses as Custom URL. 

In both cases you can update the catalog as you wish. The first case is somewhat simpler for your users, though.

leamas, 2020-01-01:
> It is not possible to make a PR to
> https://github.com/OpenCPN/plugins/beta
> because it does not exist.

Indeed. But creating a git branch is really simple, Dave could do that as part  of giving you or someone else commit rights.  

That said, to make a PR against a non-master branch is non-trivial and should be avoided.

bdbcat, 2020-01-01:
On reflection, I like the meta-meta idea more. Consider this design/workflow:
1.  The are four channels (Release/Beta/Alpha/Experimental), implemented as four (and only 4?) branches.
2.  I'll use specific names to describe persona.
3.  Jon wants to manage some Beta testing for his plugin.  He submits a PR to Beta repo to replace his plugin content in the XML entirely with a meta tag referencing a separate XML file that he maintains, somewhere.
4.  I accept the PR, and will always do so for Beta/Alpha/Experimental branches, no questions asked.
5.  Hakan selects the Beta channel for testing, and so gets Jon's new code.
6.  Jon can work with his XML as necessary, without my involvement at all, as the testing proceeds.
7.  When everybody is happy, Jon submits PR against master channel. The final result does not contain a meta-XML callout, but contains the full plugin specification and becomes a standard controlled Release.

This is a bit more complicated than Alex's idea, but does not require granting commit privileges access to the main repo.  I think the code changes would be manageable.  I'll take a look today and propose something.
Comments?

bdbcat, 2020-01-01:
You will ask:  Why not just use the custom URL capability?
Answer:  Not "convenient" for our general tester base, who are mostly more comfortable (and reliable) with "point-and-click".

leamas, 2020-01-01:
>  I think the code changes would be manageable

I probably miss something here -- I don't see  that the code needs to be changed in any way for this usecase. It should work out of the box. It's just about merging the PRs to proper branches.

OTOH, why not give Jon commit rights? I'm sure can respect a simple rule to just commit to a specific branch. And it would make life much simpler for him.

Note that this is *not* about giving anyone any rights whatsoever in the main opencpn project. The plugins project is a separate one.

leamas, 2020-01-01:
> Q: Why not just use the custom URL capability?
> A: Not "convenient" for our general tester base,

Agreed. After all, I did create the channels mechanism :)

rgleason, 2020-01-01:
> That said, to make a PR against a non-master branch is non-trivial and should be avoided.

It's actually quite easy, and I've done it a number of times between branches in my repository and between branches in different repositories.  Why should it be avoided though?

I like Dave's description here. I think it has good workflow characteristics.
https://github.com/OpenCPN/OpenCPN/issues/1586#issuecomment-570063590

leamas, 2020-01-01:
This basically boils down two things:
  - User should be able to select channel using point-and-click
  - Someone (Jon, you, ? i. e, Bob) should be able to work independently with the 
    catalog corresponding to a specific branch.

Dave's proposal should work (it could be implemented in different ways). I still consider giving Bob commit rights in the plugins project is a simpler and cleaner solution. After all, the idea means that the project trusts Bob to maintain the catalog for a given branch anyway. Why not just enshrine this in giving him commit rights?

rgleason, 2020-01-01:
Clarification re content of the Plugin metadata.xml  (Experimental/Alpha/Beta) to be tested.
Initially say at "Beta" stage the file would include all OS versions uploaded to cloudsmith (or wherever).
As each Plugin OS flavor gets the green flag, the PI maintainer could then move into ocpn_plugins.xml and make PR to the Release Branch?

jongough, 2020-01-01:
I must be missing something here. The discussion seems overly complex and requires premissions to be given out and managed. Here is another suggestion:

We already have 3 cloudsmith repositories for built code (the naming is still not standardised). But if we call them 'Prod', 'Beta', 'Dev'/'Alpha', there is no need to go near git. These repositories are selected based on how the plugin owner (git owner) builds using circleci.
* 'Prod' is built when a new 'Release' version tag is created
* 'Beta' is built when new 'Draft' version tag is created
* 'Dev'/'Alpha' is built when a pull is accepted

Now all that is needed is in place from a plugins perspective.

To use this we would then need Plugin Manger allow the user to select 'Prod', 'Beta', 'Dev'/'Alpha' as the catalog type. This would then result in:
* 'Prod' - use ocpn-plugins.xml as provided by OCPN and managed by the OCPN team.
* 'Beta' - use ocpn-plugins.xml with meta data pointing to the 'Beta' cloudsmith repository. 
  * The Plugin Manger would download all the xml files found in the repository and build the full ocpn-plugins.xml file containing all the beta plugins found. 
  * When a plugin is selected to be installed the user will be given the opportunity to select which version, i.e. 'Latest', 'Previous', or one from the version list. 
  * The install would then continue as today.
* 'Dev'/'Alpha' - same as 'Beta' but using the 'Dev'/'Alpha' cloudsmith repository

This has the following 'advantages':
1. There is no need for multiple git branches (there are already enough)
1. The deployment is contained in the cloudsmith repositories which is what they are being used for anyway 
1. The user will have a GUI that will allow them to select the plugin and version they want.
1. The user can go back to a 'working' version if there are issues with the test
1. The user can easily compare different versions
1. There is no need for 'trusts' for access to git
1. There is no need for 'trusts' for access to cloudsmith

This has the following 'disadvantages':
1. The Plugin Manager will need to be able search cloudsmith repositories for xml files
1. The Plugin Manager will need to validate the xml before using it
1. The Plugin Manager will need to write out the 'new' xml file for use
1. The Plugin Manager will become more complex as it has to handle merging xml files
1. The Plugin Manager will need some changes to the UI to make picking versions possible
1. There may need to be a 'ocpn-plugins-release.xml' file and a 'ocpn-plugins-test.xml' file although I think the xml schema could be changed to allow the one file to contain both types of data.

I do think that some of the disadvantages will need to be addressed anyway.

rgleason, 2020-01-01:
Also, perhaps this will help.  We have a Cloudsmith Organization which can have repositories.
See https://github.com/OpenCPN/OpenCPN/issues/1573
Also OpenCPN public profile https://cloudsmith.io/orgs/opencpn/
Ah ah now I have it, this link should show the Opencpn Organization directories at the bottom
https://cloudsmith.io/~opencpn/repos/

Beta  https://cloudsmith.io/~opencpn/repos/plugins-beta/packages/
Pkg  https://cloudsmith.io/~opencpn/repos/plugins-pkg/packages/
Prod https://cloudsmith.io/~opencpn/repos/plugins-prod/packages/

I've invited Alec, Jon and Dave to be owners of this Organization. I believe they can all upload to these repositories.   We can also ask other members of the team as "owners", managers, member or collaborator. 

I think it might be useful to learn more about this how these repositories can be used for our benefit.

rgleason, 2020-01-01:
I've uploaded 14 most recent files for weather_routiing_pi  to
https://cloudsmith.io/~opencpn/repos/plugins-beta/packages/?sort=-format
These files include   *.tar.gz  and corresponding metadata.xml,  also (4) packages such as .deb, pkg and exe. 

It took a little while doing them one by one, downloading from my cloudsmith repository
https://cloudsmith.io/~rick-gleason/repos/opencpn-plugins-beta/packages/

Then uploading to the OpenCPN organization repository, plugins-beta
https://cloudsmith.io/~opencpn/repos/plugins-beta/packages/?sort=-format

Perhaps someone can make an automatic ocpn_plugins_beta.xml that lands in this plugins-beta repository?
Then the "Beta" channel can point to this repository?   ....or some other clever technique?

bdbcat, 2020-01-02:
@jongough 
Thanks for the new "thought bubble".  I see some concerns.
1. The code changes you propose will tightly link OCPN Plugin Manager runtime directly to cloudsmith, and to our ci process.  I see this as a liability in a runtime package.  Who knows where cloudsmith will be in a year?  We may end up chasing cloudsmith API changes indefinitely.  We are already doing this elsewhere (wxWidgets, GTK, crowdin, etc).  It is not much fun.

Using four branches (alias for channels) at the github ocpn-plugins project avoids much of that problem, or at least ties us only to well structured github artifacts.  As I have proposed, using meta-indirection will not require managing distributed commit rights.  

This is not to say that cloudsmith is not useful.  Perhaps the available repos could be used if desired by any plugin developer.  But as we have said, we do not want to build a process that absolutely requires cloudsmith to even exist.  And certainly not for runtime activities.

I will soon have more to say about the redirection tag in the ocpn-plugins.XML file.  But for now, recall that the redirected XML file can exist anywhere (github artifact, dropbox account, OCPN web server, even cloudsmith).  Also, if I was not clear, the redirected XML file will be merged at runtime, as you envision.

jongough, 2020-01-02:
Don't get me wrong, I was only suggesting CloudSmith as that seems to be flavour of the month, we just need a location to store the requisite files. I just want separation of git from the packages store as I think it gets complicated when trying to use git for that. We just need a 'packages' repository which could be any 'cloud' based, free, storage app.

jongough, 2020-01-02:
I was also suggesting that the 'core' ocpn-plugins.xml file has pointers to wherever the plugin developer wishes to store their package sets. The only thing that needs to be standardised is the naming of the repositories, although this could be done by an extension to the xml schema to include <Prod>, <Beta>, <Alpha> tags. These could either specify the repository name, i.e. what the store is called within the location, or it could be the full url to the store. If this is done then there should be no tie in for the mainline code. All the variable information will be provided by the xml file and the child xml files it will consume.

bdbcat, 2020-01-02:
@jongough 
OK, we are on the same page.  I fully agree that the XML should allow arbitrary specification of storage location for package sets. I am prototyping something now for further consideration.

bdbcat, 2020-01-03:
Closing this, starting new clean issue.

closed 2020-01-03 by bdbcat