OpenCPN/OpenCPN #1591: PI Installer:  weather_routing metadata,  does not load yet

rgleason, 2019-12-26:
Using   https://raw.githubusercontent.com/rgleason/plugins/master/ocpn-plugins-beta.xml
in the advanced URL  

There are some difficulties. also it does not upload yet, fails.Shoort on time. Wo
[image]

Some of the windows don't resize so the list of plugins is not fully available... etc

[image]
[image]
[image]
[image]
[image]
[image]

Short on time. no Inet 2 days.  Sorry.

leamas, 2019-12-26:
Rick,
I really, really appreciate your efforts to modify this and other plugins to work with the installer. 

Really.

That said, we cannot use the opencpn bugtracker for this. That weather_routing does not work with the installer is a problem with weather_routing and should be handled at the plugin's issue tracker. If you discover some bug in the opencpn code, it's another thing. But I see nothing about that in this issue.

rgleason, 2019-12-27:
I will revisit this when I have I better inet. Also Jon has made changes. 
This issue is about how PI Installer presents, in addition to the failure, which may or may not be caused by the gz files.

Next post will be much clearer. Sorry.

closed 2019-12-27 by rgleason