OpenCPN/OpenCPN #1592: Plugin Manger - When debugging with OPENCPN_PLUGIN_DIRS set plugins dont work

jongough, 2019-12-26:
I am trying to test the Weather routing plugin so I am running in debug mode with `OPENCPN_PLUGIN_DIRS=./lib` to have the libs/plugins got from a debuggable directory. I have updated the ocpn-plugins.xml file to point to the latest cloudsmith versions. I open the Plugin Manager and select the '+' icon. I can see the Weather routing plugin, which I try to install by the 'Install' button. The installation process gets the error dialog
[image]
This is the log of the installation

```
08:51:06 AEDT: PluginHandler: using metadata path: /home/jon/.opencpn/ocpn-plugins.xml
08:51:07 AEDT: Installing weather_routing
08:51:09 AEDT: filesize https://dl.cloudsmith.io/public/rick-gleason/opencpn-plugins-beta/raw/names/weather_routing-ubuntu-16.04-tarball/versions/1.13.1.0+82.5c14018/opencpn-plugin-weather_routing-1.13.1.0-ov50-1.16_ubuntu-16.04.tar.gz: 3825733 bytes
08:51:21 AEDT: PlugInManager: loading plugins from /home/jon/opencpn/OpenCPN-5.0/build-kde181/lib/opencpn
08:51:21 AEDT: Using LD_LIBRARY_PATH: /home/jon/opencpn/OpenCPN-5.0/build-kde181/lib/opencpn:/home/jon/opencpn/OpenCPN-5.0/build-kde181/lib/opencpn:/home/jon/opencpn/OpenCPN-5.0/build-kde181/lib/opencpn
08:51:21 AEDT: Using PATH: /home/jon/opencpn/OpenCPN-5.0/build-kde181/lib/opencp/bin:/home/jon/opencpn/OpenCPN-5.0/build-kde181/lib/opencp/bin:/home/jon/opencpn/OpenCPN-5.0/build-kde181/lib/opencp/bin:/home/jon/development/esp/esp-idf/components/esptool_py/esptool:/home/jon/development/esp/esp-idf/components/espcoredump:/home/jon/development/esp/esp-idf/components/partition_table/:/home/jon/.espressif/tools/xtensa-esp32-elf/esp-2019r2-8.2.0/xtensa-esp32-elf/bin:/home/jon/.espressif/tools/xtensa-esp32s2-elf/esp-2019r2-8.2.0/xtensa-esp32s2-elf/bin:/home/jon/.espressif/tools/esp32ulp-elf/2.28.51.20170517/esp32ulp-elf-binutils/bin:/home/jon/.espressif/tools/openocd-esp32/v0.10.0-esp32-20190708/openocd-esp32/bin:/home/jon/.espressif/python_env/idf4.1_py2.7_env/bin:/home/jon/development/esp/esp-idf/tools:/home/jon/bin:/usr/local/sbin:/usr/local/bin:/usr/sbin:/usr/bin:/sbin:/bin:/usr/games:/usr/local/games:/snap/bin
08:51:21 AEDT: Loading plugins from dir: /home/jon/opencpn/OpenCPN-5.0/build-kde181/lib/opencpn
08:51:21 AEDT: PlugInManager searching for PlugIns in location /home/jon/opencpn/OpenCPN-5.0/build-kde181/lib/opencpn
08:51:21 AEDT: Found 2 candidates
08:51:21 AEDT: Checking plugin candidate: /home/jon/opencpn/OpenCPN-5.0/build-kde181/lib/opencpn/libsquiddio_pi.so
08:51:21 AEDT: Checking plugin candidate: /home/jon/opencpn/OpenCPN-5.0/build-kde181/lib/opencpn/libocpn_draw_pi.so
08:51:21 AEDT: ChangeLocale: Language load for:  en_US
08:51:21 AEDT: ChangeLocale: Locale Init OK for:  en_US
08:51:21 AEDT: Loading catalog for opencpn core.
08:51:21 AEDT: Getting DefaultSystemLocale...
08:51:21 AEDT: install: Reloading plugins, before: 2, after:  2
```
In this mode the Plugin Manger installs the plugins to the default locations `./local/lib/opencpn` for the lib, `./local/share/opencpn/plugins` for the data and `./opencpn/plugins/install_data` for the Plugin Manager information. 

When the plugin is looked for none of these locations are referenced. This makes debugging Plugin Manager issues difficult if not impossible.

leamas, 2019-12-26:
In what way is ~/lib "debuggable" and ~/.local/lib not? 

"mystified"

jongough, 2019-12-26:
Try it out. Use your favourite IDE, build OCPN, run it from the debugger using `OPENCPN_PLUGIN_DIRS` set to a location that is not used by the Plugin Manager by default. Open Plugin Manager, install a plugin, Squiddio or Weather routing, using the '+' sign installation process. You will find it will fail to run the plugin. All the parts are in place, but the Plugin Manager does not look in them it looks in the `OPENCPN_PLUGIN_DIRS` location which of course does not have anything in it as it was installed in the default Plugin Manager locations. I explained this in https://github.com/OpenCPN/OpenCPN/issues/1592#issue-542679734. 

I need to be able to run the lib that has been downloaded not the one I can build myself. The build processes are quite different. If I start copying files around you will then claim I should do the testing without doing that. So, I am not copying files around, but I am raising an issue when using the environment variable `OPENCPN_PLUGIN_DIRS`. Fix this issue and then I will continue trying to debug the Weather routing plugin.

jongough, 2019-12-27:
I have now got the plugin to load by adding `~/.local/lib:~/.local/share` to the `OPENCPN_PLUGIN_DIRS` environment variable. So for debugging purposes I now set:
`OPENCPN_PLUGIN_DIRS=./lib:~/.local/lib:~/.local/share`.

These 'default' locations should probably be added after the one provided by `OPENCPN_PLUGIN_DIRS` rather than having to be specifically added.

jongough, 2019-12-27:
Have now removed the 'share' folder and re-ordered the list. I am now using `OPENCPN_PLUGIN_DIRS=~/.local/lib:./lib` to ensure the plugin manager versions get loaded before the non-plugin manager ones.

closed 2021-04-26 by jongough