OpenCPN/OpenCPN #1594: Plugin Manager - Ubuntu - does not check for plugin dependencies

jongough, 2019-12-27:
When using the Plugin Manager installation process on Ubuntu it is using a tarball to do the installation. There is no ability to add dependencies that would normally be installed by the system installation process. Therefore the installation will apparently work but there will be unmet dependencies, so in the best case the plugin will not load, in the worst case it will crash OCPN. This will not be an issue with the plugin but an issue with the installation process.

If a plugin is installed in the non-Plugin Manager method the 'deb' file contains the dependencies that the system installation process can satisfy. This process is needed with the Plugin Manager. Some plugins will use system capabilities that are not needed by OCPN so they should be able to specify extras that need to be added.

bdbcat, 2019-12-27:
@jongough 
I understand.
Such plugins will need to use the "legacy" package installation mechanism.   Only the Plugin author can know this requirement.  If the added functionality of the plugin requiring extra dependencies is important enough to the author, then that plugin cannot participate in the PluginManager scheme.  The plugin author decides whether to play or not.
Such situations will be unlikely, since the goal of cross-platform stability will be difficult to achieve if plugins contain platform specific dependencies beyond the core set..

It is not the intention, and certainly is not practical, to extend the PluginManager into a fully dependency-aware package manager across all platforms.

jongough, 2019-12-27:
I am not suggesting making much of a change to the Plugin Mananger, just that it use platform packages rather than tarballs. The xml file can still be used, the plugins I have been working with generate platform install programs as well as tarballs, and these are stored on CloudSmith alongside the metadata xml. At least using the platform package installer the dependencies will be either installed automatically or the plugin will not be installed because of their lack. The Plugin Manager only needs to 'execute' the platform installation package the platform installation process will do the rest. There is no need to build extra dependency awareness into it. This change should be minor.

With the current method, on Linux and macOS at least, if you 'install' a tarball there is no check that the installed module will actually work on the environment (due to missing dependencies). If it does not work it tends to crash OCPN. The problem then is that for a user to uninstall the plugin they need OCPN running to get to the Plugin Manager to do the uninstall. But if OCPN crashes you cannot get to the Plugin Manager. Whilst you 'can' delete the files if you know where they are stored for most users they are in 'hidden' directories and do not show up in file managers.

I have worked with 2 of the plugins trying to use this new process and Squiddio (modified and built by leamas) crashes OCPN when it is used via the new tarball method and I don't know why and it is very difficult to debug. I am trying to get the Weather Routing plugin working and am assisting rgleason with this process. But again, the tarball process seems to either not load or crash OCPN. The deb file (I am working on Ubuntu mainly) downloaded from CloudSmith installs OK and works.

bdbcat, 2020-03-17:
Jon...
The Managed plugin interface for this coming version of OCPN will not attempt to install platform packages, for at least two reasons:
1.  Installation of packages requires elevation of privileges.  We are attempting to move away from this model for most, or all, users.
2.  Platform packages come in a lot of flavors, creating a geometric plugin maintenance burden.  Installing by tarball, created by relatively stable CI process, greatly eases the plugin dev's workload.  It also allows plugins for all platforms to appear simultaneously.

jongough, 2020-03-18:
1. Installation, until this change, required this so there is no real change. Unfortunately plugins are not always simple and have a multitude of files and may have extra requirements on the platform being installed on. In the case of Ubuntu the 'deb' file contains all the information needed for a package manger to ensure the plugin is suitable to install. If so it will let the user install it. When it comes time to uninstall the package manger can do this (apt remove) and clean up completely (apt purge). Currently with the plugin manager this is not possible, so data and programs can be left lying around on the system. This is why package mangers were built. If we don't want to do that then we should be using 'flatpak', 'AppImange', 'docker', or 'user visible' (i.e. ~/opencpn/plugins/....) directories. The current installation process could even use 'deb' files and then check to ensure that the system is suitable to install the plugin and then install, by extraction, the plugin into the OCPN controlled locations and avoiding using elevated privileges.

2. The tarball process does not help with the number of flavours being needed. In fact it exacerbates this as there is no way to easily add missing packages to the underlaying system. The fact that we are now generating taballs for individual releases of Ubuntu and Debian and gtk2 and gtk3 demonstrates this.The circle-ci and travis-ci build process and the use of XML to define packages is good, but I think we are throwing the baby out with the bathwater and going to cause maintenance issues and continual upgrades to the build process.

bdbcat, 2020-03-18:
Jon...
in reply:
1.  Managed plugins are under strict repo control. Every managed plugin will be tested against all the core systems that it recognizes.  Thus, no plugin tarball will be available in the master catalog that will not install, and function to some degree, on all the systems supported.  Probably we have not yet identified all the system corner cases due to limited alpha testing.  But we will during Beta testing.
Plugins that cannot be installed by the current management process (probably due to dependencies), will need to be published as legacy un-managed packages, or exactly as we are now publishing. Hopefully this will be few in number.
I think it may be a worthwhile goal to explore installation of .debs, by unpacking and moving, etc.  But not for this release.

2.    I guess you are not familiar with Ubuntu PPA publishing.  This is not very convenient, and requires bot-builds for 4 platforms across at least three distros.  For 20 or so plugins. Each one needs some tweak, usually.  That is why we get CF complaints about missing plugins on this or that platform.  It would be a full time job to keep this matrix populated with the latest releases. I know for a fact that the CI process now used for managed plugins just works better, from a developer perspective.

jongough, 2020-03-18:
I do publish my plugins with PPA as well as git and through the OCPN process. The cloudsmith build process is better as, once the new build process is working (almost there), it should be easy for developers to provide the requisite files via cloudsmith. There should be no need for you to build all the environments as it will be done by the developers themselves. If you want to build for production there should only be a couple of changes needed to the new method. I will need to work on that when I have it working for myself and Rick G.

I think the idea of the new process is good, I think the new build process is worthwhile but quite complex (which is why I am trying to simplify it). I think the implementation needs a bit more thought. As a first pass it may work, I just am concerned that if users start to get issues it will be difficult to debug and will be confusing for non-developers. 

Whilst still in 'beta' I think that the edge cases need to be explored and attempts to see what happens when things are not quite as expected should be looked at. I am sort of doing that, but have backed off a bit recently as it seemed I was trying to hit a moving target.

bdbcat, 2020-03-18:
"There should be no need for you to build all the environments as it will be done by the developers themselves. "
Well, that has not happened in the past.  Pavel and I have built essentially all of the "official" PPA packages.  And Pavel is now offline...
So the new CI method is a definite improvement in availability.
I agree that the CI process is complicated, perhaps unnecessarily so.  And a bit fragile. Could be further modularized to allow simple adaptation to an existing CMake build.  That is a worthy goal.
But it is a tall order to build for all platforms with one simple commit....
Edge cases: We will see them in Beta, for sure.  But probably at a rate no greater than today.  I think that if we cover Win, Mac, raspbian, and Ubuntu, we have probably got >95% of our possible base handled.

rgleason, 2020-03-18:
Jon, let me know when testplugin_pi has a good frontend that you think I should deploy to update weather_routing and weatherfax, etc.

jongough, 2020-03-19:
Hi Rick,
   The master branch of testplugin_pi should be working now. I still have intermittent issues with the flatpak builds, but that is to do with one of web resources not being available. It may be that there is 'maintenance' going on that blocks my access at time. Anyway, all the builds should work for you and put all the builds into cloudsmith.

   If it works I would like to see if we can make it so that Dave can build 'production' environments if needed.

rgleason, 2020-03-19:
Jon,  Good news! This looks very good https://github.com/jongough/testplugin_pi
I have your other branch building and uploading here https://github.com/rgleason/testplugin_pi/tree/fix_tpdc_name  and it was so easy to get it all going.

I will pull your master and set it up, then move that to WeatherRouting and Weatherfax next. Thank you.

rgleason, 2020-03-19:
Jon,  Good news! This looks very good https://github.com/jongough/testplugin_pi
I have your other branch building and uploading here https://github.com/rgleason/testplugin_pi/tree/fix_tpdc_name  and it was so easy to get it all going.

Testplugin_pi after a very quick change of personal settings in about 4 places at the top of CMakeLists.txt I had the master branch building and uploading all 12 enviroments except flatpak, which seems to be reluctant. (I tried restart from failure 3-4 times.)   This is great. I'm going to move on to wxroute.

https://app.circleci.com/pipelines/github/rgleason/testplugin_pi/53/workflows/8c77c583-ec1b-46ae-a5f7-f3122efdb61b/jobs/218
```
chmod: changing permissions of '/home/circleci/project/build/cloudsmith-upload.sh': Operation not permitted
+ STABLE_REPO=rick-gleason/opencpn-plugins-prod
+ UNSTABLE_REPO=rick-gleason/opencpn-plugins-beta
+ STABLE_PKG_REPO=rick-gleason/opencpn-plugins-pkg
+ PKG_EXT=gz
+ '[' -z true ']'
+ '[' true ']'
+ BUILD_ID=218
+ BUILD_DIR=/home/circleci/project/build
+ set +x
+ dnf --version
/home/circleci/project/build/cloudsmith-upload.sh: line 45: dnf: command not found
+ apt-get --version
+ sudo apt-get install python3-pip python3-setuptools
E: Could not get lock /var/lib/dpkg/lock - open (11: Resource temporarily unavailable)
E: Unable to lock the administration directory (/var/lib/dpkg/), is another process using it?

Exited with code exit status 100

CircleCI received exit code 100

```

leamas, 2021-01-11:
hm... revisiting this bug after making some initial attempts with the shipdriver_pi plugin. First: the dependency problem is real, and not only on Debian. A quick platform overview:

  - Debian: as stated here, plugins from time to time depends on system packages which must be present for plugin to load.
  - Flatpak: similar, but the flatpak runtime is much larger and well defined. It often (not always) has required packages installed.
  - MacOS: plugins depends sometimes on brew packages. This is OK for developers, but questionable as a user requirement
  - Windows: Similar to MacOS, but using choco. Here, we more often use bundled, pre-compiled libraries to walk around the problem

While the problem is real, we will not try to declare and resolve package dependencies. It's too expensive. Things we perhaps could do:

  - Bundle and rebuild library sources. This is the only option when using flatpak, and tends to work reasonably well here. Could be used also on other platforms, but tends to lead to long dependency chains including stuff which is available in the flatpak runtime.
  - Add a set of "standard plugin libraries" as a dependency of main opencpn, corresponding to the flatpak runtime. Could be Recommends or Suggests, basically making it opt-in or opt-out for users. Should break the long dependency chains, but can't include each and every library used by arbitrary plugins.
  - Instead of statically linking the libraries, load them dynamically. This would make it possible for the plugin loader to load the plugin which then could present a suitable GUI error message for missing libs/packages when libraries cannot be found.

These options could, and probably should, be used in parallel. 

What did I miss?

jongough, 2021-01-11:
Is there anyway that we could use the 'deb' packages within our tar.gz file as that file does contain everything that is needed for the install AND it includes the dependencies in the './control' file. In this way it should be possible to use apt/dpkg to check if the 'Depends' and 'Recommends' are installed during the installation process, and get the user to install them if not before allowing the plugin to be installed. This would be a 'halfway house' in that we are using the Plugin Manager installation process, but gaining the benefits of the platform installation information. I am assuming the other platform install packages have something in them to let the installation process know what is expected so it can be checked for.

leamas, 2021-01-11:
Again: no.

This is too complicated, and also leads to a need to elevate privileges. All of which inevitably will create a mess. I have listed alternatives to this route for the very reason it just won't work. Also note that the problem isn't Debian-specific, we basically have the same situation on all platforms. So the "parse-dpkg" route also means "parse-brew", "parse-choco" and what not.

Of course, this is just me.  But as I understand it, @bdbcat  agrees.

That said, I have listed a few alternatives. No silver bullet, but used together they should resolve most  plugin's need.  Or?

jongough, 2021-01-11:
The install process does not need elevated privileges as all it does is check to see if the packages are installed. In Ubuntu you can use dpkg -s 'package_name' to get package details and dpkg -S file_name to see if a file exists in an installed package, and a user can issue these commands with there normal privileges. apt can also be used in a similar manner as the idea is not to install anything, rather just to check that it is installed and if not let the user know which packages need to be installed.

leamas, 2021-01-11:
There is a need to give a reasonable error message if the libraries are missing, agreed. That's the reason I have listed the option to load libraries dynamically in runtime rather that build time. This would mean that the plugin could be loaded and present a reasonable error message like "Required library xxxx missing, see htttps://whatever" even if the library is missing

Yes, it requires changes to the plugin code. OTOH,the delta wouldn't be that terrifying. And, important, this is a platform agnostic solution which would work on all platforms besides flatpak.

If we combine this with a standard set of plugin dependencies we just assume exists (and are insatlled as a dep of main opencpn) we should be able to cope with most situations. In other cases, we could just bundle and rebuild libs from sources, like flatpak. 

As a starter: How do we compile a list required libraries for different plugins?  Such a  list would make this discussion a bit mor concrete.

jongough, 2021-01-11:
This is from the 'control' file in the control.tar.gz which is found in the deb file for ubuntu:
```
Package: ocpn_draw
Version: 1.8.0.12-ov50
Section: misc
Priority: optional
Architecture: x86_64
Depends: opencpn, bzip2, gzip
Recommends: xcalib,xdg-utils
Installed-Size: 3773
Maintainer: Jon Gough
Description: ocpn_draw_pi PlugIn for OpenCPN

```
Not a particularly good example bit it shows what is available. A different plugin may have extra requirements, ODraw only uses what OCPN does.

leamas, 2021-01-11:
I'm a Debian packager, I know the control file format. And I'm saying: it won't work.  And, repeating myself: Dave agrees.

Can we please drop this particular path, to use the packaging info?

leamas, 2021-01-11:
But OK... we could compile the legacy  packages deps to a list of libraries used. It should not be hard, but boring...

jongough, 2021-01-11:
What is the objection to the 'deb' file, it is only a tar/zip file which contains all the stuff and information needed by the plugin. If the 'deb' file were used in the plugin package and the xml said it was 'deb' format couldn't all the requisite information and files be extracted from that by Plugin Manager and used to do the install in the current way? This process would not use the platform install process, but could use all the useful information that is provided by the package.

You may be repeating yourself, but I don't quite know what will not work. Is it that using the platform installation process will not work? If so, I am no actually suggesting that. Is it that the platform installation package does not contain the information needed by Plugin Manager to install the package, then how come? As the platform installation process works OK. The current xml file could be extended to have extra information if needed. 

I want the build process to be completely automatic, I don't want to be hand building anything. Hence the suggestion of using the platform installation package to carry the contents of what is needed but use Plugin Manager to move the files to the correct locations AND ensure that all the required packages are in place and if not don't do the install with useful messages to the user.

I think we are after the same result, I am just trying to suggest solutions to the issue. I responded to your last sentence

> .....
> 
> As a starter: How do we compile a list required libraries for different plugins? Such a list would make this discussion a bit mor concrete.
> ....
and suggesting a way to achieve the required list.

leamas, 2021-01-11:
The basic objections:
   - It's only works for Debian.
   - The debian package does indeed contains all information, including how to build (/d/rules), how to install (d/\*.install), post-install scripts (d/post\*) and what not. The software handling this is utterly complex. We cannot use this in a meaningful way.
   - OTOH, the upstream packages normally has the required info in a usable way, including build, install and dependencies. This is normally usable info, rebuilding a upstream, well-maintained library isnẗ that hard. 
   - A library with a long, complicated dependenchy chain is IMHO just not usable in a plugin context. Here, we need to get into details in the form of a list of used libraries in different plugins. Before we have this, this discussion is just too abstract.

leamas, 2021-01-11:
Taking a quick look at the legacy plugins at launchpad they all seems to have more or less the same dependencies, most likely copy-pasted rather than designed:

 `debhelper (>= 8.0.0), cmake, cdbs, libgtk2.0-dev, gettext, gpsd, gpsd-clients, libgps-dev, wx-common, libwxgtk3.0-dev, libglu1-mesa-dev, libgtk2.0-dev, wx3.0-headers, libbz2-dev, libtinyxml-dev, libsdl1.2debian, libportaudio2, portaudio19-dev
`
Of these, cmake and cdbs  has nothing to do with runtime deps. 

The current opencpn package deps, removing administrative stuff:
 libarchive-dev,
 libbz2-dev,
 libcurl4-gnutls-dev | libcurl4-openssl-dev,
 libdrm-dev,
 libelf-dev,
 libexif-dev,
 libgdk-pixbuf2.0-dev,
 libgtk-3-dev,
 liblz4-dev,
 liblzma-dev,
 libpango1.0-dev, 
 libsqlite3-dev, 
 libtinyxml-dev,
 libunarr-dev,
 libwxsvg-dev (>= 2:1.5.21),
 libwxgtk3.0-gtk3-dev,
 lsb-release,
 portaudio19-dev

Looking at these, it seems that the common set of plugin dependencies is the opencpn dependencies + 

   - gpsd, gpsd-clients, gpsd-dev
   - libsdl1.2

Given all these libs, which are the plugins requiring other libs? And, which libs are we talking about in these cases?n I'm currently aware of photolayer, requiring libtiff, libproj and libgeotiff.  Have you more examples?

leamas, 2021-01-11:
hm... as of your example [above](https://github.com/OpenCPN/OpenCPN/issues/1594#issuecomment-757694679) draw also need gzip. That package is not libraries but tools like gzip, gunzip etc.  Havn't looked at the code, but it this is about compressing/decompressing files it might be better to code it using for example libarchive. Overall,  using libraries tends to be stable than invoking command line tools. It's also much more portable

leamas, 2021-01-11:
OTOH, I wouldn't be that worried about adding gzip to the list of "common libraries". It's a package most users should have installed anyway.

Ergo: it should be possible to deal with dependencies on a case-by-case basis.

leamas, 2021-01-11:
I have slept on this, and made an important discovery: debian -dev packages contains static libraries. For example, the libgeotiff-dev package contains the libgeotiff.a static lib. (I'm from Fedora, which strongly discourages static libs and wasn't aware how different Debain is w r t this)

The solution then of course should be to link dependencies like libgeotiff and friends statically to the plugin. I havn't done anything like this, but I know that linking could be tricky. But using these libs should basically resolve the Debian situation. Will need some time to test.

rgleason, 2021-01-11:
I ddon't see lib rtldsr  (something like that) used by weatherfax with and rtlsdr dongle to recieve weatherfaxes.

bdbcat, 2021-01-12:
Rather than blue-sky, may we please discuss some (perhaps hypothetical) plugin that is not servered well, or even at all, by the current  architecture?  I don't know of any that are currently blocked by the current situation.  Does anyone?

leamas, 2021-01-12:
My example is the photolayer_pi, which depends on two external libs libgeotiff and proj (sic!).  I have asked @jongough  if he has other examples of plugins which depends on libraries outside the set pulled in by main opencpn.

In any case, my point is that it should be possible to handle such plugins on a case by case basis using static linkage, bundled sources etc. It *might*  perhaps also be useful to add some libs to the set pulled in by opencpn go benefit plugins.

On a sidenote: Is the copy-pasted build-depends [above](https://github.com/OpenCPN/OpenCPN/issues/1594#issuecomment-757793743) sane, does it reflect packages actually required by these plugins?

leamas, 2021-01-12:
@rgleason: Right on spot, these are the examples we need. 

As it is now, this dependency means that Debian users need to install the  librtldsr package, otherwise the plugin won't load. This is a problem, so to speak. Could be handled either by rebuilding the upstream package from sources, or static linkage of the library into the plugin. In either case, problem solved fro Debian users.

A question is of course how how MacOS and Windows are handled with respect to this.  

I don't know

rgleason, 2021-01-12:
Sean, would be better to comment, but librtlsdr is optional for Weatherfax, it will run without it too, it just enables weatherfax to use the dongle's output directly and create a weatherfax.   I have not tried it, but Sean used it a lot crossing the atlantic, so I have included it with the plugin releases.

leamas, 2021-01-12:
Still a problem: If the Debian plugin indeed is linked to librtlsdr it will  fail to load if that package isn't installed.

This illustrates the general problem: We have a number of plugins which are coded for the older scenario when they was installed as .deb packages. It's very easy to add a dependency to such a package which ensures that install the weatherfax plugin on LInux you will also have librtlsdr installed.

However, in the new, pluginloader context there are no dependency mechanisms. This means that in order to make the plugin work, or even load successfully that package must be installed. The situation is a mess, there is even no sane error message since the plugin just not load.

We could perhaps just live with the situation. But having supported plugins which doesn't load is IMHO a plain bug which should be fixed. There are a number of ways outlined [above](https://github.com/OpenCPN/OpenCPN/issues/1594#issuecomment-757652031). In all cases, this means patching the plugin code.

leamas, 2021-01-12:
> The solution then of course should be to link dependencies like libgeotiff and friends statically to the plugin. 

Doesn't work, the Debian archive is compiled without -fPIC which makes it impossible to link against a shared lib. I'll go with the "rebuild from source" option for photolayer_pi. armhf needs some extra care, the build time is too long (hours).

leamas, 2021-01-12:
To  make my point clear: using ldd on weatherfax in plugins as of current mastyer:

    hemulen:~/$ ldd libweatherfax_pi.so  | grep rtl
    librtlsdr.so.0 => not found

I. e., weatherfax has an unresolved runtime dep, and opencpn will rightfully refuse to load it. Can be fixed by installing librtlsdr package, but how should user know?

rgleason, 2021-01-13:
I thought that Linux will install dependencies needed and tat was one of the really nice features of something like Ubuntu.
If not, then shouldn't the llbrary be included with weatherfax release and be installed if needed?
I am sure there are all kinds of probblems. maybe I sould justt forget rtlsdr and that feature is not possible unless you are a Ubuntu wizard!

Thanks Allec.

bdbcat, 2021-01-13:
Devil's Advocate here....

" But having supported plugins which doesn't load is IMHO a plain bug ..."
There is, of course, another way to look at this.
A plugin that does not load on a specific platform should be withdrawn from the catalog, for that platform, and thus will be declared not (officially) supported, or supportable, as written.
To say yet another way: "Yes, there are some desired OpenCPN plugin functionalities that are simply not supportable in the current release.  Sorry about that.  Maybe we could accommodate these plugins by forcing OCPN to link with some specific external libs (not otherwise needed, such as libgeotiff), in some future version.  Or maybe there is some other workaround.  Meanwhile, adventurous users are directed to the plugin's github source..."
And one more observation:  We speak here mostly of linux.  Please understand that the IS NO generally used package manager for Windows, our primary user base.  So, for Windows, any special libraries are going to need to be included with the plugin, probably by static linkage.  Maybe the same for Mac.
So we maybe should consider a "really cool plugin that only can work under linux" to be important, but not so important that we break the present model.

Food for thought...

bdbcat, 2021-01-13:
Speaking of Windows and Mac library issues, now we can understand why OCPN has  large pieces of libGDAL source embedded within.  No other way to get the required cross-platform functionality.  But we live with it...

bdbcat, 2021-01-13:
@leamas...
On this list:
`debhelper (>= 8.0.0), cmake, cdbs, libgtk2.0-dev, gettext, gpsd, gpsd-clients, libgps-dev, wx-common, libwxgtk3.0-dev, libglu1-mesa-dev, libgtk2.0-dev, wx3.0-headers, libbz2-dev, libtinyxml-dev, libsdl1.2debian, libportaudio2, portaudio19-dev`

Probably we can drop the gpsd stuff, ancient history.  Not sure about libsdl1.  From the description, sounds like something added for some specific plugin.

bdbcat, 2021-01-13:
@leamas...
re: "photolayer_pi, which depends on two external libs libgeotiff and proj"
I would consider dynamic loading, and symbol lookup, if the use case is easily confined.

jongough, 2021-01-13:
The thing to remember is that the current build process, frontend2, does actually build the platform installation package as part of the packaging process. It generates the required tar file for plugin manager and the exe/deb/pkg/rpm/... for the platform normal installation process. Three files are stored in cloudsmith for each platform build, xml, tar & platform package. The user can always install the platform package if required. It is not the build process that has the issue, it is the packaging process and the processing of the packaging that is the issue.

What does windows do when you execute a dll which has unresolved requirements (unmet requirements)? Does it gracefully tell the user what the issue is? Does it just 'not work'? Does it crash something? I don't know. Luckily most plugins don't require anything out of the ordinary to be used, however some want to interact with the internet, deal with JSON & XML, process cloud data, etc. and these 'may' require functionality that is not on the users implementation, but may well be on many other users implementations. Shouldn't OCPN, regardless of platform, do a check of some sort to inform the user of any issues due to missing dependencies? This is the current behaviour for installs using the platform installation process. We should not be taking retorgrade steps, or needlessly marking plugins as 'unfit' due to issues with one platform.

I thought the idea behind plugin manager was to be user friendly not user hostile.

Is there a way with the cmake process to identify the underlaying plugin 'requires capability', i.e. a list of 'libraries', and putting this into the xml file so that the plugin manager can check if that functionality is available. If it is not available then a message can be given to the user suggesting installation of the required capability. If we require all plugins to have static linked libraries we are going to get executable bloat as each plugin will be loading the same capabilities just for itself.

Another thought, some of the capabilities probably have to be installed at the system level to make them function (in linux terms, as a root user with administration permissions), so embedding them in a plugin would not necessarily work.

leamas, 2021-01-13:
> A plugin that does not load on a specific platform should be withdrawn from the catalog, for that platform, and thus will be declared not (officially) supported, or supportable, as written.

Another, less drastic way to look at it is to publish the errata and the known work-around to install the corresponding Debian package. All  plugins already have a link to their "website" even if they can't be loaded, so if published here the info is reachable for users, at least with some help

Doing so is of course also to recognize that this is a bug which should be fixed, the info should contain a link to the bug. Which means that we should file separate bugs for the plugins involved, like weatherfax  and photolayer

leamas, 2021-01-13:
> I would consider dynamic loading, and symbol lookup, if the use case is easily confined.

Note that flatpak has the same problem, and solves it by rebuilding the upstream sources. I think this is a better solution in general, since it actually doesn't require plugin changes. I have some "works for me" stuff in place for photolayer,  but it needs polish before it can be published.

leamas, 2021-01-13:
@jongough  Have you concrete examples of plugins with problematic dependencies in the general context of this bug, like Rick's weatherfax and my photolayer?

> Another thought, some of the capabilities probably have to be installed at the system level to make them function (in linux terms, as a root user with administration permissions), so embedding them in a plugin would not necessarily work.

Any examples on this?

leamas, 2021-01-13:
> Speaking of Windows and Mac library issues, now we can understand why OCPN has large pieces of libGDAL source embedded within. No other way to get the required cross-platform functionality. But we live with it...

Indeed. However, in hindsight it would have been better if the bundling was explicit and patches traceable so we could benefit from updated upstream sources over time.

jongough, 2021-01-13:
No I don't have explicit examples for that process as my plugins only make use of the same capabilities as OCPN, currently. But, we are replacing a more complete installation process, which does checks to make sure everything needed is in place, with one that does no checking that everything is installed. This makes the process more susceptible to issues in the future if a platform changes anything that is currently being used. There will be no warnings, just issues. That is the whole thrust of this issue as we are now opening ourselves to problems caused by underlying system updates. If an update invalidates a platform installed object it will not get installed and the user will be warned. We have now removed that sanity check.

I am concerned that we are focusing on our toe nails and the immediate and not really looking forward a bit to try and build robustness into the new process. I seem to be a loan voice in this, but I like to look forward and try to pre-empt easily identifiable issues such that they don't occur.  I am uncomfortable with the 'chuck it in and see what happens' approach. I have said before we are going back to the time before platform install processes were built. They were built to get around exactly this type of problem. Why are we not learning from their experience and building on it?

I have now said enough and am backing out of this thread as my input is not proving useful.

Hakansv, 2021-01-13:
If you don't mind a Windows user to join:
The advice to a non engineer user on my OCPN training sessions is to keep on Windows for a simple life. Of course, for security and possible stability, the Linux world is preferred. "If you for these or more reasons would like Linux but not enjoy digging into commands and stuff go for Openplotter and its inbound update capabilities.
But from now on also Openplotter has to rely on O pluginmanager to update plugins instead of the previous inbound "apt get" So a "dumb" user looking forward approach would be crucial. If that's what Jon is pointing to it would be worth serious consideration. Or -- " If you not are a developer, stick to Windows!"

leamas, 2021-01-13:
The goal here is simple, and could somehow be described like "make the linux world behave like Windows" :) At least, when it comes to plugin dependencies.

One way of describing the situation is that some plugins (not many) needs to be ported from the old plugin framework to the new. Basically, this means that no plugins should have a runtime dependency to anything not already pulled in by opencpn. This is how Windows and Mac always has worked, since they havn't any package managers, at least not available by end users. 

When done, we will all be happy, both Linux and Windows users since plugins will just work.   

> The advice to a non engineer user on my OCPN training sessions is to keep on Windows for a simple life

Can we please put that genie back into the bottle? ;) Otherwise, refer to #2096.

leamas, 2021-01-13:
This seems to boil down to a few plugins which needs fixes  we need to file separate bugs for those. When it comes to weatherfax, where should this be filed?  Sean's main repo doesn't seem appropriate since he hasn't adopted the new model (?). Rick's fork doesn't have issue tracking enabled, neither has Jon's. 

"Confused"

leamas, 2021-01-13:
Bottom line: that the plugin loader doesn't resolve package dependencies should not be considered a bug, it's rather a feature. An important aspect of this is that it reduces complexity. It also imposes a common way to handle dependencies on all platforms, a good thing for a project like this.

Hakansv, 2021-01-13:
> Can we please put that genie back into the bottle? 

He's not flying wild! And as said. I use to advice Openplotter to a native Windows user looking for cheep power saving alternatives. But the fact is >90% of the course attendees are bringing Win PCs. But it may be a Linux user don't need O training?

leamas, 2021-01-13:
> He's not flying wild!

 But, what it he is a she?

Hakansv, 2021-01-13:
Not a ghost! ( Or he -> hen?)

leamas, 2021-01-13:
he -> hen is probably  a better description, given actual genie properties... Unfortunately only applicable in Sweden, though.

rgleason, 2021-01-13:
> A plugin that does not load on a specific platform should be withdrawn from the catalog, for that platform, and thus will be declared not (officially) supported, or supportable, as written.

I am never sure about the operation of all flavors of a plugin and can only test Windows one flavor of Ubuntu, and users don''t seem to report problems that often. Yes more sophisticated users report through github or perhaps the forum, but I wonder if there is a way of sending a failure message... so that the pllugin could be removed and Plugin Manager does not seem broken.   Perhaps this is too complicated to do and would not really solve the problem.  We need better reporting about outcomes I think.

> 
What does windows do when you execute a dll which has unresolved requirements

In windows I've seen dll messages when there is a missing dll..

rgleason, 2021-01-13:
> This seems to boil down to a few plugins which needs fixes we need to file separate bugs for those. When it comes to weatherfax, where should this be filed? Sean's main repo doesn't seem appropriate [...]

Is is possible to enable "Issues"?  I thought all forks were "forced" to use the parent "Issues".  Perhaps I have this wrong. Will investigate.

leamas, 2021-01-13:
> Is is possible to enable "Issues"?  

Yes

leamas, 2021-01-13:
@rgleason: but the important thing right now is weatherfax. Also note the overall mess here: Is it your or Jon's fork which is the master  one here (since it isn't Sean's).

leamas, 2021-01-13:
> If not, then shouldn't the llbrary be included with weatherfax release and be installed if needed?

No. Linux is moving, and a binary library will not link correctly after an update, and will be different for different Ubuntu versions. The way to go is the same as flatpak's which rebuild dependencies from sources.

rgleason, 2021-01-13:
OpenSource  is naturally an organic operation, and is inherently messy.  I can't fix that.

leamas, 2021-01-13:
No, but we can agree on which fork should host the issues, your or JOn's

leamas, 2021-01-14:
I have made a PR against photolayer at https://github.com/Rasbats/photolayer_pi/pull/23. Here, the libs/libgeotiff contains some code which should be possible to use in other plugins when for example rebuilding dependencies from source.

CMake actually shines here, with good mechanisms to download and rebuild an external project in less than 10 lines of code.

[3 more replies]

closed 2021-04-26 by jongough