OpenCPN/OpenCPN #1596: Plug-in mgr; list shows wrong version for oeSENC

bcn2, 2019-12-27:
Typo in the database?
[image]

Hakansv, 2019-12-27:
No 1.11.0 is the API, 111, version not the plugin version. Other issues and and a PR is trying to correct that. Pls stay tuned.

Hakansv, 2019-12-30:
@bcn2 
See my shot.
I think this issue is solved as far as possible. For plugins built for API <= 116 no info about version patch is able to transfer to pluginmanager and O so therefore is that info omitted. For plugins from 117 and up the patch version will be known for O and will also be shown here. The plugin itself can though print its own version as in the preferences. 
The required API is from a recent commit not longer printed in the selected view. I'm not sure I personally like that and it's easy to get back. Opinions?

[image]

rgleason, 2019-12-31:
Hakan, 
Looks good to me, will check it out later, but I wonder if it is possible to show
Version maj. Min.patch (as you have done)
and plugin (minimum) API 
at the top of the plugin's  descriptive listing? 

I think that would be helpful, and not dependent on the internal plugin's programming.  This way, all plugins would automatically conform to some standards for sharing version and api!

Hakansv, 2019-12-31:
We could put the API info back to the selected view like the shot. For the average user I think the information is worthless but not disturbing. I think Alec took it out just because the debate here and don't mind if I PR it back?

[image]

bcn2, 2019-12-31:
A user will not need to know about the API version, as you say - and the info appears in the log just in case. My option would be to omit it.
"Maj.min.patch" shall be there, as it shows: "there is a new version you can get".

Hakansv, 2019-12-31:
OK, betcha. Close this?

rgleason, 2019-12-31:
Incorrect for manual installations. Users should be aware of plugin api.

rgleason, 2019-12-31:
We had this discussion before, several times, bcn.

rgleason, 2019-12-31:
Can you automatically show it at the top under preferences?

bdbcat, 2019-12-31:
I agree with @bcn2 
It is only clutter, and only interesting if there is a problem.
Another viewpoint:  The required API is not shown on current O5 production.  So this discussion applies only to next release. If we do our job correctly, we will not offer to install a plugin that is not API compatible.  All plugins participating in PI will automatically be compatible, and all legacy plugins will also be automatically compatible.  So no user actionable information is communicated by showing the API  requirement.

rgleason, 2019-12-31:
> If we do our job correctly, we will not offer to install a plugin that is not API compatible. All plugins participating in PI will automatically be compatible, and all legacy plugins will also be automatically compatible.

Ok, knowing that this crew follows through on these commitments...

On another matter, are users going to be treated to the Opencpn full version number and opencpn plugin api used, in an About tab our under "?"

closed 2020-04-29 by bcn2