OpenCPN/OpenCPN #1598: Plugin Manager - Ubuntu 16.04 - Attempt to install Weather routing - incorrect error message

jongough, 2019-12-28:
I have attempted to install the Weather Routing Plugin with Plugin Manager into OCPN 5.0.522+5ea57e2. I updated the ocpn-plugins.xml to include this xml:
```
<plugin version="1">
  <name> weather_routing </name>
  <version> 1.13.1 </version>
  <release> 0 </release>
  <summary> Plugin to complete optimal routing with weather files </summary>

  <api-version> 1.16 </api-version>
  <open-source> yes </open-source>
  <author> Sean d'Epagnier </author>
  <source> https://github.com/rgleason/weather_routing_pi </source>

  <description>
    Weather_Routing Plugin optimizes weather routing
  </description>

  <target>ubuntu</target>
  <target-version>16.04</target-version>
  <tarball-url>
    https://dl.cloudsmith.io/public/rick-gleason/opencpn-plugins-beta/raw/names/weather_routing-ubuntu-16.04-tarball/versions/1.13.2.0+91.316ad7b/opencpn-plugin-weather_routing-1.13.2.0-ov50-1.16_ubuntu-16.04.tar.gz
  </tarball-url>
  <info-url> https://opencpn.org/OpenCPN/plugins/weather_routing.html </info-url>
</plugin>
```
I then started OCPN and went ot 'Options/Plugins/+' and selected the Weather Routing PI and clicked install. The tarball was download, but I got a message saying the install had failed:
[image]
I clicked 'OK' and 'Done' and the new plugin did not show up in the list. So I clicked OK to close the options panel. On the OCPN toolbar there was a new Icon:
[image]

Clicking the green puzzle icon (still trying to get to the bottom of why the icon is not showing) it started Weather Routing.
[image]
Going back into 'Settings/Options/Plugins' I can see that the plugin has been installed.
[image]
Here is the OpenCPN.log of the test run.
[opencpn console output from install of weather routing plugin.txt](https://github.com/OpenCPN/OpenCPN/files/4006695/opencpn.console.output.from.install.of.weather.routing.plugin.txt)
Here is the console log from the same run
[opencpn log - weather routing installed but plugin manager saying it wasnt.txt](https://github.com/OpenCPN/OpenCPN/files/4006692/opencpn.log.-.weather.routing.installed.but.plugin.manager.saying.it.wasnt.txt)

jongough, 2019-12-28:
When this occurs if you expand the WeatherRouting plugin you will see that you cannot uninstall it.
[image]

leamas, 2019-12-28:
hm...  From my perspective this is closely related/similar to/duplicate of #1570. The core issue is that there is no code to clean up after failed installations. 

Obviously, here is something missing in the log as well...

jongough, 2019-12-28:
I have the icon working now as I changed to using SVG icons in the build process. However, all other aspects of the issue still exist.

jongough, 2019-12-28:
This is quite different as there is internet access, the plugin downloads, it installs all the files it is supposed to but the Plugin Manager thinks it has failed. There is no real information in the log to help diagnose the issue. The plugin is usable after the 'failed' install. However, it cannot be uninstalled.

jongough, 2019-12-30:
This issue has not been fixed with OCPN 5.0.522+1.0280e45

closed 2020-03-17 by bdbcat