OpenCPN/OpenCPN #1600: [tracker bug] cmake install  update.

leamas, 2019-12-28:
We need to update the cmake install targets + docs:

  - Support uninstall - #1531 
  - Add functionality similar to DESTDIR to windows  - https://github.com/OpenCPN/OpenCPN/issues/1535#issuecomment-568953882
  - Document plugin workflows which installs the plugin directly into the same paths
    as the installer. 

The last bullet is developer support to avoid the need to build, upload and download  plugin just to test them.

jongough, 2019-12-29:
https://github.com/OpenCPN/OpenCPN/issues/1600#issue-543101679
What is meant by 'Support uninstall'? 
What changes are you suggesting to a plugin cmake that would allow a taball install to uninstall itself?

leamas, 2019-12-30:
Made a PR #1608. This is mostly to simplify life for devs, which installs opencpn in different locations from time to time.

Also added some docs in https://github.com/leamas/OpenCPN/wiki/Local-installations-(debugging)

A small step for man...

jongough, 2019-12-30:
The issue then is that you are not really testing the plugin. I can do all that this does from my IDE. Because the plugin is built on my system it will work on my system (all the required stuff is in place). Testing plugin functionality is one thing, testing the installation/uninstallation process is another. 

During testing of the circleci build process I had to build a far more complex environment to try and replicate what circleci does, i.e. I had to have a 'local' web version of git, a Linux VM that looked similar to circleci's, docker running in that VM to enable the circleci docker process to run and generate the installable package, and a web location to download the package from. Then I could test the installation process. None of this is for the feint hearted.

Most of the issues I have been raising have been to do with the installation process of the built tarballs and the management of them after an installation attempt. I also have to keep a couple of 'clean' VM's that I can test the whole process on to ensure I have not contaminated them with newer versions of packages that are likely to be co-installed (installed alongside) on my development machine.

Once the Plugin Manger installation/uninstallation process has been sorted out then it would be unusual for testing of this type to be carried out frequently. It would just be an extra test that is done before moving from beta to release status.

rgleason, 2019-12-31:
Ithink it is highly unlikely many devs will take the time and effort to build an environment for testing such as Jon's, but this is the best way to understand what is happening.

One way around this is to use circlci settings to record failures and leave a report, but these don't seem to work when using containers. At least I've had not success at this.  Thus I was very fotunate and  it was very lucky that Jon was able to do the debugging.

leamas, 2020-02-23:
Filed a PR https://github.com/bdbcat/oesenc_pi/pull/66.

This is a tool for the oesenc flatpak build which installs the plugin directly, without the upload/download cycle. It might be possible to create something similar for other platforms, but  it's somewhat more complicated.

leamas, 2020-02-23:
Status:

   -  Support uninstall - #1531  (FIXED).
   -  Add functionality similar to DESTDIR to windows - https://github.com/OpenCPN/OpenCPN/issues/1535#issuecomment-568953882 -- WONTFIX
   -  Document plugin workflows which installs the plugin directly into the same paths
      as the installer --  Script templates under way (bdbcat/oesenc_pi#66)

closed 2020-03-05 by bdbcat