OpenCPN/OpenCPN #1604: Course up mode causes strange pan issue

transmitterdan, 2019-12-29:
In course up mode some text is not rotated along with the rest of the chart.  Hard to explain but maybe easier to see.

https://www.icloud.com/iclouddrive/0CLufRsjxjb1ReIZ0ajFe_ncg#OpenCPN_5.0.522.1b18d91_2019-12-29_11-27-23

bdbcat, 2019-12-30:
Text is intended to be "unrotated" to a natural horizontal display in course-up mode.  One of the "features" of ENCs.
I can see that some of the legends are not properly located with respect to their objects, though.  Is this what you mean?

transmitterdan, 2019-12-31:
The text orientation is right.  But the anchor point of the text item appears to not respect the rotation of the view-port for some items (.e.g light names).

bdbcat, 2019-12-31:
Should be fixed by d75f5dbdbcd2e9e8086cdca288ec5484a4730c02

transmitterdan, 2019-12-31:
Confirmed.

closed 2019-12-31 by transmitterdan