OpenCPN/OpenCPN #1605: Plugin Manager:  Windows PI destination

rgleason, 2019-12-30:
cd %localappdata%/opencpn/plugins   goes to   "C:\Users\Rick\AppData\Local\opencpn\plugins"

yet how different is    "C:\Users\All Users\opencpn\plugins"  which is being used now?
and contains all the current plugins.

The environment variable %ALLUSERSPROFILE%  used to be  "C:\Documents and Settings\All Users" in WindowsXP and in Windows Vista & &   it was  "C:\ProgramData"

In fact for most modern Windows OS
cd %programdata%/opencpn/plugins  goes to "C:\ProgramData\opencpn\plugins>"

So why not install to a longtime familiar Windows Directory which is the same as
  "C:\Users\All Users\opencpn\plugins"  ?

 Directory of C:\ProgramData\opencpn\plugins
09/09/2019  12:10 PM    <DIR>          .
09/09/2019  12:10 PM    <DIR>          ..
10/24/2017  06:05 PM    <DIR>          celestial_navigation
06/01/2018  08:00 PM    <DIR>          CL-DATA-1.0.tar
06/01/2018  07:52 PM         5,830,716 CL-DATA-1.0.tar.xz
09/03/2018  07:44 AM    <DIR>          climatology_pi
06/04/2019  03:36 PM    <DIR>          deviation_pi
12/26/2019  11:47 AM    <DIR>          install_data
05/28/2016  08:11 AM    <DIR>          logbook
06/29/2016  09:02 PM    <DIR>          ocpn_draw_pi
05/01/2018  07:22 PM    <DIR>          plots
05/29/2018  09:14 PM    <DIR>          testplugin_pi
01/21/2016  05:15 PM    <DIR>          watchdog
03/09/2019  11:36 AM    <DIR>          weatherfax
04/27/2018  09:29 AM    <DIR>          weather_routing
09/16/2017  06:50 AM    <DIR>          weather_routing-old
----
Which is the same directory as

C:\ProgramData\opencpn\plugins>cd C:\Users\All Users\opencpn\plugins

C:\Users\All Users\opencpn\plugins>dir

Directory of C:\Users\All Users\opencpn\plugins

09/09/2019  12:10 PM    <DIR>          .
09/09/2019  12:10 PM    <DIR>          ..
10/24/2017  06:05 PM    <DIR>          celestial_navigation
06/01/2018  08:00 PM    <DIR>          CL-DATA-1.0.tar
06/01/2018  07:52 PM         5,830,716 CL-DATA-1.0.tar.xz
09/03/2018  07:44 AM    <DIR>          climatology_pi
06/04/2019  03:36 PM    <DIR>          deviation_pi
12/26/2019  11:47 AM    <DIR>          install_data
05/28/2016  08:11 AM    <DIR>          logbook
06/29/2016  09:02 PM    <DIR>          ocpn_draw_pi
05/01/2018  07:22 PM    <DIR>          plots
05/29/2018  09:14 PM    <DIR>          testplugin_pi
01/21/2016  05:15 PM    <DIR>          watchdog
03/09/2019  11:36 AM    <DIR>          weatherfax
04/27/2018  09:29 AM    <DIR>          weather_routing
09/16/2017  06:50 AM    <DIR>          weather_routing-old

rgleason, 2019-12-30:
PS: I hope to catch up on testing, to understand this process better.

leamas, 2019-12-30:
The basic reasoning is based on trying adhere to platform (in this case, windows) standards. I googled around and discussed this issue last spring. After some crucial input from Håkan I basically landed in https://stackoverflow.com/questions/16276139/what-is-the-difference-between-programdata-and-appdata.

Reading this + the references it seems that the distinction between user data and program data has been neglected by opencpn up to now.  Linux and MacOS also has this distinction, with different implementations.

To not make things worse, and to keep the overall picture the same on the different platforms I went for %LOCALAPPDATA% when it comes to installer data.

Now, this is certainly not a black/white issue. I could live with a patch moving tings around. Unless there are more important arguments than "we are used to have it this way" I'm not motivated to make such a patch, though.

Also note that users are free to store the data under %LOCALAPPDATA%/opencpn
anywhere by making a symlink as administrator. It's nothing the code cares about.

Hakansv, 2019-12-30:
First the plugin loader should work for the user who just want a one button click to install/update a plugin regardless of earlier, <=v5.0, installed plugins. (Otherwise the present 5.0 release is good enough)
Second is to possible make a tool for "hackers" using alternative plugin versions. But that's so far of less importance?

leamas, 2019-12-30:
> Second is to possible make a tool for "hackers" using alternative plugin versions. But that's so far of less importance?

The Use Custom URL is the basic loophole. It allows a user to define an arbitrary source of the catalog, including updates. A  hacker can of course  also  modify ocpn-plugins.xml. It should IMHO be sufficient for now.
 
> [PI]  should work for the user who just want a one button click to install/update a plugin regardless of earlier, [...]

This is an important issue. Could you please file a separate bug ? It's not really related to the topic here.

jongough, 2019-12-31:
Use the platform installation process from Plugin Manager and avoid all these issues. The platform installation process will put executables and data where it deems to be the best place. It will also uninstall these if needed.

Many years of effort have gone into the platform installation processes, why does OCPN now need to redo all this work? OCPN is not an installation manager.

rgleason, 2020-01-04:
Since both Plugin Packages and Plugin Manager will be supported going forward (certainly during a long transition period as Plugin Manager improves) 

Since the Plugin Manual (wiki) is being revised to 
1. Clarify the three Plugin Types that are available: Default, Plugin Packages and Plugin Manager
2. Clarify the differences in installation and removal.
3. Assist with "manual" removal of Plugin Packages by having good descriptions.
Please review  https://opencpn.org/wiki/dokuwiki/doku.php?id=opencpn:opencpn_user_manual:plugins:install_and_enable
and advise re improvements.
 
Since we now know that the paths used for these Types are: 

Plugin Packages 
- User Plugin Data  %programdata%/opencpn/plugins or C:\Users\All Users\opencpn\plugins
- Executable and Plugin Data  C:\Program Files (x86)\OpenCPN [version number]\plugins

Plugin Manager
-  User Executables and User Plugin Data  %LOCALAPPDATA%\opencpn\plugins
-  or the same C:\Users\[User]\AppData\Local\opencpn\plugins 
-  Thus all plugin executables and the plugin data are in a user accessible folder.

Default Plugins
 - Executable and Data  "C:\Program Files (x86)\OpenCPN [version]\plugins

One last concern is what happens when a Plugin Manager installation or uninstallation fails and Opencpn becomes corrupted and will not start?

rgleason, 2020-01-06:
Closing, with new Issue for last concern.

closed 2020-01-06 by rgleason