OpenCPN/OpenCPN #1606: Plugin Manager: Un-resizeable Plugin Listing, Buttons partially shown

rgleason, 2019-12-30:
Please see the screenshot.  Menu is not resizeable and buttons are partial.  Also I think it should size to what is needed, if that screen space is available.  Thanks.

[image]

leamas, 2019-12-30:
Right. Will look into it, but it might be later rather than sooner (assuming this is non-blocking).

Hakansv, 2019-12-30:
Just to add to the list:
Similar on Win10 and RPi3 for the dialog window "Catalog Manager". It's able to resize but that's not saved to next opening.
[image]

rgleason, 2020-01-02:
Also re Catalog Manager, please see Jon Gough's   "Cancel" and "OK" do the same thing.
https://github.com/OpenCPN/OpenCPN/issues/1612

rgleason, 2020-01-04:
OpenCPN 5.0.22 Windows   cloned and built tonight after Alex's PR was merged.
No joy.  Sorry.   Cannot resize and cannot see any of the controls.  Worse.
[image]

rgleason, 2020-01-04:
Under Custom Catalog URL, after setting the URL to
https://raw.githubusercontent.com/opencpn/plugins/master/ocpn-plugins.xml

Then hit  "Use default location"  and the URL changes to 
https://raw.githubusercontent.com/opencpn/plugins/experimental/ocpn-plugins.xml

Why?  I have  "Master" set in the Catalog Channel!

[image]

rgleason, 2020-01-04:
Hit "Advanced Catalog Update"
Why is Catalog Manager greyed out and blank?  Is it not supposed to be there?
It seems to open and quickly show text, but then it quickly goes grey and the next window opens.
[image]

Notice that I can now see the buttons and there is a More and Less to open the rest of the plugin description.  The earlier problem with not being able to resize to see the buttons is due to the plugin descriptions that are displayed (the plugins allready loaded in the program)

rgleason, 2020-01-04:
Do we need a scroll bar on the icons on the top?
[image]

I know some of this is difficult.

rgleason, 2020-01-04:
More and Less description in the list works, but only for the plugins that are in experimental...
not with other catalogs.

[image]

rgleason, 2020-01-04:
OpenCPN 5.0.22 Windows cloned and built tonight after Alex's PR was merged.
The final problem is that it is impossible to pick  master, Alpha, Beta, master, because it always seems to be set back to "Experimental".

Even when I pick 
Default Catalog    date    "Use as active Catalog"
Latest Available Catalog  date   "Use as active Catalog"

Also when I set  Advanced URL to   https://raw.githubusercontent.com/rgleason/plugins/master/ocpn-plugins.xml

or  https://raw.githubusercontent.com/rgleason/plugins/rg-beta/ocpn-plugins-beta.xml

Nothing Changes in the list. and it always seems to go back to Experimental.

leamas, 2020-01-04:
> Why is Catalog Manager greyed out and blank?

It should really be hidden, and is on other platforms. Don't care about how it looks, it's on the radar and will be fixed.

leamas, 2020-01-04:
> Under Custom Catalog URL, after setting the URL to...

From what I see all looks OK. 
 
  - When you set the Custom URL you don't affect (or is affected by) the branch setting. 
    It's a simple, plain URL unrelatedrelated to the branch setting, overriding everything else. 
  -  The default channel is formed formed using the actual branch (which I guess is experimental)

leamas, 2020-01-04:
> The final problem is that it is impossible to pick master, Alpha, Beta, master,

So far I have not been able to reproduce this...

rgleason, 2020-01-04:
Alec I am telling you that It does not work, the Experimental channel is always reselected.
I cannot get Master, Alpha or Beta to show.

It would also be very useful to have the Plugin Manager Listing to show at the top, which Channel is being used.

rgleason, 2020-01-04:
Also why do some plugins cause the description area to extend the width so that the buttons are not availalbe?  That should not happen ever.  

The new Plugin Manager has to be able to handle all plugin descriptions properly.

rgleason, 2020-01-06:
Ok, this doesn't exist.  Just kidding, but I am closing it anyway.

closed 2020-01-06 by rgleason