OpenCPN/OpenCPN #1607: Plugin Manager: When Plugin is managed by Opencpn, if PI is Involved, it is not uninstallable.

rgleason, 2019-12-30:
How can I easily uninstall this version of Weather_routing so that I can install the tar version from PI Installer?
Both the Install and Uninstall buttons seem to be frozen.

[image]

rgleason, 2019-12-30:
So I went to
C:\ProgramData\opencpn\plugins
and
C:\Users\Rick\AppData\Local\opencpn\plugins
and deleted wxrouting dll and directory

rgleason, 2019-12-30:
Then I start Opencpn again, wxrouting is gone.  
The Advanced URL is the default path, to Leamas repository, so I change that to mine
https://raw.githubusercontent.com/rgleason/plugins/master/ocpn-plugins-beta.xml

and then click "Update Catalog"   The plugin list does not change.
I click "Done" to close that menu, and then have to close the Catalog menu as well (should not have to.)
I am back to the Normal Options> Plugins List.   The list has not changed and I did not expect it too change.

I click on the PI Installer "+"  Now Weather_routing_pi v1.13.1 shows in the list.
I try to install it and there is a note that it fails. This is the most current version from yesterday because I have added all the xml files and rebuilt the ocpn_pluginx_beta.xml

https://raw.githubusercontent.com/rgleason/plugins/master/ocpn-plugins-beta.xml

https://cloudsmith.io/~rick-gleason/repos/opencpn-plugins-beta/packages/

When I check to see if the files have been installed, I find weather_routing_pi installed in
C:\Users\Rick\AppData\Local\opencpn\plugins

but the weather_routing directory is found under 
C:\Users\Rick\AppData\Local\opencpn\plugins\plugins
along with oesenc_pi and radar_pi directories (both dated 11/2)

So I moved weather_routing directory up one directory and deleted the other two directories.

Oddly, there is a Weather_Routing Icon and it works.

[image]

 

[image]

rgleason, 2019-12-30:
So there is another problem.  PI Installer is having trouble when something installs properly, or the weather_routing data files are installing in  plugins/plugins thus causing the problem.

leamas, 2019-12-30:
It might become easier if one understands how the installer works:

  - When it installs a plugin, it creates a list of files. This list is used when uninstalling.
  - When it loads a plugin it checks if the list of files is available. If not, it deems the plugin as installed by something else (probably, a package).  In this case it treats the plugin as read-only, all buttons frozen.

Which boils down to that a plugin installed by for example a package must be uninstalled the same way. The PI will not change it.

The last commits should give a little more feedback on this, an icon with a tooltip  on the very right.

leamas, 2019-12-30:
Sorry, my last comment was a bit off-topic, referred to earlier posts.

leamas, 2019-12-30:
>  PI Installer is having trouble when something installs properly [...]

This should be easy to debug: download the tarball and inspect that paths within it. If those contains plugins/plugins the tarball is broken. If not, PI is.

leamas, 2019-12-30:
That tar file contains:
```
bsb4_pi-1.5.0.3_msvc-10.0.14393/plugins/
bsb4_pi-1.5.0.3_msvc-10.0.14393/plugins/bsb4_pi.dll
```
The dll looks ok, and should end up in the correct location. And so it seems, since it's loaded.

But I cannot see any data files on this tarball...

rgleason, 2019-12-30:
Alec, I should have separated topics.
First - Why is Uninstall button on previous installation of weather_routing_pi greyed out so I have to manually remove the dll?
I think user should be able to uninstall the pi.

Second - Why does the PI install of weather_routing_pi show failure, but the pi icon actually works?

rgleason, 2019-12-30:
Alec wrote:

"Which boils down to that a plugin installed by for example a package must be uninstalled the same way. The PI will not change it."

But the PI Installer should not freeze buttons of previously installed pi so user has to manually remove the dll.

leamas, 2019-12-30:
It basically has to. Packaged plugins (the normal case) are installed using administrative privileges, so PI cannot uninstall it. Furthermore, even with the necessary privileges it doesn't know what files to install besides the dll.

However, Håkan raised an important aspect: user should be able to upgrade using PI even if there is a packaged plugin -- currently this doesn't work. I have asked him to file an issue about this.

leamas, 2019-12-30:
> Second - Why does the PI install of weather_routing_pi show failure, but the pi icon actually works?

No idea. I see no immediate problem when I check the correct tarball. It has both data and the dll at usable locations.

You should check what's installed in %LOCALAPPDATA% and compare with the tarball contents. Does it make sense?

rgleason, 2019-12-30:
Alec, If I understand you correctly, are you saying that a previously installed plugin, installed by the old manual method, gets it's "Uninstall" button greyed or "frozen" so that the user cannot uninstall it normally?

---------

I checked the tarball and it looks normal.
https://cloudsmith.io/~rick-gleason/repos/opencpn-plugins-beta/packages/detail/raw/weather_routing-msvc-10.0.14393-tarball/1.13.2.+80.c284e85/

I checked the ocpn-plugins-beta.xml used to install
https://github.com/rgleason/plugins/blob/master/ocpn-plugins-beta.xml
and it looks normal

```
  <plugin version="1">
    <name> weather_routing </name>
    <version> 1.13.1 </version>
    <release> 0 </release>
    <summary> Plugin to complete optimal routing with weather files </summary>
    <api-version> 1.16 </api-version>
    <open-source> yes </open-source>
    <author> Sean d'Epagnier </author>
    <source> https://github.com/rgleason/weather_routing_pi </source>
    <description>
Weather_Routing Plugin optimizes weather routing
</description>
    <target>msvc</target>
    <target-version>10.0.14393</target-version>
    <tarball-url>
https://dl.cloudsmith.io/public/rick-gleason/opencpn-plugins-beta/raw/names/weather_routing-msvc-10.0.14393-tarball/versions/1.13.1.+70.5c14018/weather_routing_pi-1.13.1.0-ov50-1.16_msvc-10.0.14393.tar.gz
</tarball-url>
    <info-url> https://opencpn.org/OpenCPN/plugins/weather_routing.html </info-url>
  </plugin>

```

So why did it install the weather_routiing data directory to 

**C:\Users\Rick\AppData\Local\opencpn\plugins\plugins?????**

rgleason, 2019-12-30:
Here is the log file copied from good old 
C:\ProgramData\opencpn
or C:\Users\All Users\opencpn
or %programdata%\opencpn

[opencpn.log (2).txt](https://github.com/OpenCPN/OpenCPN/files/4010107/opencpn.log.2.txt)

There are just a couple of errors for testplugin_pi icons.

rgleason, 2019-12-30:
The PI Installer "**+**"  is NOT SELECTED, yet the button "Uninstall" is greyed out and I cannot "Uninstall" the plugins.

[image]

Unfortunately this changes the way the old manual set of plugins operates and is not user friendly.

leamas, 2019-12-30:
Hm... According to the text, this is managed by opencpn; that is, the install manifest is found. And still, the uninstall button is disabled. Need to look at it, BBL.

leamas, 2019-12-30:
> Here is the log file copied from good old [...]

At a glance, it looks OK. In particular, the plugin seems to be successfully loaded. Perhaps the error message is a false positive? We need to wait until the enhanced error reporting is merged to sort this out.

leamas, 2019-12-30:
>  are you saying that a previously installed plugin, installed by the old manual method, gets it's "Uninstall" button greyed [...]

Yes. 

This is basically as it should, but it should not block an update to a pi-managed version.

rgleason, 2019-12-31:
You have confirmed that it is intended to prevent the user from uninstalling an "old" style plugin.

IMHO This changes the original plugin interface and it detrimental, contrary to the user's expectations, idiosyncratic and contrary to the original intentions and objectives for PI Installer to be compatible and xoexist with the existing plugin system.

leamas, 2019-12-31:
> You have confirmed that it is intended to prevent the user from uninstalling an "old" style plugin.

Not preventing from uninstalling. Compared to last version (5.0.0) the situation is exactly the same when it comes to packaged plugins: they must be removed using the same tools as when they where installed, typically using an uninstaller left by the package installation.  Nothing has changed here.

PI *cannot* uninstall such plugins for two reasons: it lacks administrative privileges, and it doesn't know what files to remove.

Hakansv, 2019-12-31:
For Windows it could be possible to look up and, if found, start the uninstall program left by the installer. No rights needed. But I don't think neither Linux nor Mac has any similar binaries?

[image]

leamas, 2020-01-01:
@Hakansv  wrote:

> For Windows it could be possible to look up and, if found, start the uninstall program left by the installer. No rights needed. 

Yes, Rick has also mentioned this, citation needed.

> But I don't think neither Linux nor Mac has any similar binaries?

IIRC, Mac and Android doesn't have these binaries, but OTOH well-defined procedures.  Linux is a complete mess, each distro have their own idea (including flatpak).

It is certainly possibly to make a best-effort attempt to locate and start the windows uninstaller. This is windows-only, and might fail for example if user have used a non-default installation directory

What I have been thinking about is to add some kind of extended info  like a '?' button with extended information about the situation, and how user could remove the packaged plugin. This would at least work as intended on all platforms.

Hakansv, 2020-01-01:
@leamas 
Yes I agree it's a bit of a overdo to try to autotomize this process. 
An info message about the need to uninstall previous installed plugins would be good enough. Maybe including some hints for "normal" uninstall methods for different systems?

leamas, 2020-01-01:
>  Maybe including some hints for "normal" uninstall methods for different systems?

My thinking as well. Assuming that users have internet access while updating plugins, a simple link would do.  But someone needs to write those docs, and we need to agree on the actual URL.

Perhaps we could use the under-used wiki?

rgleason, 2020-01-01:
> You have confirmed that it is intended to prevent the user from uninstalling an "old" style plugin.
>
> Not preventing from uninstalling. Compared to last version (5.0.0) the situation is exactly the [...]

Alec, think about this a little more please.
1. The PI Installer "+" button has NOT been selected
2. "Packaged" Plugins that were previously installed (NOT using PI Installer) should operate exactly as before, Including being able to be "Uninstalled". Which means the Uninstall button should not be "frozen".

I believe that installation of weather_routing_pi was the "packaged" version from my cloudsmith opencpn-plugins-beta repository, that I had manually installed.

I believe that if PI Installer interferes with previous operations of old style manually installed Plugins, there will be a lot of user grief and it will come down on our heads.

I will be testing this further.

I am going to close this report now, it is getting old, although little seems to have been resolved.

closed 2020-01-01 by rgleason

leamas, 2020-01-01:
@rgleason :
But the Uninstall button didn't exist before, it was introduced by the Pi installer. So, compared to  5.0.0,  there is nothing PI  could interfere with?!

@Hakansv : since Rick closed this, could you open a new one about uninstalling packaged plugins  and updating from packaged version to a new, PI managed version?

rgleason, 2020-01-01:
Alec,  You're absolutely right. Version 5 does not have "Uninstall" for Plugins. I am very sorry for thinking so. --Mea culpa.
[image]

rgleason, 2020-01-01:
Alec, since the "Uninstall" does not exist in Version 5 and previous, is there a way to simply not show it when it is "frozen out" by PI Installer?  I think that would be less frustrating for users. It is simply not available.  However I expect that the "Disable" button would work properly for those plugins, just as it has before.

leamas, 2020-01-01:
> I am very sorry for thinking so. (It seemed so natural.)

Don't be sorry! There is absolutely no reason for that ;)

>  is there a way to simply not show it when it is "frozen out" by PI

It is possible. However, it would probably cause even more confusion when the button sometimes exist, sometimes not. The read-only nature of packaged plugins *is* a real problem we need to handle somehow instead of trying it hide it.

When @Hakansv and I discussed it (above) we concluded that some kind of info button which would give more detailed info could be the way to go. That info could then include hints how to actually remove the package on different platforms.

Thoughts?

rgleason, 2020-01-01:
Yes, also, maybe it would be good to make it clear that these plugins (with the frozen "Uninstall" button) are "Old version 5.0 Plugins", and they where never able to be "Uninstalled" from a button. The "Uninstall" button is a new PI Manager/Installer feature that only works with the newer versions of plugins that are installed using PI Manager/Installer.  

Also that it is necessary to use the techniques used previously to remove these "Old version 5.0 Plugins" either by removing the appropriate files (dll, dmg, etc) or using the appropriate uninstall-[pluginname].exe if your OS is windows.

rgleason, 2020-01-01:
I am afraid that we do not discuss removal of a plugin very much in the Manual.
https://opencpn.org/wiki/dokuwiki/doku.php?id=opencpn:opencpn_user_manual:plugins:install_and_enable

Perhaps that is the first place we should add a paragraph.
Also I think the github wiki would be a very good place to have removal information.

rgleason, 2020-01-01:
Please see  the new Paragraph
https://opencpn.org/wiki/dokuwiki/doku.php?id=opencpn:opencpn_user_manual:plugins:install_and_enable#plugin_uninstall_removal

I've made a description for Windows users. 
We need Linux, MacOS and ?
Perhaps someone can add these?