OpenCPN/OpenCPN #1609: Plungin Manager - New icons wrong

jongough, 2019-12-30:
I have just installed OCPN 5.0.522+0280e45 to test a possible fix for internet connectivity missing during plugin installation. I notice that in the Plugin Manager there are now icons on the right hand side of the list of plugins. These are supposed to provide information about the plugin, however they are not always correct.
In the screenshot you can see that OCPN Draw 1.6 is marked with a 'ghost' icon. This is incorrect as this plugin is only installable through a package and not through the new process.
[image]

jongough, 2019-12-30:
Further, weather routing is shown as good (a tick mark) but Plugin Manager is not able to uninstall it even though it did install this plugin.
[image]
[image]

rgleason, 2020-01-05:
Confirmed:
"Further, weather routing is shown as good (a tick mark) but Plugin Manager is not able to uninstall it even though it did install this plugin.
image"

What is the Ghost Icon intended for? I have not seen it yet:
"n the screenshot you can see that OCPN Draw 1.6 is marked with a 'ghost' icon. This is incorrect as this plugin is only installable through a package and not through the new process."

bdbcat, 2020-03-17:
Please test current code, with new UI.

jongough, 2020-03-18:
I now get 'missing' icons and I don't know where it is looking for them as there are no messages in the log to say they are missing.
[image]
[log when plugin dialog showing missing icon.txt](https://github.com/OpenCPN/OpenCPN/files/4346838/log.when.plugin.dialog.showing.missing.icon.txt)

bdbcat, 2020-03-18:
Have you installed the app once?
This is required in order to get the icons moved to the proper run-time location.  Some are old, and already in place.  Others are new...

jongough, 2020-03-18:
No, the app is not installed for this latest test as I am running it from my debugging environment. I have copied all the bitmaps from src/bitmaps to the uidata directory, but it is obviously looking somewhere else as that did not work.

bdbcat, 2020-03-18:
Copy to :
{PREFIX}/share/opencpn/uidata/traditional

jongough, 2020-03-19:
I forgot to mention that in my development environment to make it work and find all the required directories I have to run in 'portable' mode, i.e. adding  '-p' to the command. This makes the build directory the base for finding all artefacts. I have copied the files to my build directory under share/opencpn/uidata/* and I still get the jigsaw icon:
[image]
This is the log that shows there is no error message being generated for these:
[icons not available log.txt](https://github.com/OpenCPN/OpenCPN/files/4352165/icons.not.available.log.txt)

bdbcat, 2020-03-19:
Jon...
What can I say.  It works for production build/install.
Your choices:
1. Live with it.
2. Read the code, and fix it.

bdbcat, 2020-03-19:
pluginmanager.cpp:775

jongough, 2020-03-19:
OK. I have tried running OCPN under Ubuntu 16.04 with the -p option and it works OK. I can live without the icons showing for my build environment.

closed 2020-03-19 by jongough

jongough, 2020-03-19:
Pull request generated to provide a message in opencpn.log