OpenCPN/OpenCPN #1610: Plugin Manager/Installer - OCPN hang after trying Plugin Manger reinstall without internet access

jongough, 2019-12-30:
Installed OCPN 5.0.522+0280e45 to test if [#1570](https://github.com/OpenCPN/OpenCPN/issues/1570) was resolved. I did the same sequence as in the original issue. I installed oesenc 3.3.0, which successfully installed. I disconnected the internet went into the plugin manager and selected 'Resinstall' for oesenc. This resulted in:
[image]
Clicked 'OK'
[image]
Clicked 'Cancel'
[image]
Result, OCPN hangs and has to be closed down with a 'kill' command from a terminal.

bdbcat, 2020-03-17:
Please test with current code.

jongough, 2020-03-18:
Issue does not occur with current code

closed 2020-03-18 by jongough