OpenCPN/OpenCPN #1612: Plugin Manager - Catalog Manager Cancel and OK do the same thing

jongough, 2019-12-31:
There appears to be no difference between 'Cancel' and 'OK' on the Catalog Manager dialog If you go to 'Advanced catalog update...' you get presented with the Catalog Manager dialog box and then the Manage Plugin Catalog dialog displays. If you make changes on this dialog, i.e. select Default Catalog 'Use as active catalog' then select 'Done' it makes no difference if you select 'Cancel' or 'OK' on the Catalog Manager dialog, the last change in the Manage Plugin Catalog dialog does not get applied. This is shown by opening up the 'Advaned catalog update...' and and there will be no difference on the display.

bdbcat, 2020-03-17:
Code and UI redesigned.
Time to close?

jongough, 2020-03-18:
Closing this issue but raised another issue with the dialog: See https://github.com/OpenCPN/OpenCPN/issues/1785#issue-583430585

closed 2020-03-18 by jongough