OpenCPN/OpenCPN #1613: Plugin Manager - Installation of weather routing plugin fails - incorrect error message

jongough, 2019-12-31:
Using Win10 with OCPN 5.0.522+d75f5db the following screen is shown:
[image]
Notice the text after "Installation verification error:"
Here is the log of the installation:
```
7:30:32 AM: Installing weather_routing
7:30:33 AM: filesize https://dl.cloudsmith.io/public/rick-gleason/opencpn-plugins-beta/raw/names/weather_routing-msvc-10.0.14393-tarball/versions/1.13.2.+74.316ad7b/weather_routing_pi-1.13.2.0-ov50-1.16_msvc-10.0.14393.tar.gz: 0 bytes
7:30:39 AM: PlugInManager: loading plugins from C:\Users\jon\AppData\Local\opencpn\plugins;C:\Program Files (x86)\OpenCPN 5.0.522+d75f5db\plugins
7:30:39 AM: Using PATH: C:\Users\jon\AppData\Local\opencpn\plugins;C:\Program Files (x86)\OpenCPN 5.0.522+d75f5db\plugins;C:\Users\jon\AppData\Local\opencpn\plugins;C:\Program Files (x86)\OpenCPN 5.0.522+d75f5db\plugins;C:\Users\jon\AppData\Local\opencpn\plugins;C:\Program Files (x86)\OpenCPN 5.0.522+d75f5db\plugins;C:\Program Files (x86)\Common Files\Oracle\Java\javapath;C:\Windows\system32;C:\Windows;C:\Windows\System32\Wbem;C:\Windows\System32\WindowsPowerShell\v1.0\;C:\wxWidgets-3.1.1;C:\wxWidgets-3.1.1\lib\vc_dll;C:\Program Files (x86)\Poedit\GettextTools\bin;C:\Program Files\CMake\bin;C:\Users\jon\AppData\Local\JSONBuddy;C:\Users\jon\AppData\Local\Microsoft\WindowsApps
7:30:39 AM: Loading plugins from dir: C:\Users\jon\AppData\Local\opencpn\plugins
7:30:39 AM: PlugInManager searching for PlugIns in location C:\Users\jon\AppData\Local\opencpn\plugins
7:30:39 AM: Found 1 candidates
7:30:39 AM: Checking plugin candidate: C:\Users\jon\AppData\Local\opencpn\plugins\weather_routing_pi.dll
7:30:39 AM: PLugin is compatible: true
7:30:39 AM: PlugInManager: Loading PlugIn: C:\Users\jon\AppData\Local\opencpn\plugins\weather_routing_pi.dll
7:30:39 AM: Using private plugin dir: C:\Users\jon\AppData\Local\opencpn
7:30:39 AM: Using plugin data path: C:\Users\jon\AppData\Local\opencpn;C:\Program Files (x86)\OpenCPN 5.0.522+d75f5db\plugins
7:30:39 AM: PlugInManager: Using data dirs from: C:\Users\jon\AppData\Local\opencpn;C:\Program Files (x86)\OpenCPN 5.0.522+d75f5db\plugins
7:30:39 AM: Warnińg: no data directory found, using ""
7:30:39 AM: PlugInManager:  C:\Users\jon\AppData\Local\opencpn\plugins\weather_routing_pi.dll
              API Version detected: 116
              PlugIn Version detected: 113
7:30:39 AM: ChangeLocale: Language load for:  en_US
7:30:39 AM: ChangeLocale: Locale Init OK for:  en_US
7:30:39 AM: Loading catalog for:  opencpn-chartdldr_pi
7:30:39 AM: Loading catalog for:  opencpn-wmm_pi
7:30:39 AM: Loading catalog for:  opencpn-weather_routing_pi
7:30:39 AM: Loading catalog for opencpn core.
7:30:39 AM: Getting DefaultSystemLocale...
7:30:39 AM: Found LanguageInfo for: English
7:30:39 AM: Loading plugins from dir: C:\Program Files (x86)\OpenCPN 5.0.522+d75f5db\plugins
7:30:39 AM: PlugInManager searching for PlugIns in location C:\Program Files (x86)\OpenCPN 5.0.522+d75f5db\plugins
7:30:39 AM: Found 4 candidates
7:30:39 AM: Checking plugin candidate: C:\Program Files (x86)\OpenCPN 5.0.522+d75f5db\plugins\chartdldr_pi.dll
7:30:39 AM: Checking plugin candidate: C:\Program Files (x86)\OpenCPN 5.0.522+d75f5db\plugins\dashboard_pi.dll
7:30:39 AM: Checking plugin candidate: C:\Program Files (x86)\OpenCPN 5.0.522+d75f5db\plugins\grib_pi.dll
7:30:39 AM: Checking plugin candidate: C:\Program Files (x86)\OpenCPN 5.0.522+d75f5db\plugins\wmm_pi.dll
7:30:39 AM: ChangeLocale: Language load for:  en_US
7:30:39 AM: ChangeLocale: Locale Init OK for:  en_US
7:30:39 AM: Loading catalog for:  opencpn-chartdldr_pi
7:30:39 AM: Loading catalog for:  opencpn-wmm_pi
7:30:39 AM: Loading catalog for:  opencpn-weather_routing_pi
7:30:39 AM: Loading catalog for opencpn core.
7:30:39 AM: Getting DefaultSystemLocale...
7:30:39 AM: Found LanguageInfo for: English
7:30:39 AM: install: Reloading plugins, before: 4, after:  5
7:32:10 AM: Installation of weather_routing failed
```
The plugin is installed and runs, but if 'Options/Plugins' is opened the following occurs:
[image]
So it looks like the Plugin Manager processing has an issue.

rgleason, 2020-01-05:
I get this error too. See 
https://github.com/OpenCPN/OpenCPN/issues/1635#issuecomment-570831980
There is also an attached log file

bdbcat, 2020-03-17:
Corrected in code.;
Time to close?

jongough, 2020-03-18:
Weather routing does not cause this issue now.

closed 2020-03-18 by jongough