OpenCPN/OpenCPN #1614: Plugin Manager - Malformed XML causes hang on Ubuntu 16.04

jongough, 2019-12-31:
Take the correct ocpn-plugins.xml file and modify the <tarball-url> tag to read <tarball-url1> (with matching closing tag). Goto 'Options/Plugins/Plugin Manager', select the plugin with the malformed xml and try and install it. An error message is displayed:
[image]
Click 'OK' and OCPN then hangs:
[image]
It has to be killed as it cannot be closed. 
The OCPN log has the following in it:
```
03:51:36 AEDT: PluginHandler: using metadata path: /home/jon/.opencpn/ocpn-plugins.xml
03:52:07 AEDT: Installing weather_routing
03:52:07 AEDT: filesize : 0 bytes
03:52:07 AEDT: Warning: Failed to get '' [<url> malformed]
03:52:28 AEDT: LOGBOOK:  2019-12-28 16:52:28 UTC  DR Lat   33.35800 Lon  -79.28200
03:52:47 AEDT: Installation of weather_routing failed
```
The malformed xml should 'possibly' have been written to the log to help identify the issue.

bdbcat, 2020-03-17:
Jon...
XML files are the responsibility of the Plugin developer.
Presumably we will not release a managed plugin whose metadata file is corrupt.
So this issue should not occur on normal use.

jongough, 2020-03-18:
Unless there is a change to the XML (which could well happen with the current rate of change) and the new version is not backward compatible (there is an issue with backwards compatibility for this change). There should still be validation against an XSD of any of the XML being used to ensure a hang cannot occur.

jongough, 2021-04-26:
The XSD should be used by xml processing to ensure the XML is not corrupt. This can happen for a variety of reasons, so downloading an xml file over the internet it should be check for validity. This is the whole reason for XML and the XSD.

rgleason, 2021-04-26:
Would making this verification be an important requirement for security and to prevention of tampering?

jongough, 2021-04-26:
I would have said yes, the same as for json schema validation, but.....

bdbcat, 2021-05-10:
Changing this to an "enhancement".
I agree that XSD validation of the XML at OCPN runtime is a good idea, though not strictly necessary since the plugins and associated files normally come from a "trusted" source.  And the XML files _are_ validated at source.

jongough, 2021-05-10:
Except where the user imports a file through the Plugin Manager interface. This cannot be trusted as the source is unknown.