OpenCPN/OpenCPN #1615: Plugin Manager - no XSD schema validation done of ocpn-plugins.xml file

jongough, 2019-12-31:
The standard for using xml is to provide a schema and to validate all xml against the schema and if it fails to provide error messages about the failure. There should be no attempt to use the xml until it has been validated. As the ocpn-plugins.xml is being hand maintained and is being used for internet access, installing plugins, etc. it should be validated at every use.

rgleason, 2020-01-04:
Alec made a  check-metadata-url tool.
https://github.com/OpenCPN/plugins/pull/16/commits/9c0b004d9917ae179f4348d59964d234e3c9d418#diff-cf28111ce51ac768c6e8ef8a0ae246fa

I copied it into my local github/plugins/Alpha/tools and tried it from local  github/plugins/Alpha branch  this way

C:\Users\Rick\Documents\GitHub\plugins>check-metadata-urls C:\Users\Rick\Documents\GitHub\plugins\ocpn-plugins.xml
'check-metadata-urls' is not recognized as an internal or external command,
operable program or batch file.

I must be doing something wrong

leamas, 2021-04-15:
> As the ocpn-plugins.xml is being hand maintained and is being used for internet access, installing plugins, etc. it should be validated at every use.

The current  situation:
  - The OpenCPN client by default only downloads from the trusted location at the opencpn/plugins project.
  - In the plugins project, catalogs are checked against the xsd file + some other checks before being published

This scheme accomplishes the same safety as when vallidating the xml in the client as long as cĺieny refrain from using the Custom Url option.  Perhaps not perfect, bit IMHO good enough.

rgleason, 2021-11-03:
Does it make sense to implement xml checking on ocpn-plugins.xml file now?
Should this task remain open?