OpenCPN/OpenCPN #1617: XP-OpenGl

Hakansv, 2020-01-01:
I'm checking performance and so on a WinXP. 
-Using 5.0.522 the general performance is bad using CM93. BSB-Raster are as usual compared to O 5.0
-Using OpenGl and CM93 is not useful at all. CPU load 100%. (CPU wo OpenGl ~15%)
-Raster BSB and oeSENC chart are acceptable both with and wo OpenGl but most texts, like light descriptions are black squares on oeSENC and OpenGl. (Was there a fix for this video card?)

First: should we discuss further XP support before digging deep into this?
Second: Is this OpenGl not useful in 5.0.522?

```
14:04:27: Creating MyFrame...size(1541, 983)  position(130, 0)
14:04:27:  **** Connect stuff
14:04:28: Metrics:  m_display_size_mm: 331     g_Platform->getDisplaySize():  1680:1050   
14:04:28: wxDisplaySize(): 1680 1050
14:04:28: OpenGL-> Renderer String: MOBILITY RADEON X600 x86/SSE2
14:04:28: OpenGL-> Version reported:  1.5.4898 WinXP Release
14:04:28: OpenGL-> Texture rectangle format: 0
14:04:28: OpenGL-> DISABLING VBO for Intel test.
14:04:28: OpenGL-> Vertexbuffer Objects unavailable
14:04:28: OpenGL-> Framebuffer Objects unavailable
14:04:28: OpenGL-> Using Stencil buffer clipping
14:04:28: OpenGL-> Not Using compression
14:04:28: OpenGL-> Minimum cartographic line width:  1.0
14:04:28: OpenGL-> Minimum symbol line width:  1.0
14:04:28: BuildFBO 3
14:04:28: BuildFBO 3
```

bdbcat, 2020-01-02:
@Hakansv 
Without Framebuffer Objects, it will indeed be slow on an older CPU.
Is this the same OpenGL signature seen on O5.0?

Hakansv, 2020-01-02:
No, it's about the same.
And if I remember correct there are som prerequisites given somewhere for OpenCPN:
Like WinXP SP3 - Win10
And for OpenGl >= version 2.0
If that's correct I've no excuse for this XP's OG v:1.5.4898 (Strange numbers of decimal points..)
Thanks
Håkan
```
18:02:24: 5.0.2 (My own 5.0)
18:02:24: wxWidgets version: wxWidgets 3.1.2 Microsoft Windows NT 32 bit wxMSW
18:02:24: OS Version reports as:  5.1

18:02:38: Metrics:  m_display_size_mm: 331     wxDisplaySize:  1680:1050   
18:02:38: Loading Tide/Current data source: C:\Program\OpenCPN\tcdata\HARMONIC.IDX
18:02:39: OpenGL-> Renderer String: MOBILITY RADEON X600 x86/SSE2
18:02:39: OpenGL-> Version reported:  1.5.4898 WinXP Release
18:02:39: OpenGL-> Texture rectangle format: 0
18:02:39: OpenGL-> Using Vertexbuffer Objects
18:02:39: OpenGL-> Framebuffer Objects unavailable
18:02:39: OpenGL-> Using Stencil buffer clipping
18:02:39: OpenGL-> Not Using compression
18:02:39: OpenGL-> Minimum cartographic line width:  1.0
18:02:39: OpenGL-> Minimum symbol line width:  1.0
18:02:39: ChartDB Cache policy:  Application target is 1023 MBytes
```

closed 2020-01-12 by Hakansv