OpenCPN/OpenCPN #1618: Raspberry RPi install package

transmitterdan, 2020-01-01:
As there are no Cloudsmith RPi binary packages I took the liberty to create one for raspbian/stretch.  It is here:

https://cloudsmith.io/~dan-dickey/repos/opencpn-unstable-raspbian-stretch/packages/

If anyone cares to test it this package installs into /usr/local since it is a development version.  To make this run smoothly I had to install the latest kernel 
```
sudo rpi-update
```
and add the cma=384M option to the /boot/cmdline.txt file similar to this:
```
dwc_otg.lpm_enable=0 cma=384M blah blah blah
```
The important bit is the cma=384M. Leave the rest as is. Without cma=384M OpenGL will not work.  But with that option O will run for days.

Hakansv, 2020-01-01:
@transmitterdan 
Just for info. And I'm not sure of expected difference.
On RPi3 stretch
I use to build from master and have had no OpenGl problems. Has been running for days.
I tried your cma=384M change for any difference but nothing found om my system. I haven't tried your package though. 

```
Using cma=384M
20:35:27 CET: wxDisplaySize(): 800 480
20:35:27 CET: Creating glChartCanvas
20:35:27 CET: Metrics:  m_display_size_mm: 153     g_Platform->getDisplaySize():  800:480   
20:35:27 CET: wxDisplaySize(): 800 480
20:35:27 CET: OpenGL-> Renderer String: Gallium 0.4 on llvmpipe (LLVM 3.9, 128 bits)
20:35:27 CET: OpenGL-> Version reported:  3.0 Mesa 13.0.6
20:35:27 CET: OpenGL-> Texture rectangle format: de1
20:35:27 CET: OpenGL-> Using Vertexbuffer Objects
20:35:27 CET: OpenGL Framebuffer OK, size = 782
20:35:27 CET: OpenGL-> Using Framebuffer Objects
20:35:27 CET: OpenGL-> Using FBO Stencil buffer
20:35:27 CET: OpenGL-> Using Stencil buffer clipping
20:35:27 CET: OpenGL-> Using Scissor Clipping
20:35:27 CET: OpenGL-> Not Using compression
20:35:27 CET: OpenGL-> Minimum cartographic line width:  1.0
20:35:27 CET: OpenGL-> Minimum symbol line width:  1.0

Not using cma=384M
20:54:38 CET: wxDisplaySize(): 800 480
20:54:38 CET: OpenGL-> Renderer String: Gallium 0.4 on llvmpipe (LLVM 3.9, 128 bits)
20:54:38 CET: OpenGL-> Version reported:  3.0 Mesa 13.0.6
20:54:38 CET: OpenGL-> Texture rectangle format: de1
20:54:38 CET: OpenGL-> Using Vertexbuffer Objects
20:54:38 CET: OpenGL Framebuffer OK, size = 782
20:54:38 CET: OpenGL-> Using Framebuffer Objects
20:54:38 CET: OpenGL-> Using FBO Stencil buffer
20:54:38 CET: OpenGL-> Using Stencil buffer clipping
20:54:38 CET: OpenGL-> Using Scissor Clipping
20:54:38 CET: OpenGL-> Not Using compression
20:54:38 CET: OpenGL-> Minimum cartographic line width:  1.0
20:54:38 CET: OpenGL-> Minimum symbol line width:  1.0
```

leamas, 2020-01-02:
@transmitterdan This is great! In my opinion, Rpi is one of the most important reasons to create .deb packages -- flatpak works on regular desktops. 

I presume they are built using cmake's .deb generation. We might want to consider to build using the downstream packaging which create packages good enough to be accepted in Debian. 

But, again: great work!

transmitterdan, 2020-01-03:
> @transmitterdan
> Just for info. And I'm not sure of expected difference.
> On RPi3 stretch
> I use to build from master and have had no OpenGl problems. Has been running for days.
> I tried your cma=384M change for any difference but nothing found om my system. I haven't tried your package though.

In your file /boot/config.txt there may be a line as follows:
```
dtoverlay=vc4-fkms-v3d
```
In that case I don't think cma= does anything special.  The cma option is for when I use:
```
dtoverlay=vc4-kms-v3d
```

Hakansv, 2020-01-03:
No, neither of the dtoverlay are active.
The only change I have are: (The two first for the 7" monitor)
framebuffer_width=800
framebuffer_hight=480
framebuffer_depth=32
framebuffer_ignore_alpha=1

transmitterdan, 2020-01-03:
@Hakansv 

Ah, then you are not using hardware acceleration to get OpenGL. That's ok too.

free-x, 2020-01-04:
@transmitterdan 

> and add the cma=384M option to the /boot/cmdline.txt file similar to this:

it's much better with cma=512M
@Hakansv 
This option is very usefull if you are using RNC and MBtiles

Greeting
free-x aka BlackSea@cf

free-x, 2020-01-06:
for those who want to test current development OpenCPN packages on Debian/Raspbian Buster , I have created a repository
Here's small howto:
```
sudo apt install apt-transport-https
wget -O - https://www.free-x.de/debian/oss.boating.gpg.key  | sudo apt-key add - 
sudo vi /etc/apt/sources.list.d/oss.list
deb https://www.free-x.de/debpreview/ buster main 
deb https://www.free-x.de/debian/ buster main contrib non-free
sudo apt update
sudo apt install opencpn
```

Regards
free-x

closed 2021-01-09 by transmitterdan