OpenCPN/OpenCPN #1619: Plugin Manager - build for Android version?

rgleason, 2020-01-01:
In previous versions of weather_routing,  Sean has developed the means to build for Android. We still have "buildandroid/build_android.cmake"
https://github.com/seandepagnier/weather_routing_pi/blob/master/buildandroid/build_android.cmake

I am now wondering if this still works, if we should include a script for Android and build and deploy it to cloudsmith?

There are a number of plugins that have migrated to Android, partially due to Sean's work, and it might be a good idea to extend the PI Installer support to Android, or at least have the tar and metadata available in case the PI Installer gets extended to Android?

Alec may have some concerns about overburdening the circleci and cloudsmith system with mulitple OS however.

bdbcat, 2020-01-02:
Rick...
Android is its own beast, easier to deal with in some ways, much more difficult in others.  I will need to spend some think time on this, but right now a much lower priority than the bigger picture.
Please stand by.

rgleason, 2020-01-02:
Thanks for the clarification.

rgleason, 2020-01-06:
I am closing this

closed 2020-01-06 by rgleason