OpenCPN/OpenCPN #1622: Plugin Manager: Tarballs and Gz

rgleason, 2020-01-02:
Alec, 

I know that early on you discussed the preference for use of tarballs and gz for all installations instead of packages such as exe, pkg and deb and the use of NSIS.  Can you explain this thought process for the rest of us, and your reasons for this approach in the context of services available and distribution considerations? 
It would be very helpful for us, as we keep thinking the older package system was well honed, albeit not a complete "Plugin Manager" which included an "Uninstall" button.  I think we all recognize that the older package system did require  most people to just 'click past' the admin stuff, or just type in the required password, which is something we've all gotten used to, so it would probably not be the driving reason to use tarballs.

It will help for us all to understand this decision.

Best
Rick

leamas, 2020-01-03:
This is a complicated late guest to the party. Meaningful discussions on these underpinnings needs a working prototype or so.

Design goals
-----------------

  - Simple, one-button click install for user.
  - Simple, one-button click uninstall for user.
  - Common UX on all platforms.
  - All devs: Simple, well-defined model of what a plugin is  installation-wise.
  - Ocpn devs: Platform-independent code (mostly). 
  - Plugins devs: Make it simpler to provide plugins, avoid need to provide packages

Discussion
--------------
The simple, platform-independent  one-click user actions are simply not possible using native packages. Reasons include that uninstall just isn't possible,  technical problems when communicating with privileged code, different or missing GUIs and fragmented linux solutions.

For plugin developers the new model boils down to that the they need to support *make install* instead of providing packages. This is much easier, and cmake gives tons of support.

For opencpn we get a common, unified  platform with not that much platform-dependent code (estimated < 300 lines).  

Both ocpn and plugin devs benefits from the more restricted model of what a plugin actually is. When using platform packages a plugins cold basically install all sorts of files in all sorts of places, run scripts on install/uninstall, pull in platform dependencies etc. Even if most (almost all?) plugins only installs a library and some data files, any design using platform  packages will be hampered by what a platform package *might* do.

As you see, all this boils down to a solution similar to the browsers, IDE:s, multmedia tools etc. out there: Use an application-specific format for the plugin (.tar.gz in this case).

rgleason, 2020-01-03:
Thank you Alec.
Thiught about the greyed out "Uninstall" button for older packaged plugins.
You already have identified it as an older packaged plugin, it should have its own identifying plugin type icon in the list and the greyed button should say "Manual uninstall"  There ideally would be a link to further offline documentation/ readme/ wiki about this manual process.  The plugin manual now has more detailed description about this.

rgleason, 2020-01-04:
Alec, 
I've added your newer Design Goals and Discussion to this Dev Manual.
https://opencpn.org/wiki/dokuwiki/doku.php?id=opencpn:developer_manual:pi_installer_summary

Since there are no further comments, I am closing this.  If devs wish to comment further, they still can.

closed 2020-01-04 by rgleason