OpenCPN/OpenCPN #1623: Grib - email request updated every second

Hakansv, 2020-01-02:
On Win10.
The request data, marked on attached pict, is updated every second. Some msec glimpse to zero and then back again.
[image]

Hakansv, 2020-02-25:
Some more info: The position data is flickering only in double Canvas mode, not when on a single canvas. Does it try to catch both canvases, alternate every second?

bdbcat, 2020-04-26:
Sorry, not reproducible.  For me, it picks up the location bounded by the canvas active at the moment the request (@) button is pressed.  Does not shift.

Any special conditions?  GPS active?

Hakansv, 2020-04-26:
Dave..
No, double canvas is the only condition that matters.
It's not dependent of any other status like OpenGL on/off, used charts, NMEA data on/off, selected grib data, provider or forecast model.

You wrote: "location bounded by the canvas active at the moment"
```
But this function forces it to canvas(1) only (called from "OnMouseEventTimer(...")
GetGRIBCanvas()
// If multicanvas are active, render the overlay on the right canvas only
```
Could there be another place looking at canvas(0) also?
The position values seems to alter between left and right canvas every second. If I succeed to cover the same area on both, left and right, canvas the position data shift are less. Degrees are stable.

A video link while I change the size of left canvas. (~9MB)
https://www.dropbox.com/s/hcxwql7qg9ngl8m/Grib_area.mp4?dl=1

Hakansv, 2020-04-26:
As said this is on Win only. Two different tested. 
As a try to trace I've come to put a log here if any help:
grib_pi.cpp:561
`if (m_pGribCtrlBar->pReq_Dialog) {
        m_pGribCtrlBar->pReq_Dialog->RenderGlZoneOverlay();
        wxLogMessage(wxString::Format(_T("** Test3: vp lat_max %f"), vp->lat_max));
    }`
The value is constantly changing and from different canvas: (67,6 is from the left canvas)
22:27:29.220 MESSAGE grib_pi.cpp:562 ** Test3: vp lat_max 58.020116
22:27:29.237 MESSAGE grib_pi.cpp:562 ** Test3: vp lat_max 58.020116
22:27:30.199 MESSAGE grib_pi.cpp:562 ** Test3: vp lat_max 67.654363
22:27:30.226 MESSAGE grib_pi.cpp:562 ** Test3: vp lat_max 58.019988
22:27:30.243 MESSAGE grib_pi.cpp:562 ** Test3: vp lat_max 58.019988
22:27:31.183 MESSAGE grib_pi.cpp:562 ** Test3: vp lat_max 67.654363
22:27:31.212 MESSAGE grib_pi.cpp:562 ** Test3: vp lat_max 58.019988
22:27:31.230 MESSAGE grib_pi.cpp:562 ** Test3: vp lat_max 58.019988
22:27:32.189 MESSAGE grib_pi.cpp:562 ** Test3: vp lat_max 67.654363

bdbcat, 2020-04-27:
Should be correced in github now, or very soon.

Hakansv, 2020-04-27:
Now it works fine and refer to the right canvas only.
Thanks

closed 2020-04-27 by Hakansv