OpenCPN/OpenCPN #1626: Plugin Manager: Uninstallation

rgleason, 2020-01-03:
Jon, Hakan & I have expressed concern and a need for improvements and clarity about Uninstallation of Plugins Generally.  I believe Alec agrees, and has asked Hakan to make an Issue about it.

"The current package process makes the installation process separate from the run time process. So if OCPN crashes the packages can be uninstalled easily."

"If the install breaks OCPN then it must be uninstallable easily to allow OCPN to work again. The current package process makes the installation process separate from the run time process. So if OCPN crashes the packages can be uninstalled easily."

Regarding **Un-Installation of Plugins**

**Existing plugins  (packages - exe, pkg, deb, etc)**

1. Windows - Use the uninstall[plugin].exe
2. Linux - we need to explain this better.
3. MacOS  - we need to explain this better.

Under the Plugin Manual, "Install & Enable", please see 
https://opencpn.org/wiki/dokuwiki/doku.php?id=opencpn:opencpn_user_manual:plugins:install_and_enable#plugin_uninstall_removal
explanation for Uninstalling current plugins for Windows.  We need to add Linux and MacOS descriptions as well (with locations).

Additionally with the Plugin Manager version of OpenCPN, the new "Uninstall" button, counter-intuitively shows as "frozen" or "greyed out" for all current plugins. To a user this will be very confusing and there needs to be more clarity and explanation for them how to remove the plugin. 

**New Plugin Manager plugins  (tarball & gz)**

1. Uninstallation button for new plugins only (from within OpenCPN)
2. If OpenCPN fails due to a plugin and will not open, what happens?
3. How can the failed plugin be removed relatively easily?
4. Is there an automatic cleanup technique upon restart, or some alternative action to fix it?

jongough, 2020-01-03:
Existing plugins
item 2: Linux users would use yum (redhat based) or apt or dpkg (debian/ubuntu base) command lines as well as the gui's, Synaptic Package Manger (debian/ubuntu), gnome-software (redhat)
item 3: This is less clean, but this link shows how to uninstall macOS sotware: [https://techsviewer.com/how-to-completely-uninstall-application-in-mac/](https://techsviewer.com/how-to-completely-uninstall-application-in-mac/)

rgleason, 2020-01-03:
Thanks Jon, those are added now.
https://opencpn.org/wiki/dokuwiki/doku.php?id=opencpn:opencpn_user_manual:plugins:install_and_enable

jongough, 2020-01-03:
I think the installing a plugin section needs updating as most plugins provide platform installation packages (*.exe, *.deb, *.pkg, *.dmg) that allow the plugin to be installed by downloading it from various locations, the latest of which is cloudsmith. These packages can be installed by 'running' them on the platform they are intended for (this is indicated in the file name). 

The uninstall section mentions the use of the package managers but the installation process does not. Also, OCPN 5.x changed the version of wxWidgets and the screen display (MUI) so I 'think'  most plugins had to be regenerated for this.

leamas, 2020-01-03:
hm... short of time. But I agree, we need updated docs on this.

Looking at the link, I have a first concern: From a user perspective I think it needs to be re-organized like this:

   - Windows
          + Installing
          + Enabling (link to common text)
          + Uninstalling
          + Troubleshooting

  - MacOS
          + Installing
          + Enabling (link to common text)
          + Uninstalling
          + Troubleshooting
  
  - Etc...

That is, collect everything for windows users in one place and so on. This makes it so much easier to find what you are looking for. It would also make it possible to create direct links from opencpn to the proper section since we know the platform we are running on.

More later, in particular on Linux.

rgleason, 2020-01-03:
Good points from both of you. 

Thought about the greyed out "Uninstall" button for older packaged plugins.
You already have identified it as an older packaged plugin, it should have its own identifying plugin type icon in the list and the greyed button should say "Manual uninstall"  

There ideally would be a link to further offline documentation/ readme/ wiki about this manual process.  The plugin manual now has more detailed description about this.

rgleason, 2020-01-03:
Jon and I have been working on this page
https://opencpn.org/wiki/dokuwiki/doku.php?id=opencpn:opencpn_user_manual:plugins:install_and_enable
and it is clearer, but slowly working into the outline Alec suggested.

I have not been able to see the full screen after hitting "+" as I have not been able to download a new Window package with Alec's recent improvements, but I believe the "Enable" and "Disable" selection works the same for all plugins that load.  Sometime today I might have time to download and check it out.

I have manually installed a Packaged Plugin "Dashboard_Tactics" and have included a screenshot.
This Plugin, when hovered at the right Checkbox  pops  "Plugin managed by Opencpn".  I would like it to have a different "Package" Icon (perhaps just a box with no check), and have the hover message say something like "Plugin package manually install & uninstalled"

Also the greyed out "Uninstall" button should be greyed out "Manual uninstall"
[image]

leamas, 2020-01-04:
>  have manually installed a Packaged Plugin "Dashboard_Tactics" and have included a screenshot.
This Plugin, when hovered at the right Checkbox pops "Plugin managed by Opencpn"

This is a plain bug which should be fixed. There are icons and tooltips in place, but somehow the classification seem s to fail. Will look into it

rgleason, 2020-01-04:
Thanks, would like to see a different icon to indicate a different "type" of plugin "Plugin Package" See this documentation please. 
 https://opencpn.org/wiki/dokuwiki/doku.php?id=opencpn:opencpn_user_manual:plugins:install_and_enable#plugin_types

Thanks!

rgleason, 2020-01-06:
Closing, I assume this is in process of being fixed.

closed 2020-01-06 by rgleason

jongough, 2020-01-07:
I would only close issues when they are fixed, no when it looks like someone may look at it. We have had issues automatically closed when pull requests have been merged but the issue has not been fixed. I actually raised an issue about this.