OpenCPN/OpenCPN #1627: Extensible plugin metadata implemented.

bdbcat, 2020-01-03:


bdbcat, 2020-01-03:
Okay, here is what I have done:

1.  Shifted hardcoded urls to OpenCPN/plugins project on github.
2.  Created 3 new branches on that project: (Beta/Alpha/Experimental)
3.  Implemented a meta-file structure in the plugin manager, allowing a plugin spec in any .xml file to re-direct to another .xml file, recursively.
4.  I demonstrate this simply in the Experimental branch.  There, in ocpn-plugins.xml, you will see a plugin defined like this:

  <plugin version="1">
    <meta-url> https://raw.githubusercontent.com/bdbcat/oesenc_pi/master/catalog_entry.xml </meta-url>
  </plugin>

In the re-direct target ("catalog_entry" file) you will see a normal plugin definition .xml file.  At run time, with the manager channel directed to Experimental, the plugin manager will read this file and parse the contents.  If all goes well, you will see a new oesenc plugin ready for test.  Actually, for this demo, the redirected tarball has not been updated, but it is easy to see that it could be in real life. I changed only the <summary> tag.
Some notes:
1.  The re-direct can go anywhere.  I chose github, just to experiment with a possible workflow.
2.  The re-direct can contain multiple plugin definitions. Specifying all plugins implemented for the array of targets to be supported seems natural.
3.  The re-direct can contain re-directs itself, ad-infinitum.
4.  If the re-direct contains an entry that matches an existing plugin, the existing plugin specification will be updated.  The match must be "exact", that is name, version, release, target, target_version, must match to enable updating.
5.  And if not clear, the contents of the re-direct can point to tarballs residing anywhere accessible by url.

I will accept PRs against OpenCPN/plugins, Experimental and Alpha branches, to add callouts to your plugins under test.  As you can imagine, the PRs will naturally migrate from Exp. to Alpha to Beta over time.

Or, for active plugin devs, I can imagine that a reasonable PR (to branch Alpha) might add an entry to ocpn-plugins.xml that looks like:

  <plugin version="1">
    <meta-url> https://wherever.janedoe/plugins.xml </meta-url>
  </plugin>

This gives jane a private and extensible sandbox to play with, and allows all testers to access the results.  It then also falls on jane to communicate with testers as the plugin(s) evolve.

I have tested some simple scenarios locally.  I have not tested deep recursive environments.  As noted in another issue, we do need XML validation against XSD to harden this code.  Should be a priority.

Finally, please keep in mind that this is all a development assist tool.  In Production, the "master" ocpn-plugins.xml will not contain re-directs, in the interests of stability and security.  So maybe we can live with any quirks that appear as this scheme is exercised during development.

leamas, 2020-01-03:
Before we can validate the xsd file needs to be updated. 

Note there is validation support available, and that we could provide a git hook which blocks or warns  if code to be committed fails to validate. This could be a first step.

leamas, 2020-01-03:
I'm on the xsd update, stay tuned...

leamas, 2020-01-03:
Done: https://github.com/OpenCPN/plugins/pull/14

rgleason, 2020-01-03:
Made this PR , please advise, re name of catalog file, etc. to get it right. Should it be redone, back to ocpn-plugins.xml  in the Beta branch?

Also I don't want to be too specific about the beta plugins as they will change as they get tested.
https://github.com/OpenCPN/plugins/pull/15

leamas, 2020-01-04:
Added tool to check url accessibility: https://github.com/OpenCPN/plugins/pull/16

leamas, 2020-01-04:
Checking current builds. At a glance, it looks OK on linux and flatpak. However, on windows it seems to fail badly when changing channel, get a message "Cannot parse XML" or so.

It does make sense somehow. When changing channel the xml file for the new channel is downloaded and parsed. What doesn't make sense is that the parsing fails only on windows.
Also, IIRC this was tested at some point and was working then.

Line delimiters issues?  "mystified"

@bdbcat: Thoughts?

leamas, 2020-01-04:
[keeping log here since it's likely I will have to abort this soon]

It seems to be about case-sensitive urls. On windows the effective url to the experimental branch used is
 `https://raw.githubusercontent.com/opencpn/plugins/experimental/ocpn-plugins.xml`
This gives a 404 + a bad error message about failed parsing. The correct url is
` https://raw.githubusercontent.com/OpenCPN/plugins/Experimental/ocpn-plugins.xml`

So, here is two issues:
  - 404 errors should be reported as such, not as failed parsing.
  - Also windows need to use the correct url.

rgleason, 2020-01-04:
What about changing the channels/branches to 
alpha
beta
experimental
master?

The capitals threw me to begin with.

leamas, 2020-01-05:
[keeping log here since it's likely I will have to abort this soon]

It seems to be about case-sensitive urls. On windows the effective url to the experimental bran
 `https://raw.githubusercontent.com/opencpn/plugins/experimental/ocpn-plugins.xml`
This gives a 404 + a bad error message about failed parsing. The correct url is
` https://raw.githubusercontent.com/OpenCPN/plugins/Experimental/ocpn-plugins.xml`

So, here is two issues:
  - 404 errors should be reported as such, not as failed parsing.
  - Also windows need to use the correct url.

leamas, 2020-01-05:
Part of  the windows mess seems to bethat the opencpn.ini file isn't updated the same way as other platforms. 

@rgleason : you could probably walk arund the selecting channel problems on windows by editing opencpn.ini and set CatalogChannel to one of the listed plugins (before starting opencpn).

rgleason, 2020-01-05:
Found in opencpn.ini 
CatalogCustomURL=https://raw.githubusercontent.com/opencpn/plugins/Alpha/ocpn-plugins.xml
CatalogChannel=experimental

Changed to 
CatalogCustomURL=https://raw.githubusercontent.com/opencpn/plugins/Alpha/ocpn-plugins.xml
CatalogChannel=alpha
Now it is showing "alpha" Channel with the alpha URL!
[image]

Changed to 
CatalogCustomURL=https://raw.githubusercontent.com/opencpn/plugins/Alpha/ocpn-plugins.xml
CatalogChannel=beta
Now it is showing "beta" channel with alpha URL
[image]
 
Changed to
CatalogCustomURL=https://raw.githubusercontent.com/opencpn/plugins/master/ocpn-plugins.xml
CatalogChannel=master
Now it is showing master channel with master URL, however the underlying Plugin Manager List still appears to be from the alpha channel or alpha url. because testplugin and weather routing are still in there having been added via meta-url that I created and bdbcat merged.   I do not fully understand how this is supposed to work.  Are the Catalog and URL supposed to be in lock step so  to speak?
[image]

Ok this is after using Advanced Catalog Update selecting  "Use Latest Catalog", check what Ultra Advanced says (still on the master url), Close that without "Saving" it), hit Done,  Hit Apply, then OK.
Now back to the normal Opencpn screen. Then options>plugins, and it appears that Channel=master and URL=https://raw.githubusercontent.com/opencpn/plugins/master/ocpn-plugins.xml
[image]

I think that we need to look at how the ChartDownloader works!  ChartDownloader is quite clear about how you download a new chart catalog and Apply it and does not require closing of many menus and selection of "Done" "Apply" "Close" etc.

leamas, 2020-01-07:
Some notes:

  - This is not the time to re-organize the ui. For now, we need to fix the last bunch of bugs and get it running. 
  - What you should compare to the chartdownloader update should be the "Update catalog" button which is a simple button click followed by a confirmation dialog. It can't be much simpler.
  - The Advanced menu *is* complicated, but the complexity mirrors a flexibility built in from the beginning which maps well into our needs for feeding for example beta testers with a separate catalog.
  - This is the first shot on the ui. Those shots are seldom correct, and there is probably reasons to re-organize the ui (I have some ideas for the Advanced/Catalog Manager window). But not now.

leamas, 2020-01-07:
> Changed to
CatalogCustomURL=https://raw.githubusercontent.com/opencpn/plugins/Alpha/ocpn-plugins.xml
CatalogChannel=alpha
Now it is showing "alpha" Channel with the alpha URL!

However, as long as there is a CatalogCustomURL defined it will override everything, basically disabling the selected channel.

rgleason, 2020-01-07:
1. Not the time to reorganize, ..get it running.  Agreed.
2. Re Advanced menu, I ask, Is it using previous commits and ocpn-plugins.xml to allow stepping backwards to a previous catalog?
3.  First shot... Ok
4. Showing Alpha channel with alpha url  - great!
5.  "as long as there is a CatalogCustomURL defined it will override everything, basically disabling the selected channel."   Can we eliminate the ambiguity by replacing the Channel name with "Custom" when someone has entered a Custom URL? or is that too difficult programmatically?

Thanks Alec. Let me know when/if you want some testing.

bdbcat, 2020-01-07:
Not the time, but very soon now.
Let us not spend a whole lot of time polishing the existing UI, but rather, lets become convinced that the underlying plumbing is secure and stable.  By whatever means necessary now.
I can relate from long experience that perfecting all the fussy dialogs in a UI across all platforms/languages is a tedious and never ending task.  We should not spend the effort if the essential UI structure is to change before release.

And change it shall.  The sooner the better.
Dave

leamas, 2020-01-08:
Major changes under way from @bdbcat, stay tuned.

That said, I'm not aware of any any blocking issues when it comes to the  testing underlying plumbing. So testing of these parts is welcome.

leamas, 2020-03-03:
Time to close?

closed 2020-03-17 by bdbcat