OpenCPN/OpenCPN #1628: macOS - hovering AIS targets, no data shown

bcn2, 2020-01-03:
no data shown when hovering over AIS targets under macOS

bcn2, 2020-01-03:
beta trial of course - the stable release is ok

tai-fun, 2020-01-03:
Is Options->User Interface->Enable Route/AIS Info block activated?

bcn2, 2020-01-03:
Thanks! No, hasn't been - now activated gives us the following picture (w&w/o dark scheme)
[image]

bdbcat, 2020-01-04:
Should be fixed by ca924a650af5faf4217b552e116114d5af56ad9e

bcn2, 2020-01-04:
Confirmed ok now...

Hakansv, 2020-01-04:
I've a suggestion to default set "Enable Route/AIS Info block activated" for non touch screen devices.
Let's see if Dave accept my hack or have a better method.

tai-fun, 2020-01-04:
I agree with Håkan, but we also need a bit more text in "Options" to cross reference the different options. Imagine a user who activates all alternatives in Options->Ships->Ais Targets-Rollover and don't see anything on the screen. The default action could be to ask on the forum! Said user forgot that she un-ticked the info block option a week ago when she played around with O for the first time. 
We already see a similar scenario with showing text on vector charts where about 3 settings are involved, and we have had a number of threads about this on the forum.
Also, the option should be renamed "Enable track/route/AIS Info block ", as track info is involved as well.
Thomas

bcn2, 2020-01-04:
Enabling or disabling "touch screen interface" should trigger a pop-up pointing to this option.
Or directly giving the opportunity to toggle the Rollover visibility right from there.

Will prevent a lot of questions at the forum - as Thomas already stated.

bdbcat, 2020-01-04:
I agree with Thomas that our options are in several cases fragmented across screens.  This needs a scrub.
Meanwhile, Hakan's pragmatic solution looks good for 95%  of the user base.
Thanks
Dave

bcn2, 2020-01-04:
Closed by now...

closed 2020-01-04 by bcn2