OpenCPN/OpenCPN #1633: Win10 Canvas right mouse click -> O Crash

Hakansv, 2020-01-04:
Some commit after "Correct rollover window display on MacOS." creates O to crash on right mouse click.
I'll check for more details.

Hakansv, 2020-01-04:
Well, I didn't come that far on Debug:
First exception:
```
>	opencpn.exe!ChartCanvas::DoCanvasUpdate() Line 1624	C++
 	opencpn.exe!MyFrame::DoChartUpdate() Line 8103	C++
 	opencpn.exe!MyApp::OnInit() Line 2468	C++
 	wxbase312ud_vc_custom.dll!wxAppConsoleBase::CallOnInit() Line 93	C++
 	wxbase312ud_vc_custom.dll!wxEntryReal(int & argc, wchar_t * * argv) Line 491	C++
 	wxbase312ud_vc_custom.dll!wxEntry(int & argc, wchar_t * * argv) Line 184	C++
 	wxmsw312ud_core_vc_custom.dll!wxEntry(HINSTANCE__ * hInstance, HINSTANCE__ * __formal, char * __formal, int nCmdShow) Line 305	C++
 	opencpn.exe!WinMain(HINSTANCE__ * hInstance, HINSTANCE__ * hPrevInstance, char * __formal, int nCmdShow) Line 1090	C++
```

[image]

bdbcat, 2020-01-04:
@Hakansv 
git bisect is the tool for this occasion.

Hakansv, 2020-01-04:
Commit: "Correct IsSameAs() semantics to exactly identify matching plugin metadata" runs OK.
Next commit: " Merge pull request #1620 from laroche/flr-dev      cleanup of compiler warnings" is causing the failure. And there's a lot changed.

laroche, 2020-01-04:
Thanks for the report. Checked the source again and even on Linux the compiler gives a warning on this. Patch Coming...

laroche, 2020-01-04:
Just added a new pull request #1634

Hakansv, 2020-01-04:
Yes that typo you've corrected by 1634 is clear. That one I saw as well.
After that we come to the real issue for right click menu failure. You better test yourself but the debug exception is here:

[image]
```
>	wxmsw312ud_core_vc_custom.dll!wxWindowBase::GetScreenPosition(int * x, int * y) Line 306	C++
 	wxmsw312ud_core_vc_custom.dll!wxWindowBase::GetScreenRect() Line 347	C++
 	opencpn.exe!MyFrame::GetCanvasIndexUnderMouse() Line 3538	C++
 	opencpn.exe!PlugInManager::PrepareAllPluginContextMenus() Line 2534	C++
 	opencpn.exe!CanvasMenuHandler::CanvasPopupMenu(int x, int y, int seltype) Line 859	C++
 	opencpn.exe!ChartCanvas::InvokeCanvasMenu(int x, int y, int seltype) Line 9181	C++
 	opencpn.exe!ChartCanvas::CallPopupMenu(int x, int y) Line 7364	C++
 	opencpn.exe!ChartCanvas::MouseEventProcessObjects(wxMouseEvent & event) Line 8501	C++
 	opencpn.exe!glChartCanvas::MouseEvent(wxMouseEvent & event) Line 731	C++
 	wxbase312ud_vc_custom.dll!wxAppConsoleBase::HandleEvent(wxEvtHandler * handler, void(wxEvtHandler::*)(wxEvent &) func, wxEvent & event) Line 658	C++
 	wxbase312ud_vc_custom.dll!wxAppConsoleBase::CallEventHandler(wxEvtHandler * handler, 

```

Hakansv, 2020-01-04:
For any reason this line is commented out:
```
@@ -3537,7 +3531,7 @@ int MyFrame::GetCanvasIndexUnderMouse()
         case 1:
             cc = g_canvasConfigArray.Item(0);
             if(cc ){
-                ChartCanvas *canvas = cc->canvas;
+                //ChartCanvas *canvas = cc->canvas;
                 if(canvas->GetScreenRect().Contains(/*canvas->ScreenToClient*/(screenPoint)))
                     return 0;
```
If I uncomment and again activate #ChartCanvas *canvas = cc->canvas;"  The right mouse menu is working again.

It may be every waste change like this commit have to be thoroughly tested before commit?

More or less most program functions has been changed. The test program will be extensive. All charts, options, tracks, AIS targets and decoder, connections etc.

laroche, 2020-01-04:
Oop, that change from me was also buggy. The "canvas" var from the function can be
deleted, not this var definition within the code.

My own setup is very small, just started using O the last days. I'll make sure to review/test things
more thoroughly before submitting further change requests.

best regards,

Florian La Roche

Hakansv, 2020-01-04:
Ok Florian and welcome to the team. Some code cleaning is for sure not bad but the program has been working and developed for years and is rather complex so every change need to be tested in "real life" with all components and plugins. As far as possible before beta test.
With the last bug fix I've tested around and nothing more found so far. I will continue.
Håkan

pevol, 2020-01-05:
Hi, 
there seems to be another problem introduced with commit 65c007425c9fa8a21d8d8ee4aaa1fa9ecfb59c40 (cleanup of compiler warnings): compiler error when compiling for ARM ( on raspbian bionic)

Here is the fix:

Subject: [PATCH] fix compiler error when compiling for ARM

---
 src/cm93.cpp | 7 +++++--
 1 file changed, 5 insertions(+), 2 deletions(-)

diff --git a/src/cm93.cpp b/src/cm93.cpp
index f7c2b9e2b..6e09c2315 100644
--- a/src/cm93.cpp
+++ b/src/cm93.cpp
@@ -3410,6 +3410,10 @@ S57Obj *cm93chart::CreateS57Obj ( int cell_index, int iobject, int subcell, Obje
             int ival;
 
             S57attVal *pattValTmp = new S57attVal;
+#ifdef __ARM_ARCH
+            float __attribute__((aligned(16))) tf1;
+            unsigned char *pucf;
+#endif
 
             switch ( vtype )
             {
@@ -3485,8 +3489,7 @@ S57Obj *cm93chart::CreateS57Obj ( int cell_index, int iobject, int subcell, Obje
                       pAVR = ( double * ) malloc ( sizeof ( double ) );   //new double;
                       pf = ( float * ) aval;
 #ifdef __ARM_ARCH
-                        float __attribute__((aligned(16))) tf1;
-                        unsigned char *pucf = (unsigned char *)pf;
+                        pucf = (unsigned char *)pf;
 
                         memcpy(&tf1, pucf, sizeof(float));
                         *pAVR = tf1;

Best regards
Peter

Hakansv, 2020-01-05:
@laroche 
Florian, Would you mind check this and maybe PR a change? I've tried it on my RPi3 where I received an error as well. Using Peter's patch solved it.

laroche, 2020-01-05:
Ok, will check this...

Hakansv, 2020-01-05:
The CM93 patch was probably not a good solution?  CM93 on RPi3 is working. No details or objects.

Hakansv, 2020-01-05:
For commit: "cleanup of compiler warnings"
What was the reason for taking this out? See below.
After that RPi3 don't build. OK after restore. The code from Peter was not a solution.
```
@@ -3435,7 +3408,6 @@ S57Obj *cm93chart::CreateS57Obj ( int cell_index, int iobject, int subcell, Obje
             int nlen;
             double dival;
             int ival;
-            unsigned char *pucf;
 
             S57attVal *pattValTmp = new S57attVal;
 
@@ -3514,7 +3486,7 @@ S57Obj *cm93chart::CreateS57Obj ( int cell_index, int iobject, int subcell, Obje
                       pf = ( float * ) aval;
 #ifdef __ARM_ARCH
                         float __attribute__((aligned(16))) tf1;
-                        pucf = (unsigned char *)pf;
+                        unsigned char *pucf = (unsigned char *)pf;
 
                         memcpy(&tf1, pucf, sizeof(float));
```

laroche, 2020-01-05:
I can currently only test Debian unstable on ARM64, I assume most others use 32bit arm.

I propose to change the code to be the same as all other similar "case"-Implementations in
cm93.cpp and enclose all statements within a starting "{" and a closing "}".

This should fix the compile problems and get people up and running again on arm.

And on the code that is used throughout OpenCPN with __ARM_ARCH:
I assume the problems on arm only affect 32bit and probably only
some compilers. Maybe 64bit is not affected at all? This should be verified in more detail
and checked if more elegant solutions can be found. The alignment to 16-bytes
has also just been added the last month, so I assume Android also needs this.

Patch coming the next minute.

best regards,

Florian La Roche

pevol, 2020-01-05:
This was the compiler error without the patch:

`[ 28%] Building CXX object CMakeFiles/S57ENC.dir/src/cm93.cpp.o
/usr/bin/c++  -DWXUSINGDLL -D_FILE_OFFSET_BITS=64 -D__WXGTK__ -DocpnARM -DocpnUSE_GL -DocpnUSE_SVG -DwxUSE_WEBVIEW=0 -I/home/peter/upstream/OpenCPN/build/include -I/home/peter/upstream/OpenCPN/include -I/home/peter/upstream/OpenCPN/libs/nmea0183/src -I/home/peter/upstream/OpenCPN/libs/libtess2/Include -I/home/peter/upstream/OpenCPN/libs/sound/include -I/home/peter/upstream/OpenCPN/libs/wxJSON/include -I/home/peter/upstream/OpenCPN/libs/ssl_sha1/include -I/home/peter/upstream/OpenCPN/libs/wxsvg/include -I/home/peter/upstream/OpenCPN/libs/wxcurl/include -I/usr/include/arm-linux-gnueabihf -I/home/peter/upstream/OpenCPN/libs/iso8211/include -I/home/peter/upstream/OpenCPN/libs/gdal/include -isystem /usr/lib/arm-linux-gnueabihf/wx/include/gtk2-unicode-3.0 -isystem /usr/include/wx-3.0  -pthread -O2 -g -DNDEBUG    -DPREFIX=\"/usr/local\" -fvisibility=hidden -Wall -Wno-unused -fexceptions -rdynamic -fno-strict-aliasing -Wno-deprecated-declarations -std=gnu++11 -o CMakeFiles/S57ENC.dir/src/cm93.cpp.o -c /home/peter/upstream/OpenCPN/src/cm93.cpp
/home/peter/upstream/OpenCPN/src/cm93.cpp: In member function ‘S57Obj* cm93chart::CreateS57Obj(int, int, int, Object*, cm93_dictionary*, Extended_Geometry*, double, double, double, double)’:
/home/peter/upstream/OpenCPN/src/cm93.cpp:3499:19: error: jump to case label [-fpermissive]
                   default:
                   ^~~~~~~
/home/peter/upstream/OpenCPN/src/cm93.cpp:3489:40: note:   crosses initialization of ‘unsigned char* pucf’
                         unsigned char *pucf = (unsigned char *)pf;
                                        ^~~~
make[2]: *** [CMakeFiles/S57ENC.dir/build.make:66: CMakeFiles/S57ENC.dir/src/cm93.cpp.o] Error 1
make[2]: Leaving directory '/home/peter/upstream/OpenCPN/build'
make[1]: *** [CMakeFiles/Makefile2:272: CMakeFiles/S57ENC.dir/all] Error 2
make[1]: Leaving directory '/home/peter/upstream/OpenCPN/build'
make: *** [Makefile:155: all] Error 2
peter@vasco:~/upstream/OpenCPN/build $ gcc --version
gcc (Raspbian 8.3.0-6+rpi1) 8.3.0
Copyright (C) 2018 Free Software Foundation, Inc.
This is free software; see the source for copying conditions.  There is NO
warranty; not even for MERCHANTABILITY or FITNESS FOR A PARTICULAR PURPOSE.`

This was on a RPi4 + raspbian bionic  with latest updates.

laroche, 2020-01-05:
Pull request 1645 already has the patch, I assume once the automatic checks are complete,
this should also get referenced here.

best regards,

Florian La Roche

Hakansv, 2020-01-05:
@pevol 
And what if you try to  restore the change indicated on my post above.
With your change I had some black squares on some scales using CM93.

pevol, 2020-01-05:
I tried the version from pull request 1645, works fine for me.
@Hakansv , making the changes according to your post above also works. I can see no issues with CM93 but I am also not sure what to look for exactly. BTW, I am using hardware openGL.

Hakansv, 2020-01-06:
RPi issue solved by #1644 A most refined and uncomplicated solution.  Thanks.

closed 2020-01-06 by Hakansv