OpenCPN/OpenCPN #1639: Plugin Manager: Manage Plugin Catalog Question

rgleason, 2020-01-04:
How exactly does this work?

Catalog URL Status: OK
Channel: Master
'-------------------------'
Current Active Status       [version]   [date & time] 
Default Catalog                [version]  [date & time]     Use as active Catalog button
Latest Available Catalog  [version] [date & time]       Use as active Catalog button
'-------------------------'
Ultra Advanced >>>

What is the Default Catalog?  The one that is stored locally on the Machine?
What is the Latest Available Catalog?  The one that has been updated in the cloud?

[image]

rgleason, 2020-01-05:
Selection of a "Channel" is intended to be selection of a  Github  Branch from https://github.com/OpenCPN/plugins/
There is a one to one correspondence
Channel   = URL
master     = https://github.com/OpenCPN/plugins/tree/master
Alpha      = https://github.com/OpenCPN/plugins/tree/Alpha
Beta        = https://github.com/OpenCPN/plugins/tree/Beta
Experimental = https://github.com/OpenCPN/plugins/tree/Experimental

However we are actually looking for the ocpn-plugins.xml catalog for each of these branches so it is actually this correspondence
master = https://raw.githubusercontent.com/OpenCPN/plugins/master/ocpn-plugins.xml
Alpha = https://raw.githubusercontent.com/OpenCPN/plugins/Alpha/ocpn-plugins.xml
Beta= https://raw.githubusercontent.com/OpenCPN/plugins/Beta/ocpn-plugins.xml
Experimental = https://raw.githubusercontent.com/OpenCPN/plugins/Experimental/ocpn-plugins.xml
Custom= User defined url such as 
Custom= https://raw.githubusercontent.com/rgleason/plugins/rg-alpha/ocpn-plugins.xml
Custom=https://raw.githubusercontent.com/rgleason/plugins/Alpha/ocpn-plugins.xml

When the user selects from the dropdown one of these Channels or set Github Branches both the "Channel" and the underlying "URL" must change and  the catalog data fetched from the respective website url,  read into the Plugin Manager List and finally displayed properly.
There is no reason this can't be clearer.
1. Pick the Catalog
2. Pick Update which updates to the most current catalog.
3. Plugin Manager List Displays the respective ocpn-plugins.xml

If it has trouble doing this have the appropriate failure messages:
A. Internet Connection Lost.
B. Incomplete transmission. Try again.
C. Catalog corrupted or not formatted properly and cannot be used.

This does not require a complicated menu system, it could be a drop down Channel selection at the top of the Plugin Manager Listing, with an Update button, which loads the new catalog and displays it below.

Then there there two are advanced functions needed.
1. "Custom" Catalog , treat this the same as the Channels (Branches). This also be n the drop down menu at the top of the Plugin Manager List, and when selected it brings up a URL field where the user can enter the URL for a custom catalog.  Later a programmer can add URL memory for previous URL selection, maybe even allowing change of "Custom" to some user defined name.

This then leads to the need to "rollback" to a previous "Catalog"
I believe this is the same as going back to a previous github commit and Updating.
The reason this is needed is if a new Catalog is corrupted and the user needs to rollback.
It would be nice to be able to step forward and back through commits. 

I believe this functionality is represented by the current "Advanced Catalog/Manage-Plugin-Catalog" menu to a greater or lesser degree. I think the source of this information needs to be clearer, the menus simpler, and the ability to move back and forward through commits needs to be added.

[image]

Why not just call it "Advanced Catalog"
Show the Channel that has been selected previously, for Example "Alpha"  or "Custom"
Then just show the current and last 2 commits with ocpn-plugins.xml (if they exist) and  the commit date
"Select as active Catalog"
(Current  [Date])   (As a selection button)
(Recent   [Date])   (As a selection button)
(Previous  [Date])  (As a selection button)
Selection completes the action (no need to hit "Done"), the menu closes and the Plugin Manager List redisplays the catalog.

[image]

bdbcat, 2020-01-05:
All...
Actually, this brings up a bigger point that I was waiting until the correct point to introduce.
Now that we have a (mostly) functional extended plugin manager system, we need to re-factor the UI for plugin management overall.

1.  There should be exactly one large list for all plugins, new and legacy, installed, or not.
2.  The panel for each plugin (by name) should provide information on the current state of the  plugin, and controls to allow manipulation of it (install, activate, upgrade, downgrade, etc.)
3.  There should be a simple control for the list allowing global selection of the advanced plugin channel in use, always visible.
4.  There should be an "update catalog" button, always visible.
5.  There should not be multi-level popups for advanced functions.  At most a single level of popup dialog.
6.  There should NOT be popups showing incremental status of the various operations.  I understand that this is very useful during development, to monitor the operation of some complex state-machine code.  But this is too distracting for our target user base.

I understand that the "biggie" here is integration of the two lists. Lots of code, and won't be done in a day. The rest will be subject to tweaking as we move along.  But I am certain that we should make this effort to harmonize and simplify the UI now, before we even try to introduce the new plugin concepts in the wild.  After all, we all hope that in a year, there will be NO interesting legacy plugins....
Dave

Hakansv, 2020-01-05:
Dave
The structure is clear and appreciated. Very good!. And in the end I do think needed methods to adapt any plugin will be clear. Also for underdogs like me that are bit lost for moment.

rgleason, 2020-01-05:
Perhaps have a opencpn.ini setting for PluginInstallerDebugMenus=yes during this process?

rgleason, 2020-01-05:
I like Dave's idea. Perhaps we will want to know in the list, which channel an installed plugin came from? Experimental, Alpha, Beta, master, Custom?

Its worth thinking about.

closed 2020-01-06 by rgleason