OpenCPN/OpenCPN #1641: Plugin Manager: Catalog Manager

rgleason, 2020-01-05:
Isn't Catalog Manager just an informational menu that confirms and shows exactly Catalog the Plugin Manager is going to use?  There are no choices or selections here.
[image]

If so, shouldn't   "Update Plugin Catalog" just read "Catalog Manager"?
Also if this is just informational, aren't "Cancel" and "Ok" the same? Can't it just be "Close"?

I think there is something wrong because it says Channel: "Experimental" when actually I entered "Alpha" which is the channel that this URL is intended for.

There might also be some other places where a button for "Catalog Manager" would be helpful, such as after picking  "Advanced Catalog Update" which opens Manage Plugin Catalog?   Wouldn't we find it useful to know what the "Catalog Manager" is using, with a "Catalog Button?

[image]

Also note that this is using Experimental, even when I have selected "Alpha" .  Also when I select "Alpha"  from the dropdown, pick Done and the pick Advanced again, it comes up with "Experimental" again..  This is very frustrating and it is not working.

leamas, 2020-01-07:
The selecting Alpha -> experimenatl should now be fixed.

> Shouldn't "Update Plugin Catalog" just read "Catalog Manager"?

I don't think so. After all, this is not managing the catalog in any other sense then a plain update using the existing catalog channel, Compare this to the "Advanced" button which shows a window which actually is managing the catalog update.

>  aren't "Cancel" and "Ok" the same? 

Not really. Cancel works during the update, OK not. However, on a fast networkl  it iosn't much of a difference. Perhaps one should disable Cancel when the update is complete...

> There might also be some other places where a button for "Catalog Manager" would be helpful, such as after picking "Advanced Catalog Update" which opens Manage Plugin Catalog? Wouldn't we find it useful to know what the "Catalog Manager" is using, with a "Catalog Button?

I cannot parse this, sorry...

rgleason, 2020-01-07:
So "Update Plugin Catalog"  actually rereads via the internet the currently selected catalog? It would be nice if we could visually see which catalog.  While testing this, absolutely nothing happened, no screen rewrite, nothing. So perhaps you can understand that I could not "parse" your intent.

"Perhaps one should disable Cancel when the update is complete..."
I think this might be a good idea. 

re button of Catalog Manager -- you can't parse it, because I could not determine the program intent (IE I couldn't parse it either).  Parsing sounds like a computer.

closed 2020-01-18 by rgleason