OpenCPN/OpenCPN #1642: Plugins: Version

rgleason, 2020-01-05:
Why not show the full Version Number in the Plugin List, it is available and it would help distinguish between newer plugins and older ones.  I thought we had agreed to show them when they are available.

[image]

Also there is no need to have the Vertical Scroll bar in the upper section where the Icons are located.

Hakansv, 2020-01-05:
Rick..
According to my knowledge it's not so.
The Options Plugins window is drawn by O.
O and Pluginmanager don't know about the version patch. The plugin has no way to transfer that info to O before API 117. If you you want the plugin to transfer that you've to use "GetPlugInVersionPatch()" in the plugin and join it to API 117.
I think the new Plugin Manager dialog uses the newly prepared xml document to read info about actual patch version. The new plugin loader would possible be able to create a once time "backdoor" if the plugin is installed that way but to have a consistent update after a O restart I think a plugin must be adapted to API 117.
You may try it in the "testplugin" and confirm if the version patch is printed also in the Options Plugins window.

[image]

rgleason, 2020-01-05:
@Hakansv 
Thanks, sorry I forgot that we are not on api1.17  and that Plugin Manager has this information in the Catalog.  Good points. Thanks.

I am guessing this should be closed now.

closed 2020-01-05 by rgleason