OpenCPN/OpenCPN #1643: Plugin Manager: Alpha Meta-Url  is working, but not normally, and only with URL.

rgleason, 2020-01-05:
The only way I can get the meta-url that I made a PR for and Dave merged, here
https://raw.githubusercontent.com/OpenCPN/plugins/Alpha/ocpn-plugins.xml
is to use the  Ultra  Advanced  URL: https://raw.githubusercontent.com/OpenCPN/plugins/Alpha/ocpn-plugins.xml directly.   Using the Advanced  dropdown selection and picking "Alpha" does not work, it just goes back to "Experimental".

Also once I enter the URL: https://raw.githubusercontent.com/OpenCPN/plugins/Alpha/ocpn-plugins.xml
I have to Save and Close everything and then Shutdown Opencpn, then reopen it to get this to work.  I wouldn't ask even Beta Testers to spend their time on this at this point. The only thing that really works is the URL!

Anyway, at least we know the meta-url record does work. I guess that is some progress.

[image]

bdbcat, 2020-01-05:
Rick...
We are nowhere near Beta testing of the advanced Plugin Manager yet.  There are lots of UI changes coming.  Patience, please, while the hidden plumbing is refined and validated.

rgleason, 2020-01-06:
Ok I'll close this, but it isn't really workiing.

closed 2020-01-06 by rgleason