OpenCPN/OpenCPN #1652: Chart downloader on Raspberry Pi Buster is broken

transmitterdan, 2020-01-06:
On Raspberry Pi 4 Buster, chart downloader cannot change preferences.  Preferences dialog opens but no controls are active and it cannot be closed. Even closing plugin manager window leaves preferences dialog of chart donwloader on screen but all controls are disabled.  It is as if it is a modal child of some invisible window.

Also, the preferences option on the chart downloader panel in options is no longer present so it's not possible to reach the preferences via that route any more.

In previous version of chart downloader the parent window of the dialog was different than it is now. I don't know if that is the reason for this issue or not.  Other plugins seem not to be affected like this.

On Windows this issue doesn't happen so it could be a wx version issue too.

bdbcat, 2020-01-07:
@transmitterdan 
I got this.  Please stand by

bdbcat, 2020-01-07:
Probably fixed in e23b0774502235602640ef860abeb886bcec0e22
Not tested on all platforms yet.

Hakansv, 2020-01-07:
On RPi3 - stretch.
Chart downloader is fine what I can detect but a strange effect on Route properties only seen on RPi, not Windows: An actual local time text box is shown. If it's connected to this issue I don't know?
[image]

bdbcat, 2020-01-07:
Fixed by f7e649df30f890ffa43b5d74ada637650e9f52d8
All due to "bigmerge" commit earlier.

Hakansv, 2020-01-07:
Very good! The time box in route manager is now gone on RPi3. For me this can be closed.
There are still (connected?) issues in dialog windows like mentioned in 1534. WP properties is not usable since no OK button is viewed even with font size = 7. And no slider.

closed 2020-01-08 by transmitterdan