OpenCPN/OpenCPN #1658: all core plugins not compatible

seandarcy, 2020-01-11:
I'm building today's git on linux, Fedora 31.

But none of the 4 core plugins are compatible or enabled:

16:08:50 EST: Checking plugin candidate: /usr/lib64/opencpn/libdashboard_pi.so
16:08:50 EST: Skipping not enabled candidate.
16:08:50 EST: Checking plugin candidate: /usr/lib64/opencpn/libwmm_pi.so
16:08:50 EST: PLugin is compatible: false
16:08:50 EST: Checking plugin candidate: /usr/lib64/opencpn/libchartdldr_pi.so
16:08:50 EST: PLugin is compatible: false
16:08:50 EST: Checking plugin candidate: /usr/lib64/opencpn/libgrib_pi.so
16:08:51 EST: PLugin is compatible: false

Tools->Options->Plugins shows no plugins.I would have expected the dashboard plugin to show, with an "Enable" button.

I built vfkaps 2.1.1 from source, same machine, also not compatible:

16:08:49 EST: Checking plugin candidate: /usr/lib64/opencpn/libvfkaps_pi.so
16:08:50 EST: PLugin is compatible: false

Any help appreciated.

transmitterdan, 2020-01-12:
What was your cmake command?  It could be the plugins installed in /usr/lib64 were built with a different version of wxWidgets.  Normally when you locally build O everything installs in /usr/local/..... not /usr....

So what I am saying is that you probably have an older installed version and then the newer locally built version and there is a conflict.  I build Linux builds all the time and don't have the issue you have.

bdbcat, 2021-04-12:
Cannot reproduce with current code base

closed 2021-04-12 by bdbcat