OpenCPN/OpenCPN #1665: Plugin Manager UI redesign

bdbcat, 2020-01-18:
Hello all...
With Alec's help, github master now contains a redesign of the new PluginManager GUI.
We were looking for a simplified and more intuitive interface for our generic entry-level user.  With this redesign, we are close, but some questions still remain.
Please have a look, and bring comments/questions here.

Thanks for your patience.
Dave

bdbcat, 2020-01-18:
For reference, and a path forward, we would like to avoid ever having to answer this tech support question:
"What does this <fill in the blank> button do?"
Dave

tai-fun, 2020-01-18:
> "What does this button do?"

Dave
The new interface quickly becomes unusable. Whenever I do something to a
plugin, such as just checking different alternatives, a new instance of the
"Plugin Catalog"  sub dialog is added, until you can't even see any plugins
at all.

Thomas

tai-fun, 2020-01-18:
[image]

rgleason, 2020-01-18:
Dave, This is much better... I feel Iike I know what to expect generally, and it is much simpler for comprehension.

1. The Catalog interface at the bottom seems to be working as expected for Master, Alpha and Beta channels, and the Selected channel, also shows what catalog is active and being used. This works I think. 

2. Additionally when I select the Alpha channels the Catalog changes and shows the potential of installing testplugin or weather_routing.  These both install and can be used, but weather_routing itself has a problem.  When it is installed it seems to be split into  a "Managed plugin" and an "Unmanaged plugin". The unmanaged installs, and the managed on crashes Opencpn. That is a problem with the Weather_routing compilation.  Testplugin is better behaved. 

3. I have messed with the Custom  URL https://raw.githubusercontent.com/rgleason/plugins/Alpha/ocpn-plugins.xml   putting this url in, the selected ocpn-plugins.xml  catalog is showing and the dropdown shows "Custom" as it should.

This is all much much better.  You and Alec have improved the situation considerably.

4. The next level to consider is how the individual plugins install and uninstall under different conditions. 

- When there is an unmanaged and managed plugin that is the same name, 
   - First the unmanaged plugin should be "disabled" and "removed manually" using the uninstall exe or by removing the dll or executable.  Currently the "unmanaged" version has an active "Uninstall" button which when selected closes Opencpn. I do not know if there is a search going on to find and execute the uninstall.exe, because my Opencpn installation was today and there were no uninstall.exe available for plugiins, but  shouldn't that active button that says "Uninstall" have an unactive "Manual Uninstall Required." message, if there is a managed plugin available (assuming we want users to migrate)?
     - Second when the same name"managed" and "unmanaged" plugins show in the list, the "managed" when "enable" is tried, crashes OpenCPN. I haven't figured out why this should happen, but perhaps the file exists and cannot be overwritten or removed?  Anyway, when this condition exists, rather than crashing shouldn't OpenCPN issue an explanatory message?  "Please remove the "unmanaged plugin by using manual methods similar package installations in the past, before installing this "managed" plugin." 

There are some other comments, but this is enough for now.

bdbcat, 2020-01-18:
Taifun....

Strange effect with multiple catalog managers.  Never seen here in simple testing.
Can you describe exactly the simplest way to reproduce this effect?
Dave

bdbcat, 2020-01-18:
Rick...
As I mentioned, we have simplified the user mental model.
1. We have decided and declare that the PluginManager will not ever attempt to "uninstall" an unmanaged (i.e. "Legacy") plugin.  There is no way to do this safely from user space on any platform.  In particular, on Windows, we will not search for an un-install application, ever.
2.  We will (soon) detect and notify the user (gently) that an unmanaged Plugin exists in a "shadowed" state, and may safely be removed. However, having a shadowed, but inactive, plugin exist on the file system does no real harm.  Removal is only necessary for those who worry about very small increments of disk space. Let us not get users stirred up about this, as it will become a hugely distracting red herring.  Note the languge: "may be safely removed" vs "must be removed".
3.  A "shadowed" unmanaged Plugin will not ever be enabled if there is a managed plugin by the SAME EXACT NAME available in the search path. 
4.  If an unmanaged Plugin shows an "uninstall" button, this is an error.  Please describe how to reproduce this effect, if possible.
5.  There should NEVER be two plugins with the exact same name in the plugin list.  If this happens, this is an error.  I understand that it seems to happen today on "weather_routing".  I will investigate.
Dave

rgleason, 2020-01-19:
> 1. We have decided and declare that the PluginManager will not ever attempt to "uninstall" an unmanaged (i.e. "Legacy") plugin. There is no way to do this safely from user space on any platform. In particular, on Windows, we will not search for an un-install application, ever.

Yes, that is the whole point.  There should not be an active button that says "Uninstall" which shuts down Opencpn.  There should be a message that says "Manual uninstallation is required"

2. Good. That is also good to know that it can coexist.  But why won't a "managed" plugin install if there exists and "unmanaged" plugin of the same name?  ---Or is this my imagination?  Seemed to be the situation more than several times.

3.  Good.  What about watchdog-[version]_pi for example?  If the version number is different what happens?  What about if the "unmanaged" legacy plugin is already enabled. Does it stay enabled? 

4. If an unmanaged Legacy plugin shows an "uninstall" button... Ok. Yes, this is working as you say. have some screenshots.   However, I think when I started testing there was some anomaly where I would hit an "Uninstall" button and OpenCPN would just close. I thought that was happening with "unmanaged" plugins, but I could have been wrong, and it was "managed" pi.  --Will check further. 

5. Jon and I have been working on configuring wxrouting for Plugin Manager and there are some things that are not quite right I believe.  I understand your point about never having plugins with the same NAME  (name + version).  Thanks.

Four Legacy "unmanaged" pi

[image]
[image]
[image]
[image]

Wx-Route 

Before install
[image]

Install
[image]

After try install - 2 instances
[image]

After deleted "unmanaged" instance
[image]

Its a bit weird, like we are halfway converted to Plugin Manager.

rgleason, 2020-01-19:
Can a legacy "unmanaged" pi with an earlier version number coexist with a "managed" pi with a more recent version?  Can either one be enabled as long as the other one has been disabled?
--Need to try this.

bdbcat, 2020-01-19:
Rick...
2."But why won't a "managed" plugin install if there exists and "unmanaged" plugin of the same name? ---Or is this my imagination? Seemed to be the situation more than several times."
Please show exact case.

Also, please shorten your posts.  Impossible to respond usefully to run-ons.
Thanks
Dave

bdbcat, 2020-01-19:
3. "Good. What about watchdog-[version]_pi for example? If the version number is different what happens? What about if the "unmanaged" legacy plugin is already enabled. Does it stay enabled?"

Version number shown on theUI does not matter, except controls the exact text in the button for manged plugins.
Upgrading to a managed plugin (from Legacy) should retain the enabled/disabled state.
Dave

rgleason, 2020-01-19:
OK here is an "unmanaged" with an "Uninstall"  that dumps the user out of Opencpn.
[image]
[image]

bdbcat, 2020-01-19:
Rick:
"Can a legacy "unmanaged" pi with an earlier version number coexist with a "managed" pi with a more recent version? Can either one be enabled as long as the other one has been disabled?"

Yes, this is possible.  If you install a managed plugin over a Legacy plugin, and then later un-install the managed plugin, the Legacy plugin becomes available again.  This is one good reason not to remove "shadowed" Legacy plugins, at least until one is certain that the managed version is good to go.

bdbcat, 2020-01-19:
re:
"OK here is an "unmanaged" with an "Uninstall" that dumps the user out of Opencpn."
And how did you produce that scenario, step by step?
Dave

rgleason, 2020-01-19:
1. Exit Opencpn and install oesenc2.9
2. start O options>pi
3. click on oesenc2.9 and this shows
[image]
Then just Uninstall.

Take the pdf off please.

[oesenc_pi-2.9.1601-ov50-win32.exe.pdf](https://github.com/OpenCPN/OpenCPN/files/4082116/oesenc_pi-2.9.1601-ov50-win32.exe.pdf)

bdbcat, 2020-01-19:
Rick...
The prblem producing two "weather routing" list entries is that the names in the metadat are not EXACTLY the same.  "WeatherRouting" != "weather_routing".  Since the plugin name is the primary search field, we enforce EXACT name match.  I understand that anything can happen on Alpha channel, but it is an error in the redirected metadata.  I will accept a PR fixing this...
Dave

bdbcat, 2020-01-19:
Rick...
Legacy oesenc_pi uninstall button:
Not happening for me.  May be a stray copy of oesenc_pi floating around as result of earlier testing. Will you search your system for "oesenc_pi.dll", and report results?
Dave

rgleason, 2020-01-19:
Thank you Dave.  I will make a PR to you for Alpha and keep an eye on that. I think I entered an pluggin name with caps in the cmake file.

Another odd thing. A "managed" plugin, Radar, changes to "legacy" depending on the catalog chosen. I think I kind of understand this, because a given catalog may not have current or any installation xml for that plugin?  Anyway it acts differently depending on what catalog is applied.

Start with Alpha channel  a number of screenshots

1. Radar50beta2-unmanaged-master
[image]

2. Radar50beta2-unmanaged-master-uninstall-it-will-work
[image]

3. Radar 5 0-after-uninstall
[image]

4-Radar5 0beta2-after change to Alpha-managed]
[image]

5-Radar50beta2-now-upgrade-managed]
[image]

6-Radar50beta2-installed
[image]

7-Radar50beta2-after-install
[image]

8-Radar50beta2-back-to-Master
[image]

9-Radar50beta2-now-unmanaged]
[image]

10-Radar50beta2-unmanaged-shows-uninstall]
[image]

11-Radar50beta2-managed-after-catalog-Alpha-update
[image]

12-Radar50beta2-can-be-reinstalled-Alpha
[image]

bdbcat, 2020-01-19:
Rick...
True, the plugins in one catalog may not be available in others.  This will generally be of concern only to Alpha testers and plugin devs.  In real life, we will be sure that the Beta and master catalogs are sensible, for the wider population.  And of course, with re-directed XML catalog entries, the onus is on the author of said XML to have it "make sense".
In general, I think we can live with some funny corners in the Alpha channel, especially with redirection.  This is for Alpha testing only by experienced testers under guidance by the author, not for casual users.  Really, the goal of Alpha plugin testing is to get the correct plugin installed and test it, and not focus too much on the PluginManager UI.

bdbcat, 2020-01-19:
Rick:
re:
"Thank you Dave. I will make a PR to you for Alpha and keep an eye on that. I think I entered an pluggin name with caps in the cmake file."

Actually, you do not need to make a PR.  Just correct your redirect XML file. This is the point of the redirect strategy, so that you can quickly edit changes without waiting for a PR cycle.
Dave

rgleason, 2020-01-19:
> Legacy oesenc_pi uninstall button:
> Not happening for me. May be a stray copy of oesenc_pi floating around as result of earlier testing. Will you search your system for "oesenc_pi.dll", and report results?

Found in C:\Program Files (x86)\OpenCPN 5.0.522\plugins
[oesenc_pi.dll.pdf](https://github.com/OpenCPN/OpenCPN/files/4082128/oesenc_pi.dll.pdf)

no oesenc_pi.dll found in  C:\Users\Rick\AppData\Local\opencpn\plugins  or %appdata%\Local\opencpn\plugins

I said earlier, I installed that file ino Opencpn intentionally.
 
[image]

rgleason, 2020-01-19:
```
Rick:
"Can a legacy "unmanaged" pi with an earlier version number coexist with a "managed" pi with a more recent version? Can either one be enabled as long as the other one has been disabled?"

Yes, this is possible. If you install a managed plugin over a Legacy plugin, and then later un-install the managed plugin, the Legacy plugin becomes available again. This is one good reason not to remove "shadowed" Legacy plugins, at least until one is certain that the managed version is good to go. Note that the only way to recover the Legacy plugin is to un-install the managed version.
```

This is a really nice feature, just saw it with radar_pi I think,.

bdbcat, 2020-01-19:
Unintended Legacy oesenc_pi uninstall button activated:
I see a path where this could happen.  I just committed a fix into master.  Building now...

rgleason, 2020-01-19:
I can't see where there is a problem. Should <name> weather_routing_pi </name> ?

 ```
    <name> weather_routing </name>
    <version> 1.13.2 </version>

    <source> https://github.com/rgleason/weather_routing_pi </source>

    <target>msvc</target>
    <target-version>10.0.14393</target-version>
    <tarball-url>
https://dl.cloudsmith.io/public/rick-gleason/opencpn-plugins-beta/raw/names/weather_routing-msvc-10.0.14393-tarball/versions/1.13.2.+80.c284e85/weather_routing_pi-1.13.2.0-ov50-1.16_msvc-10.0.14393.tar.gz
</tarball-url>
    <info-url> https://opencpn.org/OpenCPN/plugins/weather_routing.html </info-url>

```

rgleason, 2020-01-19:
I'm just going to try adding the _pi, see what happens. Pr coming.

bdbcat, 2020-01-19:
Rick....

The name in you redirected XML should be WeatherRouting, which is the common name of the  plugin being installed.
So, in general, the XML tag <name> should match the common name of the target plugin.

You do not need a PR to Alpha branch.  Just edit your private file:
"https://raw.githubusercontent.com/rgleason/plugins/rg-alpha/ocpn-plugins.xml"

Dave

Dave

rgleason, 2020-01-19:
Ok, thanks. I think I screwed up the last PR too. It does not make sense to me. Also Its late for me and I don't want to mess up again.  Manana.. Nite.

rgleason, 2020-01-19:
The changes are in my branch rg-alpha  which is just the ocpn_plugins.xml for the plugins compiled on my github.  There is a meta-url  xml reference in the Alpha ocpn-plugins.xml direct to that public repository, so Alpha channel picks up changes I make. Pretty nice convenient arrangement. Thanks Dave.
https://raw.githubusercontent.com/rgleason/plugins/rg-alpha/ocpn-plugins.xml

It looks like that fixed the problem with WeatherRouting.
[image]
[image]

Now I have to sort out why the icons are not working.  I did put some in the directory here for both Testplugiin and WeatherRouting
https://github.com/rgleason/plugins/tree/rg-alpha/icons

Somehow they are not getting picked up.

tai-fun, 2020-01-19:
Dave
I have now tested this on Windows 10 and Ubuntu 18.04 without any problems.
The problem is on Ubuntu 19.10.
Simplest case. Go to the plugin tab. Close it with "OK". Go back to the
plugin tab and there is a new catalog manager at the top of the dialog
window. Repeating this, there is added one new catalog manager each time.
They will disappear on restart of O.
No difference if OpenGL is on/off.
No problems running O5 on 19.10.
?
Thomas

bcn2, 2020-01-19:
On macOS:
start with on manual pi installed.
Catalog lists several plug-ins (built-in, managed(1), several non-managed)
Select oeSENC --> "enable" --> Crash
 
`Process:               OpenCPN [10795]
Path:                  /Applications/OpenCPN.app/Contents/MacOS/OpenCPN
Identifier:            org.opencpn
Version:               5.0.522+a237402 (???)
Code Type:             X86-64 (Native)
Parent Process:        ??? [1]
Responsible:           OpenCPN [10795]
User ID:               501

Date/Time:             2020-01-19 11:09:03.961 +0100
OS Version:            Mac OS X 10.15.2 (19C57)
Report Version:        12
Anonymous UUID:        3B4E7B11-B4AD-07F5-C446-A58AF64462AC

Sleep/Wake UUID:       FC399E53-D9C5-4E34-8DA4-3320A5F84565

Time Awake Since Boot: 590000 seconds
Time Since Wake:       11000 seconds

System Integrity Protection: enabled

Crashed Thread:        0  Dispatch queue: com.apple.main-thread

Exception Type:        EXC_BAD_ACCESS (SIGSEGV)
Exception Codes:       KERN_INVALID_ADDRESS at 0x0000000000000000
Exception Note:        EXC_CORPSE_NOTIFY

Termination Signal:    Segmentation fault: 11
Termination Reason:    Namespace SIGNAL, Code 0xb
Terminating Process:   exc handler [10795]

VM Regions Near 0:
--> 
    __TEXT                 0000000105c76000-00000001064e9000 [ 8652K] r-x/rwx SM=COW  /Applications/OpenCPN.app/Contents/MacOS/OpenCPN

Application Specific Information:
Performing @selector(controlAction78:) from sender wxNSRadioButton 0x7ff74e78a560

Thread 0 Crashed:: Dispatch queue: com.apple.main-thread
0   org.opencpn                   	0x0000000105f950d4 PlugInManager::UpdatePlugIns() + 532
1   org.opencpn                   	0x0000000105fabade PluginPanel::SetEnabled(bool) + 62
2   libwx_baseu-3.1.2.0.0.dylib   	0x00000001072fc632 wxEvtHandler::SearchDynamicEventTable(wxEvent&) + 210
3   libwx_baseu-3.1.2.0.0.dylib   	0x00000001072fc3ab wxEvtHandler::ProcessEventLocally(wxEvent&) + 59
4   libwx_baseu-3.1.2.0.0.dylib   	0x00000001072fc2b0 wxEvtHandler::ProcessEvent(wxEvent&) + 96
5   libwx_baseu-3.1.2.0.0.dylib   	0x00000001072fc74c wxEvtHandler::SafelyProcessEvent(wxEvent&) + 12
6   libwx_osx_cocoau_core-3.1.2.0.0.dylib	0x0000000106bd23d7 wxRadioButton::OSXHandleClicked(double) + 279
7   libwx_osx_cocoau_core-3.1.2.0.0.dylib	0x0000000106c64dea wxWidgetCocoaImpl::controlAction(NSView*, void*, void*) + 42
8   com.apple.AppKit              	0x00007fff2e340018 -[NSApplication(NSResponder) sendAction:to:from:] + 299
9   com.apple.AppKit              	0x00007fff2e33feb4 -[NSControl sendAction:to:] + 86
10  com.apple.AppKit              	0x00007fff2e33fde6 __26-[NSCell _sendActionFrom:]_block_invoke + 136
11  com.apple.AppKit              	0x00007fff2e33fce8 -[NSCell _sendActionFrom:] + 171
12  com.apple.AppKit              	0x00007fff2e33fc2f -[NSButtonCell _sendActionFrom:] + 96
13  com.apple.AppKit              	0x00007fff2e33becd NSControlTrackMouse + 1745
14  com.apple.AppKit              	0x00007fff2e33b7d4 -[NSCell trackMouse:inRect:ofView:untilMouseUp:] + 130
15  com.apple.AppKit              	0x00007fff2e33b694 -[NSButtonCell trackMouse:inRect:ofView:untilMouseUp:] + 691
16  com.apple.AppKit              	0x00007fff2e33aa13 -[NSControl mouseDown:] + 748
17  libwx_osx_cocoau_core-3.1.2.0.0.dylib	0x0000000106c62e0e wxWidgetCocoaImpl::mouseEvent(NSEvent*, NSView*, void*) + 254
18  com.apple.AppKit              	0x00007fff2e338ded -[NSWindow(NSEventRouting) _handleMouseDownEvent:isDelayedEvent:] + 4907
19  com.apple.AppKit              	0x00007fff2e2a2f3c -[NSWindow(NSEventRouting) _reallySendEvent:isDelayedEvent:] + 2612
20  com.apple.AppKit              	0x00007fff2e2a22e5 -[NSWindow(NSEventRouting) sendEvent:] + 349
21  libwx_osx_cocoau_core-3.1.2.0.0.dylib	0x0000000106c4f9bb -[wxNSPanel sendEvent:] + 123
22  com.apple.AppKit              	0x00007fff2e2a065c -[NSApplication(NSEvent) sendEvent:] + 352
23  libwx_osx_cocoau_core-3.1.2.0.0.dylib	0x0000000106ba0822 -[wxNSApplication sendEvent:] + 98
24  com.apple.AppKit              	0x00007fff2e3fe88f -[NSApplication _doModalLoop:peek:] + 388
25  com.apple.AppKit              	0x00007fff2e58eb5a __33-[NSApplication runModalSession:]_block_invoke_2 + 69
[... 20 lines cut ...]
46  com.apple.CoreFoundation      	0x00007fff30f44b89 __CFRunLoopDoObservers + 457
47  com.apple.CoreFoundation      	0x00007fff30f4427a __CFRunLoopRun + 1229
48  com.apple.CoreFoundation      	0x00007fff30f43738 CFRunLoopRunSpecific + 503
49  com.apple.HIToolbox           	0x00007fff2fa8a65d RunCurrentEventLoopInMode + 292
50  com.apple.HIToolbox           	0x00007fff2fa8a2a9 ReceiveNextEventCommon + 356
51  com.apple.HIToolbox           	0x00007fff2fa8a127 _BlockUntilNextEventMatchingListInModeWithFilter + 64
52  com.apple.AppKit              	0x00007fff2e0fbeb4 _DPSNextEvent + 990
53  com.apple.AppKit              	0x00007fff2e0fa690 -[NSApplication(NSEvent) _nextEventMatchingEventMask:untilDate:inMode:dequeue:] + 1352
54  com.apple.AppKit              	0x00007fff2e0ec3ae -[NSApplication run] + 658
55  libwx_osx_cocoau_core-3.1.2.0.0.dylib	0x0000000106c4569e wxGUIEventLoop::OSXDoRun() + 174
56  libwx_baseu-3.1.2.0.0.dylib   	0x00000001072dde51 wxCFEventLoop::DoRun() + 49
57  libwx_baseu-3.1.2.0.0.dylib   	0x0000000107242f75 wxEventLoopBase::Run() + 85
58  libwx_baseu-3.1.2.0.0.dylib   	0x000000010721ba74 wxAppConsoleBase::MainLoop() + 116
59  libwx_osx_cocoau_core-3.1.2.0.0.dylib	0x0000000106be470a wxApp::OnRun() + 26
60  libwx_baseu-3.1.2.0.0.dylib   	0x0000000107271bf8 wxEntry(int&, wchar_t**) + 56
61  org.opencpn                   	0x0000000105d37764 main + 20
62  libdyld.dylib                 	0x00007fff686517fd start + 1

`

- On restart only the managed Squidio and the built in PIs get listed.
- Update catalog "beta" adds the "Radar-PI" and
- Update catalog "alpha" adds "test-plug-in" and "Weather-routing"
Default icon, all three labeled "mangaged"
- After Update Catalogue "Master" the labels change to "Unmanaged" in the three cases.

transmitterdan, 2020-01-19:
On Windows 10 I converted old Radar plugin to the new beta plugin using the new plugin manager. O crashes in:
```
void PluginListPanel::Clear()
{
    for (auto it = GetChildren().GetFirst(); it; it = it->GetNext()) {
        if (dynamic_cast<PluginPanel*>(it->GetData())) {
            it->GetData()->Destroy();
        }
        else if (dynamic_cast<wxStaticLine*>(it->GetData())) {
            it->GetData()->Destroy();
        }
    }
```
the iterator (it) points to an already deleted object so this statement: 
```
if (dynamic_cast<PluginPanel*>(it->GetData()))
```
 crashes on deref of already deleted pointer.

rgleason, 2020-01-19:
Hubert re:

> On restart only the managed Squidio and the built in PIs get listed.
>     Update catalog "beta" adds the "Radar-PI" and
>     Update catalog "alpha" adds "test-plug-in" and "Weather-routing" [...]

True but see bdbcat  https://github.com/OpenCPN/OpenCPN/issues/1665#issuecomment-575955959

bdbcat, 2020-01-19:
I say again:
Alpha channel is for experienced testers and plugin devs.  Expect some funny corner case displays on the plugin list when changing channels, especially when some entries contain redirected XML sources.  All these inconsistencies get resolved on restart of O.
In Production, master and Beta will mostly have the same contents, just some Version number changes most likely.

We may want to consider putting a manual config file entry (PluginExpert=0/1) to enable access to the Alpha channel.  This will save inexperienced users some grief...

I will take another look at adjusting the messages, if sensible.

Finally, crashes on enabling a particular plugin are most likely related to the plugin itself, and not the management function.
Dave

Hakansv, 2020-01-19:
Dave..
A made some specific issues as own items. Most about UI.
Functional I've tested to handle install/uninstall of the Squiddio plugin on Win10 and it works as expected. Very handy.
I couldn't test on the RPi3, Stretch, though. No plugins available, the list is empty. I've a RPi4 on Buster but Openplotter installed. I may get me another SD card to build there as well?

leamas, 2020-01-23:
hm... One of the main reasons to build .deb plugins is Rpi and friends. After all, regular Debian desktops can use flatpak as well.  So that we don't build debian plugins is IMHO a major problem. 

And why are we not building pure debian opencpn?

leamas, 2020-01-23:
OTOH: we need to focus right now. Let's leave the Debian situation aside for now and focus on windows, macos, flatpak and Ubuntu. It's more than enough...

bdbcat, 2020-01-24:
Thomas:
re:The problem is on Ubuntu 19.10.
Simplest case. Go to the plugin tab. Close it with "OK". Go back to the
plugin tab and there is a new catalog manager at the top of the dialog
window. Repeating this, there is added one new catalog manager each time.

Cannot reproduce on 19.10 (Virtual) with current code built locally.
Any other tips?

tai-fun, 2020-01-25:
Dave
Thanks for testing on 19.10.
No other tips  for now.
Let's just wait and see if anyone else will report the same problem.
It's not the first time I see a "unique"  issue.
Thomas

rgleason, 2020-01-25:
Using newest version of Opencpn with Signal K Dashboard.  I still seem to get two instances of weather_routing when
1. Custom Catalog  https://raw.githubusercontent.com/rgleason/plugins/master/ocpn-plugins.xml  and update Custom.
2. Reinstall Weather_Routing  1.13.1
3. Two Instances show. 
4. Now go back to Alpha Channel
5. Update Alpha
6. Now there is just one instance of Weather_Routing_pi

Is this something wrong with the plugin's configuration or something amiss with Plugin Manager?
As "expert users" should we just be aware of this issue?

Custom Catalog
[image]

First Instance
[image]

Second Instance
[image]

Now go back to Alpha Catalog and Update Alpha Catalog
[image]

Just one instance of Weather_routing_pi is shown
[image]

rgleason, 2020-01-25:
Multiple Toolbar Icons for testplugin and weather_routing_pi

[image]
[image]

bdbcat, 2020-01-25:
Rick....
There is a problem with the custom .XML
As mentioned before, plugins are identified by their common name. So, **WeatherRouting** is not the same as **weather_routing**.  These are effectively two different plugins, and so are both being shown in the interface.
Dave

rgleason, 2020-01-25:
Thank you! I'll dig into it.  
Any ideas on the multiple icons?  Is this the same as has been reported by Thomas in rpi?

bdbcat, 2020-01-25:
Probably same problem as the plugin name confusion.  Fix that first, and then retest.
At this point I think we can declare that if the UI does not seems correct, the problem is in the XML.  And with custon URL, anything can happen.  That should be your first point of investigation.

leamas, 2020-01-25:
hm... The new design basically display all plugins, installed or not, in a single list with various status flags (text for now, to be replaced/augmented with other elements.

I think there is a need to separate two basica classes of plugins: those installed and the others.

When I (most users?) open the plugins tab I expect the installed plugins. Those are my most  important focus with regular maintenance commands like enable/disable, update, configure etc. Having to to cope with the complete list of plugins which can be installed when doing these tasks is problematic.

OTOH, when I want to install a new plugin I want to see plugins available and not already installed. These plugins only have some info + a single Install action available.

I think we need to separate the regular plugin maintenance and installation of new plugins somehow. Besides simplifying both actions it would also be similar to how other applications work.

Thoughts?

bdbcat, 2020-01-25:
Well, its a thought.
I prefer to see all possible plugins, installed or not, in one common UI.
We could consider the idea of filtering the list to "installed only", by using some modal control button.  However, this is almost certain to create tech support events like this:
Q: "Where is plugin XYZ_pi?"
A: "Enable the "AllAvailablePlugins" button, and refresh the catalog, please.  Then you will see XYZ..."

At the very least, we can sort the list to show installed plugins at the top of the list, followed by (alpha?) sorted available plugins.

More?
Dave

rgleason, 2020-01-26:
Ok, please explain what is wrong with this.  ...Nothing.

 **<plugin version="1">
    <name> WeatherRouting </name>
    <version> 1.13.2 </version>
    <release> 0 </release>
    <summary> Plugin to complete optimal routing with weather files </summary>
    <api-version> 1.16 </api-version>
    <open-source> yes </open-source>
    <author> Sean d'Epagnier </author>
    <source> https://github.com/rgleason/weather_routing_pi </source>
    <description>
Weather_Routing Plugin optimizes weather routing
</description>
    <target>msvc</target>
    <target-version>10.0.14393</target-version>
    <tarball-url>
https://dl.cloudsmith.io/public/rick-gleason/opencpn-plugins-beta/raw/names/weather_routing-msvc-10.0.14393-tarball/versions/1.13.2.+80.c284e85/weather_routing_pi-1.13.2.0-ov50-1.16_msvc-10.0.14393.tar.gz
</tarball-url>
    <info-url> https://opencpn.org/OpenCPN/plugins/weather_routing.html </info-url>
  </plugin>**

And by the way, who sets the standard for the naming of plugins?  Where is that?
I had "weather_routing"  without caps and expected that to work properly. Where is this "filter" located and why is it there?

rgleason, 2020-01-26:
Also the Custom URL from my branch rg-alpha 
https://github.com/rgleason/plugins/blob/rg-alpha/ocpn-plugins.xml

appears to be the same as what shows as a result of the meta-URL that is in the Alpha Catalog.
https://github.com/OpenCPN/plugins/blob/Alpha/ocpn-plugins.xml
```
<plugin version="1">
    <name> Sean Depagnier Alpha Plugins meta-url </name>
    <version> 1.0 </version>
    <summary> meta-url to Sean's Alpha Plugins by rgleason </summary>
    <meta-url> https://raw.githubusercontent.com/rgleason/plugins/rg-alpha/ocpn-plugins.xml </meta-url>
  </plugin>
```

bdbcat, 2020-01-26:
Rick....
I see nothing wrong with that XML either.
What problem are you observing, in simple sentences please.

Naming:
I set the standard.  It is as it always has been.  Plugins are, and always have been, identified by common name.  If you use weather_routing as a plugin name, this is not going to be the same plugin as WeatherRouting.  So update management should not be expected to work. Apples and Oranges are different.

rgleason, 2020-01-26:
> There is a problem with the custom .XML
> As mentioned before, plugins are identified by their common name. So, WeatherRouting is not the same as weather_routing. These are effectively two different plugins, and so are both being shown in the interface. Probably they will collide at runtime.

Who defines the plugin's  "Common Name"?

bdbcat, 2020-01-26:
The plugin author.  It is in the source code.

rgleason, 2020-01-26:
Ok, if we look at cmakelists.txt
https://github.com/rgleason/weather_routing_pi/blob/master/CMakeLists.txt
There are lots of names  Package Name, Verbose Name, Title Name,  also Set (Package "weather-routing")
and also a second Verbose name which redefines it!

```
set(PACKAGE_NAME weather_routing_pi)
set(VERBOSE_NAME weather_routing)
set(TITLE_NAME weather_routing)

```

```
set(PARENT "opencpn")
set(PACKAGE "weather_routing")
set(VERBOSE_NAME "Weather_Routing")
#set(GIT_USER "seandepagnier")
set(GIT_USER "rgleason")
set(GIT_REPOSITORY_NAME "weather_routing_pi")
```

So the problem with the double version showing in the Plugin Manager list appears to be coming from the CMakeLists.txt"  double entry of set(VERBOSE_NAME ?  One with caps and one without?

Does this then mean that I need to fix it, increment the version number  and Push a new PR with a Tag to create new tarballs in Cloudsmith, then I guess I have to update the rg-alpha branch with new xmls?   Is that correct?

bdbcat, 2020-01-26:
Rick...
Not sure exactly what your present configuration is, but note this:
1.  The Common Name comes from the source code.  From weather_routing_pi.cpp, in github:

```
wxString weather_routing_pi::GetCommonName()
{
      return _("WeatherRouting");
}

```
2. In order for the plugin manager to work consistently, the <name> tag in the XML must match this name.
3.  The (sometimes inconsistent) values in CmakeLists.txt do not affect the plugin manager at run-time.

So I would start from scratch, manually delete all your plugins, double check the <name> tag on the XML, and try installation.

In Production, this should not occur, since we would not commit an XML to master without verifying the <name> tag.

leamas, 2020-01-26:
> I prefer to see all possible plugins, installed or not, in one common UI.
> At the very least, we can sort the list to show installed plugins at the top of the list, followed by (alpha?) sorted available plugins.

I have had the same idea. Basically, show two independent lists in the same UI.
 
> More?

For the uninstalled plugins at the end:

  - Hide the Preferences button  completely
  - Replace leftmost icon with a standard package one
  - If possible, change the background colour some other , interesting grey nuance

> followed by (alpha?) sorted available plugins.

Sorting is a separate and somewhat complicated issue. Let's leave this for now, besides having the uninstalled at the end. 

Just my 5c...

rgleason, 2020-01-26:
Thanks.  So just to be clear, which CMakeLists.txt variable?   set( ???_name WeatherRouting) needs to match the "common name in source code"?   IE in weather_routing_pi.cpp 
```
wxString weather_routing_pi::GetCommonName()
{
      return _("WeatherRouting");
}
```

Also when I removed all weatherrouting.dll and directories (there were two located at

- C:\Users\Rick\AppData\Local\opencpn\plugins\weather_routing_pi
- C:\Users\Rick\AppData\Local\opencpn\plugins\plugins\weather_routing_pi
- (and they seem to often appear this way after some use) 

and reinstalled Weather_Routing through PI Manager, everything seemed fine when usiing the Alpha catalog.  I could reinstall and only the directory 
- C:\Users\Rick\AppData\Local\opencpn\plugins\plugins\weather_routing_pi
existed.  - But why is it another level down?

Weather_Routing was enabled and working with Alpha Catalog.
Then I picked Master catalog and applied it.  WeatherRouting disappeared from the list, but still has and Icon that works in the Toolbar.

Also the Options > PI screen shows two instances of the Catalog selection.
[image]

bdbcat, 2020-01-26:
Rick...
The CMakeLists .txt variables do not matter so much to the manager.
What is required is that the "name" tag in the XML file must match the "common name".

bdbcat, 2020-01-26:
Rick:
"Also the Options > PI screen shows two instances of the Catalog selection."

That is a bug.  Thomas has seen that also.  Hard to reproduce...

rgleason, 2020-01-26:
Dave, see my detailed steps above please to reproduce. I will try it again later. Must paint now.

jongough, 2020-01-29:
Rick,
In testplugin (the one I was working on to try and work out a 'standard' set of files for every plugin to use) VERBOSE_NAME is only used in creating the package names. The name used in the XML is 'PACKAGE', so if you look in the testplugin CMakeLists.txt file  at line 37 this is what puts in the XML the name of the package. There is an error in this file in that VERBOSE_NAME is used twice, the last one being what is actually used later on. Perhaps there needs to be another change to use the 'PACKAGE' name in the 'GetCommonName()' api call  in the same way that 'PLUGIN_VERSION_MAJOR' is.

rgleason, 2020-01-29:
Thank you Jon, for completing the missing information for me.
https://github.com/rgleason/weather_routing_pi/blob/master/CMakeLists.txt

   #name must match plugin's "common name"
   set(PACKAGE_NAME weather_routing_pi)  

   #name used for creating the package names
   set(VERBOSE_NAME weather_routing)

Jon, did not find getcommonname() api call for PLUGIN_VERSION_MAJOR in cmakelists.txt, are you suggesting a change in the Cmake setup or in the API?

If the former, can you make that change in your repos testplugin and I will follow it with weatherrouting, so that the Common name is used in the set(PACKAGE_NAME [variable])?

rgleason, 2020-01-29:
Dave, are these statements correct as far as you know? Thanks.

**Legacy and PI Manager Plugins**

1. A legacy "unmanaged" plugin with an earlier version number can coexist with a more recent "managed" plugin. If you install a managed plugin over a Legacy plugin, and then later un-install the managed plugin, the Legacy plugin becomes available again, one good reason not to remove "shadowed" Legacy plugins, at least until one is certain that the managed version is good. The only way to recover the Legacy plugin is to un-install the managed version.
2. If there is a more recent "Managed" version installed and enabled, the user will be notified that an "unmanaged" plugin exists in a "shadowed" state and "may be safely removed", but this only necessary to save disk space. 
3. Plugin Manager will not ever uninstall legacy "unmanaged" plugins and will never show an "uninstall" button.
4. A "shadowed" unmanaged Plugin will not ever be enabled if there is a managed plugin by the same exact name available in the search path even if "disabled", regardless of version number.
5. To use a "shadowed" unmanaged Plugin, the managed plugin must be removed.
6. The plugin list should never have two plugins with the exact same name. (conflicts with #3?)
7. The plugin common name and version number, except it controls the exact text in the "Action" button for manged plugins. <-- Not entirely clear.
8. Upgrading to a managed plugin (from Legacy) should retain the enabled/disabled state. Thus when upgrading to "managed" the legacy plugin does not need to be disabled.
9. The correct plugin "common name" must match and be correct for the "managed" plugin to be paired with a previously installed legacy "unmanaged" plugin, otherwise there will be two different plugins shown.

**Plugin List (Views)  Installed and Available plugins**

1. All possible plugins listed, installed or not, in one common UI.
2. Sort the list to show installed plugins at the top, followed by (alpha?) sorted available plugins.
3. Consider filtering the list to "installed only" & "all available) a control button, but creates tech support questions_pi?"

bdbcat, 2020-01-29:
Rick...
(2) We do not yet know how/where we will notify the user of this.  We do not want to make a big deal here, since it will become a "red herring" masking other troubles.
(3) "Plugin Manager will not ever uninstall legacy "unmanaged" plugins and will never show an "uninstall" button for unmanaged plugins."
(4) Duplicates (1)
(5) Duplicates (1)
(6) Guaranteed by code construction.
(7) I don't understand the context here.
(9)  This is for developer info only.  We will make best efforts not to make incorrect XML's available for download in Beta and master catalogs.  We can expect bugs of this sort in Alpha and custom.  That 's what Alpha testing is for...

**Plugin List (Views) Installed and Available plugins**
Final design still under consideration. Description not valid until design and implementation is complete.

[30 more replies]

closed 2020-03-15 by bdbcat