OpenCPN/OpenCPN #1666: Plugin manager Download message box hangs OCPN

Hakansv, 2020-01-19:
Download message box hangs OCPN while no Internet connection.
Win10
Code including commit: "Ensure Legacy plugins are not marked as "uninstallable""

Method to reproduce:

With no internet connection:
1. Click "Install version 1.0.17
2. The "Installation error" box appear. See shot 1 - Click "OK" and it disappears. 
3. Now the "Downloading" box would have had disappeared as well but it's still there with an active "Cancel" button
4. Click "Cancel" and the "Cancel" Button is deactivated but nothing more. It hangs the O-system so hard I've to kill OCPN process to get rid of it. Shot 2.
5. The log message referred to in the "Installation error" message box is misleading. There's no "/etc/os-release" on a Windows system:
8:57:15 AM: Error: can't open file '/etc/os-release' (error 3: The system cannot find the path specified.)

[image]

[image]

bdbcat, 2020-01-20:
Should be fixed by 98fb06c1a4169c20f84f9a25c60b23104ac80fd1
Dave

Hakansv, 2020-01-20:
Dave..
Couldn't check the initial issue. Commit 98fb06c introduced a new failure.
After a uninstall attempt the new message box appeared but the plugin is still installed and with a new status. See shot
If I try "Upgrade......." the "Uninstall" button is there again but the same is repeated if I use it. Please check

[image]

bdbcat, 2020-01-20:
Hakan...
Unrelated issue, but another bug.  Sometimes, on Windows only, the plugin .dll is not actually removed on the un-install.  I will investigate.
Dave

bdbcat, 2020-01-20:
Hakan....
Windows .dll deletion error should be fixed by 2c7f500ae965df8442f96bfd301e5bf6fc4a0c86.

Hakansv, 2020-01-20:
Dave..
Confirmed in all details.
The misleading log message while try to install without working Internet is still there though.  
8:57:15 AM: Error: can't open file '/etc/os-release' (error 3: The system cannot find the path specified.)

closed 2020-01-20 by Hakansv