OpenCPN/OpenCPN #1667: Plugin Manager: OCPN-system crash on click "Enable" button

Hakansv, 2020-01-19:
OCPN-system crash on click "Enable" button. The buttons would not have been there?
Win10
Code including commit: "Ensure Legacy plugins are not marked as "uninstallable""

Method to reproduce:
See shot.
1. The Squiddio plugin is not installed and the "Install Version 1.0.017"  button is shown.
2. The "Enabled" "Disabled" radio buttons are visible even though the plugin is not installed.
3. If I try to switch to "Enabled" O crashes.
Either:
- hide the radio buttons while not installed.
or
- make the "Enabled" button to install the plugin.
or
- Message the user to first install the plugin.

[image]

bcn2, 2020-01-19:
I can confirm this on macOS with no plug-in installed.
Click on enable will crash O independent from the individual plug-in.

bdbcat, 2020-01-20:
Corrected by de1130f646e80935460ace103b2c5928acaccab1
Thanks
Dave

Hakansv, 2020-01-20:
Confirmed om Win.
Thanks

closed 2020-01-20 by Hakansv