OpenCPN/OpenCPN #1670: PM, UI: Plugins tab/page - sizer

Hakansv, 2020-01-19:
This is about the behavior of entire "Plugins" tab.
Experiences from Win10 and the RPi-7" monitor

The Plugins tab sizer is not made in the same way as the other Options-Tabs. 
If I need to shrink it from any side the right scroll bar disappears. See the first two shots and compare the Connections tab on the third shot. 
The Plugins tab is also to very vertical narrow on small screens. This is especially clear from the RPi shots. As seen only one not expanded plugin row can be fully viewed on the same time but not two. The vertical size can't be changed. Any expanded plugin view is using more than available space. If there's anything to adjust to make the RPi view better I don't know. It's a really small screen. On the other hand we'll probably come to the same situation when we start to deal with Androids and phones? Play store will handle the most but activate/deactivate will it still be need for?)

Vertical vice it's of course better on a ordinary Win10 screen, fourth shot.

[image]
[image]
[image]
[image]

bdbcat, 2020-01-20:
Hakan...
I agree that the various dialogs need re-layout.  After we are convinced that all the information and process is correct, we shall do so.  And tackle Android....

rgleason, 2020-02-23:
Also when a plugin is selected from the Plugin List, the information presented could be more compact possibly, so that it is all seen and selections can be made.

Maybe separate Enable and Disable buttons could become just an Enable button that is toggled or something, and Reside with "Plugin Status"

Hakansv, 2020-03-13:
With the new icons and corresponding text moved to the help text the small screen outlook is better. Now at least one plugin's row is fully viewed in collapsed mode. 
[image]

closed 2020-03-13 by Hakansv