OpenCPN/OpenCPN #1673: CI scripts

transmitterdan, 2020-01-21:
Would anyone be opposed to removing the hard coded repository names and substitute them for environment variables? Would be nice for a forked repo to do test builds before making a PR but now you have to hand edit the yml files every time. By using environment variables for things like artifact upload targets and which branches build in CI it would be more generic.

rgleason, 2020-01-21:
Dan, I suggest you review all the work that Jon Gough has done to make cmake and CI setup much easier. It is not perfect yet, but it is what we used for testplugin_pi and weather_routing too.
https://github.com/jongough/testplugin_pi
Look at the head of the cmakelists.txt  for your variables that get changed.
There a a bunch of directories and files that are interdependent, but this is an excellent start to simplifying the move to Plugin Installer. 

See also https://github.com/rgleason/weather_routing_pi
and the work Jon has done https://github.com/jongough/weather_routing_pi which has been merged into my repository.

squiddio_pi is using an earlier setup that Alec created in the beginning, just to get it working and Jon has taken it a step further for ease of use.

rgleason, 2020-01-21:
Also, perhaps see Dev Manual "PI Manager" Pages. 
PI Manager Dev Setup https://opencpn.org/wiki/dokuwiki/doku.php?id=opencpn:developer_manual:pi_installler_build_deploy

PI Manager Dev Procedure
https://opencpn.org/wiki/dokuwiki/doku.php?id=opencpn:developer_manual:pi_installer_opencpn_plugins

Also Alec has a github wiki here https://github.com/leamas/OpenCPN/wiki

leamas, 2020-01-22:
The setup in the plugins is something different than the main project.

As long as patches uses constructions with hardcoded defaults I don't see any problem. See for example `ci/appveyor-upload.sh` which has a hardcoded name for the upload repo which can be overridden by $CLOUDSMITH_REPO

Bottom line: if possible, we should IMHO not patch the scripts in a way that forces the main project to set environment variables etc currently defined in scripts. That said, simplifying test builds is an important goal.

rgleason, 2020-01-22:
I believe Jon's improvements for Plugin Devs follows Alec's guidelines.

transmitterdan, 2020-01-22:
I am not too worried about plugin builds. 

But the OpenCPN build config specifies specific branches to be built. Is there a way to make that environment dependent?

leamas, 2020-01-22:
The situation differs
  - On appveyor, the branches built is defined in the web ui, does not depend on appveyor.yml
  - On travis and circlec only some branches are built as defined by .yml files.

I have walked around these issues by pushing my feature branches to master on my own repo  and thus have them built. It's a kludge, but it works.

leamas, 2020-01-22:
> But the OpenCPN build config specifies specific branches to be built. Is there a way to make that environment dependent?

This would probably mean generating the yml fiels from yml.in or so. It adds some complexity which needs to be motivated.

transmitterdan, 2020-01-23:
Can the .yml file be configured to build any pushed branch?  For the main repo it would only build branches that Dave pushes which is infrequent and generally only master.  What is the point of building a branch that hasn't changed (been pushed)?

leamas, 2020-01-23:
Technically, yes. It's just a question about removing the filter sections. Dunno if it's a good idea, but I cannot really see any arguments why not.

Why don't you make a PR som we have something to tinker with?

bdbcat, 2020-02-23:
OK to close?

closed 2021-02-01 by transmitterdan

rgleason, 2021-02-01:
I don't know if this helps, but Jon and I have worked out a way to build or not build certain environments by making changes in the circleci config.yml.   Also we have a true/false switch to deploy or not.  We also use  [skip travis] and [skip appveyor] in the commit to control those.  It shortens build times when needed.

TDan,  I have plugins that I have added collaborators who are allowed to submit a PR and then accept and merge it, so it will build on my repository.  That has worked a number of times when I haven't been around, or I needed an expert programmer to fix something.

Yes Bdbcat you can close (as far as I am concerned).