OpenCPN/OpenCPN #1674: compile problem

dlmult, 2020-01-21:
a lot of redefinition problems in winsock2.h when I compile

nohal, 2020-01-21:
AFAICT there are none: https://ci.appveyor.com/project/bdbcat/opencpn/builds/30247672
So at least try to tell us what you actually do.

dlmult, 2020-02-09:
I follow the steps of "https://opencpn.org/wiki/dokuwiki/doku.php?id=opencpn:developer_manual:developer_guide:compiling_windows". My windows version is windows 10.

transmitterdan, 2020-02-09:
You have to tell us more than that.  Nobody else has the problem you describe.  We compile on Windows 10 many times per day.  So you are doing something different than everyone else.  It's possible the instructions are not up to date.  But you have to tell us a lot more information if you want advice.  For example, what version of Visual Studio do you use.  Most of us are using Visual Studio 2017 Community Edition to compile the current master branch.

Also, tell us why you want to compile the Windows version.  Normally you don't need to do that as there are artifact files containing the latest master branch build for Windows always available.

dlmult, 2020-02-12:
I have an idea. Why not provide "opencpn.sln" like "wxWidgets" does. Usually, "cmake" process creates a lot of problem.

transmitterdan, 2020-02-12:
That’s how it works. Running cmake generator creates opencpn.sln.

dlmult, 2020-02-12:
However, the created opencpn.sln has build problems.
For example "LNK2001	unresolved external symbol "__declspec(dllimport) public: __thiscall wxUniCharRef::~wxUniCharRef(void)" (__imp_??1wxUniCharRef@@QAE@XZ)	opencpn	E:\Program2\OpenCPN\build	E:\Program2\OpenCPN\build\S57ENC.lib(Osenc.obj)	1"
and
"LNK2001	unresolved external symbol "__declspec(dllimport) public: __thiscall wxString::const_iterator::~const_iterator(void)" (__imp_??1const_iterator@wxString@@QAE@XZ)	opencpn	E:\Program2\OpenCPN\build	E:\Program2\OpenCPN\build\S57ENC.lib(s57chart.obj)	1"
"LNK2019	unresolved external symbol "__declspec(dllimport) public: __thiscall wxString::iterator::~iterator(void)" (__imp_??1iterator@wxString@@QAE@XZ) referenced in function "public: enum AIS_Error __thiscall AIS_Decoder::Decode(class wxString const &)" (?Decode@AIS_Decoder@@QAE?AW4AIS_Error@@ABVwxString@@@Z)	opencpn	E:\Program2\OpenCPN\build	E:\Program2\OpenCPN\build\AIS_Decoder.obj	1".

I have confused a few days.
Can you give me more details about the compiling.
Thank you.

Hakansv, 2020-02-12:
I say as TDan. Build with VC2017 is no problem for current O master.
Did you really followed every single step here:
https://opencpn.org/wiki/dokuwiki/doku.php?id=opencpn:developer_manual:developer_guide:compiling_windows
Read carefully every word. All matters. 
Do you use a correct terminal? Correct built and version of wxWidgets. Correct system variables for wxWidgets. Correct Windows binaries?

dlmult, 2020-02-15:
Just like the release prebuilt version of wxWidgets-3.1.2.7z, can you provide me a debug version of wxWidgets-3.1.2? Thanks.
I have tried a few days. But I can not build the right debug version.

transmitterdan, 2020-02-15:
You can build any version of wxWidgets from their git repository. For debugging I find it doesn’t matter which version of wxWidgets I use as long as I build everything with the same wxWidgets version. For example, I usually debug using the latest version of wxWidgets master branch. After debugging then I build release version using the prebuilt wxWidgets.

rgleason, 2020-02-15:
Did you follow this specifically and very carefully?
https://opencpn.org/wiki/dokuwiki/doku.php?id=opencpn:developer_manual:developer_guide:compiling_windows#get_the_wxwidgets_312_sources_and_built_them
Show us screenshots or a text copy  of your commands and output here.