OpenCPN/OpenCPN #1677: Help window is not fixed at the edges of the screen

bc38, 2020-01-23:
The help display window is not fixed at the edges of the screen borders. The upper part of the window where the close button goes beyond the screen, so because of this it is impossible to close the help window!

transmitterdan, 2020-01-23:
On what operating system and what screen size does this happen?

bc38, 2020-01-23:
windows 10
screens: 15.6,` 24`
the window should be fixed at the edges at least

[image]

rgleason, 2021-04-28:
Windows 10  Ov5.2.4 Release  I find that the Help screen is working fine, and suggest that this should be closed now.  It can always be reopened.  I do not understand the margin problem and the OP has not described the issue adequately.

[image]
[image]
[image]

closed 2021-04-28 by rgleason