OpenCPN/OpenCPN #1681: Incompatible plugins displayed and loaded.

leamas, 2020-01-24:
The current checks for plugins displayed and eventually loaded looks for PKG_TARGET i. e., the operating system. However, to be compatible a plugin msy not only be built for the correct os but the correct OS version. 

This means that current code will try to load a plugin built for windows 10 on a windows 7 host causing all sorts of interesting issues. Same applies for plugins built for wrong Ubuntiu version etc.

We need to check not only OS but also version before deeming a plugin as usable.

Hakansv, 2020-01-24:
Alec.

> This means that current code will try to load a plugin built for windows 10 on a windows 7 host causing all sorts of interesting issues. Same applies for plugins built for wrong Ubuntiu version etc.

Today's Windows installer package are valid for all Win versions. From XP to W10. What will change to make a difference with the new installer? I suppose every Win version will reply correct when e.g. questioned for the program folder.

leamas, 2020-01-24:
This is not about the installer but the plugins. The most obvious problem is that a plugin built for Ubuntu 19.04 cannot be used on for examle 18.04. Same goes for MacOS and, to some extent, flatpak.

Are you saying that a windows plugin built for W7 can be used on W10?

Hakansv, 2020-01-24:
Yes I am.
The only Windows version adaption we do is to tell VS2017 to use the build for XP like this: 
cmake -T v141_xp ..
Then the plugin installer package and also the .dll itself will fit on all Win versions XP - W10.
Note:
After next version of wxWidgets or anything that force us to use a VS above 2017 we also have to end the XP support.
(VS = Ms Visual Studio)

leamas, 2020-01-24:
Thanks!

This basically boils down to that on windows we should not check version level. Simple enough.

rgleason, 2020-01-24:
Agreed with Hakan, If we ever change to 64 bit windows, it might make a difference. Also when XP support is dropped, you'll want to have a message or something?

leamas, 2020-01-25:
We definitely need os version checking (and thanks for 32/64 bit remainder, another part of this not only  for Windows). The only question is how to handle windows which seems to be a special case. 

For now, as Hakan has explained, I'm thinking all windows plugins should be considered compatible. This is certainly not the case for other OS:es

Hakansv, 2020-01-25:
On Win: If O will not be supported by any aged version now or in future that's not a plugin issue. If no OCPN is present no need for any plugin. If O is there, so far so good.

rgleason, 2020-01-25:
Agree with Hakan.  

Windows XP, Win7 & Win10 Plugins are the same, and are generally forward compatible from the OpenCPN supported Plugin API version except when there has been a wxwidgets upgrade and O has changed versions with a major upgrade (4.0, 5.0, 6.0?). Then the plugin needs to be recompiled.  If Opencpn changes to 64 bit at some point, that may need to be checked. Also Windows XP support may soon have to be dropped.

PR #1691 (2020-02-03): Check also version in compatibliity check (#1681)
Check not only operating system but also version before deeming a plugin as compatible (that is, loadable).

The previous code which basically checked for Mint and used Ubuntu in this case case been removed, mostly because it wasn't possible to determine the version to use this way.

New code checks that operating system and operating system version are compatible with the  plugin by comparing the configuration constants  PKG_TARGET and  PKG_TARGET_VERSION with plugin metadata.

There are two ways to override the default checks:

   - By settin [... cut]
closed 2020-02-03 by bdbcat