OpenCPN/OpenCPN #1685: PPA: anyarch packages depend on arch specific package

FredericGuilbault, 2020-01-28:
Opencpn does not provide packages for arm64 (which is sad but ok) 

So I have to compile opencpn by hand. 
Then I want to use the tcdata and gshhs packages marked as multiarch on the PPA . 

But they have the `opencpn` package as depedency and it cause error as it's not available for my architecture
```
$ apt install opencpn-tcdata

Reading package lists... Done
Building dependency tree       
Reading state information... Done
Some packages could not be installed. This may mean that you have
requested an impossible situation or if you are using the unstable
distribution that some required packages have not yet been created
or been moved out of Incoming.
The following information may help to resolve the situation:

The following packages have unmet dependencies:
 opencpn-tcdata : Depends: opencpn but it is not installable
E: Unable to correct problems, you have held broken packages.
```

As I understand. tcdata and gshhs are generic dataset that you guys/gals package, and it does not strictly depend on opencpn to be useful., Even, others projects like xygrib are packaging the same dataset.   

Im not a guru at debian packaging, It feel to me that it's more of a `Suggested package` or `Recommended package` then a direct dependency. The fact that files are installed in the `share` folder kinda speak for it self. 

In a perfect world it could even be outside opencpn project and be a common ressource for any other package that would need it. like the `geoip-database` package.

---
For the record, there is a workaround but it's quite ugly : 
```
	apt-get download opencpn-tcdata
	dpkg -i --ignore-depends=opencpn opencpn-tcdata*
	apt-get download opencpn-gshhs-*
	dpkg -i --ignore-depends=opencpn opencpn-gshhs-*
```

transmitterdan, 2020-01-28:
There is a much simpler solution.

cd ~/opencpn/build
cmake -DOCPN_BUNDLE_TCDATA=ON -DOCPN_BUNDLE_GSHHS=CRUDE ..
sudo make install

FredericGuilbault, 2020-01-30:
Thx maps are included by default now. That's a way better workaround. 

I was using flags documented here, and it wasn't working: 

https://opencpn.org/wiki/dokuwiki/doku.php?id=opencpn:developer_manual:developer_guide:compiling_linux:building_on_armhf_linux_-_armbian_-_orange_pi
 
Maybe an outdated procedure ? Should I open a new issue for this ?

transmitterdan, 2020-01-31:
The documentation is not in this repo.  You can become and editor of the Docu-wiki and edit it yourself.

FredericGuilbault, 2020-01-31:
It seem like the wiki is not open to public registration 
https://opencpn.org/wiki/dokuwiki/doku.php?do=login

rgleason, 2020-01-31:
Please log into CF and send me a PM. with your email.  rgleason. Thanks

FredericGuilbault, 2020-02-06:
im sorry, but what is `CF` ? @rgleason

transmitterdan, 2020-02-06:
www.cruisersforum.com - that is where users interact and discuss things like documentation, use cases, hardware interface ideas, etc. It is free to sign up.

rgleason, 2021-04-27:
TDan wrote:
```
There is a much simpler solution.

cd ~/opencpn/build
cmake -DOCPN_BUNDLE_TCDATA=ON -DOCPN_BUNDLE_GSHHS=CRUDE ..
sudo make install
```

This has been added as an alternative to 
https://opencpn.org/wiki/dokuwiki/doku.php?id=opencpn:developer_manual:developer_guide:compiling_linux:building_on_armhf_linux_-_armbian_-_orange_pi

This could be added to the newer Dev Manual.
https://rgleason.github.io/ocpn-manuals/ocpn-dev-manual/5.3.1/od-compile-linux.html

However there is no logical place for it  @leamas ?

leamas, 2021-04-27:
Also here it seems like we (you?) should file a bug against ocpn-dev-manual.

It belongs to OpenCPN development | Compiling  | Cmake

rgleason, 2021-04-27:
I can certainly do that, but others will need to fix them.. Then can be closed. That was the objective.

DONE - https://github.com/rgleason/ocpn-dev-manual/issues/61

Can this be closed please?

closed 2021-04-27 by rgleason

FredericGuilbault, 2021-04-28:
thx @rgleason