OpenCPN/OpenCPN #1687: Plugin Manager: Doubled up Catalog Selection Area

rgleason, 2020-01-29:
This is reproducible in Windows 10  Opencpn 5.0.522 built  including Jan 25 commits for Dashboard with SignalK.

1. From Master Catalog, select Alpha channel and Update Catalog. Close Options Plugin.
2. Open Options Plugin, select Master channel and Update the Catalog. The double selection area appears.
3. Then do it again and you will have three  Selection Windows, occupying the entire screen.
4. Continuing this adds more Selection areas.

Tai-fun reported this and so did I earlier.

Doubled selection Area
[image]

Tripled selection Area
[image]

bdbcat, 2020-01-29:
Rick...
Yes, I see this now.
Patch coming...

Dave

bdbcat, 2020-01-29:
Rick/Thomas...
Should be fixed by 981b105c944da9f1553883fd95658883ef087155
Dave

tai-fun, 2020-01-30:
Dave
Yes...your patch solved the problem. No more multiple plugin catalog manger
panels, just the one at the bottom.
Thanks
Thomas

bdbcat, 2020-02-02:
Rick...
Ready to close?

rgleason, 2020-02-02:
Fixed. Thanks Dave.

closed 2020-02-02 by rgleason