OpenCPN/OpenCPN #1688: Plugin Manager: Managed changes to Legacy dependent on Catalog

rgleason, 2020-01-29:
Dave,

Please see https://github.com/OpenCPN/OpenCPN/issues/1665#issuecomment-575955401
Here are some ideas to consider.

**PROBLEM**
> A "managed" plugin, Radar, changes to "legacy" depending on the catalog chosen. I think I kind of understand this, because a given catalog may not have current or any installation xml for that plugin? Anyway it acts differently depending on what catalog is applied.
> 
> Start with Alpha channel a number of screenshots

9-Radar50beta2-now "Unmanaged" in Master Catalog
[image]

11-Radar50beta2   "Managed" after Alpha Catalog update 
[image]

**IDEA**
PI Manager "managed" is installed to %LOCALAPPDATA%/opencpn/plugin
Package Legacy "unmanaged" is installed to %PROGRAMDATA%/opencpn/plugin

For Plugins that are in the  Options > Plugin List that do not have an XML record because a different Catalog has been selected, is it possible to check if there is a plugin by that name in the  %LOCALAPPDATA%/opencpn/plugin directory to help determine if it is actually "Managed from another Catalog"?  

Then is it possible to then name it "Managed from another Catalog".

bdbcat, 2020-01-29:
Rick....
Let us agree that we are focused on the class of plugins that are in the process of being updated from Legacy (un-managed) to fully managed.

In production, we will have only Beta and master catalogs.
A (normal) Beta testing  user selecting Beta catalog may expect to see a managed candidate to replace the Legacy plugin.  Hooray!  Progress! 
I do not think that trying to guess the condition of a Legacy plugin is going to be profitable.  What we should expect is that Legacy plugins in the process of being upgraded to Managed will quickly migrate from Beta to master.  Then the management process takes over, and the Legacy plugin is completely uninvolved.

You are seeing these inconsistencies because you have access to more than two catalogs, some of which include redirection.  This is more of a developer view, with all sorts of corner cases and inconsistencies before the correct code is committed to Beta.

Dave

rgleason, 2020-01-30:
Yes, I understand it is a Dev issue when using Alpha. 
I was just trying to get the orphaned (no xml data available) "managed" plugin named something other than "unmanaged", some other state or something because it would be clearer to everyone.

Impossible I guess.

rgleason, 2020-01-30:
OK, so why spend valuable programming effort making Alpha work when switching catalogs back and forth?  I've got it now.  Hopefully all legacy plugins will disappear fast! 

(but we need to make the setup and cmake + other files easier for plugin devs)  
Jon's [testplugin ](https://github.com/jongough/testplugin_pi)is the most developed example so far. We just need to get the icons working better.

Closing

closed 2020-01-30 by rgleason