OpenCPN/OpenCPN #1692: SignalK Connection lost timeout.

Hakansv, 2020-01-30:
We need some watchdog for the SignalK network connection.
If the data stream is lost, e.g. by a SignalK server restart, the communication is lost and not reestablished.
To restart the connection either OCPN has to be restarted or to deactivate the connection in Connections tab and activate it again.
Valid for both Win10 and Linux, RPi

Hakansv, 2020-02-01:
Some more connected observations:
- The SignalK connection is slow to disconnect. If you in Connections tab deselect a SK connection and select another you have to wait at least 5 seconds until the Apply button is dimmed or unselected. If next connection is selected to fast and a the Apply button is clicked the config file connections list is some times damaged and O crash upon next "Apply". (Two simultaneously running threads?) 

- If one SK connection is deselected and another SK connected, like via the WAN IP, is selected for a while and then again deselected and returned to the first. Now the SignalK server reports three connected clients. So neither of the two deselected connections are really closed. When I afterwards close O all three clients disappears from SK-server.

Hakansv, 2020-02-03:
Dave..
Timer adjustment and isThreadrunning check did the thing for disconnecting SK in a proper way. 
Now I can instantly switch between two connections wo failures.  That one is solved. Thanks.

About SignK connection while data stream is lost:
If my own local network is gone it's detected and SignalKDataStream tries to restart every 10 seconds. It also succeeds when the net is up again. That works fine.
But, as mentioned, if SK is restarted and the data stream is lost it's not detected but the socket to SK is lost forever so nothing is received when SK is up again. I've to disable - enable the connection or restart O. We may need a data stream or better an active socket watchdog as well. (It seams that "vessel.self " is always sent. )

bdbcat, 2020-02-03:
Hakan...
Yes, we need a re-connect watchdog.  I wanted to confirm the new shutdown logic before implementing that.  Next step...

Hakansv, 2020-02-03:
Dave..
Yes, of course. Hopefully my wordings was read as info and not demanding. :)

bdbcat, 2020-02-05:
Hakan...
I have implemented a watchdog re-connect logic in signalK client. c8c5d2524d3d2b61d66388445163f821e47c17dd

In normal operation, the watchdog activates after 5 seconds of no activity.  The re-connect timer then has a 10 second retry period.
Tested reasonably on linux.  Minimal Windows testing so far.
Dave

Hakansv, 2020-02-05:
Dave..
Splendid!
Tested on Win10 and RPi3 (Stretch)
O SK-connection now withstands SignalK server restart, SignK PC reboot and network down-up.
Thanks.
Håkan

closed 2020-02-05 by Hakansv