OpenCPN/OpenCPN #1694: Plugin Manager: Installation destination directory

rgleason, 2020-01-31:
Also I notice that radar_pi, testplugin and weatherrouting seem to install to
%localappdata%\opencpn\plugins\plugins
C:\Users\Rick\AppData\Local\opencpn\plugins\plugins
    radar_pi    (has data and icons)
    testplugin_pi   (has data and icons)
    weather_routing_pi  (has data and icons)

I find in C:\Users\Rick\AppData\Local\opencpn\plugins
    oesenc_pi
    and an empty directory "testplugin_pi" but it has no data or icons.
    Also all the dll are in this directory.

Is this an issue with Plugin Manager or our plugins?

leamas, 2020-01-31:
To answer this you need to check the tarball layout and compare that to the docs in https://github.com/leamas/opencpn/wiki/Tarballs

rgleason, 2020-01-31:
That's helpful and I will check this, but why do I find oesenc in plugin/plugin too?
That is a pi you setup.

bdbcat, 2020-01-31:
Rick...
Disregard oesenc in current catalogs.  I have this now, and there will be a new version shortly.

rgleason, 2020-02-01:
From everything I see weather_routing is ok (except perhaps the name which needs to be fixed)
When I download the tarball and inspect its directory structure, it appears to be fine and as expected.

When I use the exe version and try to execute it, picking C:\Program Files (x86)\OpenCPN 5.0.522
and the plugin installs properly in the plugins directory with the dll there and the data directory is here  C:\Program Files (x86)\OpenCPN 5.0.522\plugins\weather_routing_pi\data

So, the PI Installer is not doing the right things IMHO.
Please see attached and take the "zip" Off.

[weather_routing-plugin-msvc-10.0.14393.xml.zip](https://github.com/OpenCPN/OpenCPN/files/4141948/weather_routing-plugin-msvc-10.0.14393.xml.zip)

[weather_routing_pi-1.13.2.0-ov50-1.16_msvc-10.0.14393.tar.gz.zip](https://github.com/OpenCPN/OpenCPN/files/4141950/weather_routing_pi-1.13.2.0-ov50-1.16_msvc-10.0.14393.tar.gz.zip)

[image]

leamas, 2020-02-01:
The path of the dll file in the tarball is    top-dir/plugins/weather_routing_pi.dll. However, according to the docs it should be top-dir/plugins/weather_routing/weather_routing_pi.dll (assuming that the plugin name is weather_routing).

Here is most likely also a regression in the installer code. Attaching untested patch for now:

[0001-PluginHandler-Fix-broken-install-path-on-windows-169.patch.gz](https://github.com/OpenCPN/OpenCPN/files/4142190/0001-PluginHandler-Fix-broken-install-path-on-windows-169.patch.gz)
 .
 
I assume that you remove all files under %localappdata%\opencpn\ manually before testing...

rgleason, 2020-02-01:
Are you sure about this? 
Legacy has all dll's here: C:\Program Files (x86)\OpenCPN 5.0.0+9065270\plugins with weather_routing_pi\data directory below.

Also I find the managed dlls here:
C:\Users\Rick\AppData\Local\opencpn\plugins (radar_pi.dll, squiddio_pi.dll, testplugin_pi.dll, weather_routing_pi)
and I find the weather_routing_pi directory here
C:\Users\Rick\AppData\Local\opencpn\plugins\plugins  (testplugin_pi, oesenc_pi, weather_routing_pi)

So what is it supposed to be?

bdbcat, 2020-02-01:
Alec...
Where is this documented?
Dave

leamas, 2020-02-01:
https://github.com/leamas/OpenCPN/wiki/Tarballs

leamas, 2020-02-01:
Basically, here seem to be two errors:

  - The plugin layout is not according to the docs
  - PluginHandler seems to have a regression involved in the `plugins\plugins` path, hence the 
    patch...

bdbcat, 2020-02-01:
Alec...
Maybe not so simple.
I am investigating now.  I'll figure it out and report soon.
Dave

bdbcat, 2020-02-01:
Alec...
Not sure about "share" directory destination on Windows.
After installation, do we want to see it as a child of "plugins", or as a perfect sibling of plugins?
The tarball spec has it as a perfect sibling of "plugins"
Dave

bdbcat, 2020-02-02:
I conclude by comparison with Legacy plugins, that "share" should be a sibling of "plugins".
So, some touch-up required for Windows platform.

bdbcat, 2020-02-02:
Rick...
Your suspicions about install location confirmed.
New code with corrections building now. https://github.com/OpenCPN/OpenCPN/commit/bf56307e390260d4c24a53df8f12fa64cfe69590
You should delete ALL of the directories and contents in 
"C:\Users\Rick\AppData\Local\opencpn"
and start over.
Then maybe we can investigate what has happened to the toolbar icons.

leamas, 2020-02-02:
Great job, both of you! 

Now,  this does *not* affect how the tarball should be organized. Nor does it fix the tarball layout which not is up to the docs, see https://github.com/OpenCPN/OpenCPN/issues/1694#issuecomment-580997218

The patches fixes how the paths in the tarball actually are installed. To be able to close this bug we need to verify that correct paths in the tarballs are installed in correct locations (there is a possibility that it works by sheer luck rather than the proper way...)

rgleason, 2020-02-02:
Thanks Alec. Removed all localappdata/opencpn/plugins and installed newly built O then opened the Alpha catalog and Installed 1.13.3.1 managed.   It installed to /plugins/dll + weatherrouting directory.

We seem to have changed the plugin name to "weatherrouting"  instead of legacy's "weather_routing_pi.dll" and "weather_routing_pi" directory. Should this be fixed?

I find a legacy weatherrouting plugin and a managed. The managed works, but when I close I seem to have 3 dummy icons for weatherrouting. 

So it installs to the right directory now!
[image]
[image]
[image]
[image]

rgleason, 2020-02-02:
Should we close this?

bdbcat, 2020-02-02:
Alec...
There is an inconsistency in the specified tarball layout.
Consider oeSENC. 
In the tarball, the helper oeserverd is at
`{topdir}/plugins/oesenc_pi/oeserverd.exe`

At installation,in order to find it in the $PATH,  we need it to be at:
`{installDir}/plugins/oeserverd.exe`

You see the difference....

bdbcat, 2020-02-02:
Rick...
Once again, you really should fix the name of the managed plugin so that it properly overlays the Legacy plugin.  Just do it.
Rule:  The Common Name must match the "name" tag in the metadata.  The name of the actuall .dll file, and the installed sub-tree, do not matter.

leamas, 2020-02-02:
> At installation,in order to find it in the $PATH, we need it to be at:

Not really. On Windows, the path to the plugin library, for example` ...\plugins\oesenc\`,  is added to PATH. See SetLoadPath in PluginHandler.h.

The possible problem with installing helper libs and binaries  in .`..\plugins `  is name collisions. We know the plugin dlls are unique, but cannot make the same assumption about helper stuff.  Installing them in local libraries is not a complete remedy, but makes IMHO the situation somewhat better.

rgleason, 2020-02-02:
Yes, agreed, thats what we are trying to do, but something is amiss Will check it again.

bdbcat, 2020-02-03:
Alec...
On Windows:
Sorry, nope, I cannot find any code where the $PATH is modified (dynamically?) to include a specific plugin's directory.  Best it does is "{installDir}/plugins".
Perhaps I "optimized this away" in some big UI code merge?

I also am concerned about collisions in "plugins" directory, which got me thinking about this.  Thought came to me as I was working on oeSENC and oeRNC simultaneously...
I agree the best solution is to keep all plugin helpers and dependent dlls in the plugin's owned directory.

leamas, 2020-02-03:
> Sorry, nope, I cannot find any code where the $PATH is modified (dynamically?)
 
But I can ;) It is in pluginmanager.cpp, in `static void setLoadPath()`, called from  `PlugInManager::LoadAllPlugIns()`.

This is old code.  While looking at it I see that $PATH will accumulate the same leading entries over and over again for each call to LoadAllPlugIns(). To be fixed, but it should not affect the basic functionality

bdbcat, 2020-02-03:
Alec...
Still nope.  This code does what it does just fine.  But it simply does not work for Windows.

For Windows, to honor the tarball layout, we will need something like
void AddLoadPath(PlugInContainer *pic){
...add "pic->m_plugin_filename_without_extension" to $PATH
}
We would need to call this as each plugin is discovered and added to the list.
Or, simpler, just let the plugin find it's helpers without dependence on $PATH.  Only Windows would need special handling.

leamas, 2020-02-03:
I don't really follow you here. Are you saying that we need to add the directory part of discovered plugins to $PATH? 

We know for sure where the plugin foo's dll  is located, right? And thus also the `...\plugins\foo` directory (?) So why do we need to determine the path from the dll file?

"confused"

bdbcat, 2020-02-04:
Alec...
It is simply this:
This code, extracted from the oesenc_pi you adapted, will not work on Windows.  
```

      // Specify the location of the xxserverd helper.
      wxFileName fn_exe(GetOCPN_ExePath());
      g_server_bin = fn_exe.GetPath( wxPATH_GET_VOLUME | wxPATH_GET_SEPARATOR) + _T("oeaserverd");
      
      if (!wxFileExists(g_server_bin)) {
        std::string path(find_in_path("oeaserverd"));
        if (path == "") {
            wxLogWarning("Cannot locate oeaserverd binary in $PATH");
        }
        else {
            g_server_bin = wxString(path.c_str());
        }
      }

```Now I know that this may be obsolete code, and I now "own" oesenc_pi.  But I also know that this will not work for Windows, while it works fine for linux.  And this is due to the ever-so-slightly different tarball spec for Windows.  Which I agree with.
So, if we are to honor the specified tarball structure for Windows, then the plugin must "special case" Windows in order to find the helper program(s).
And this is no problem, really.
I merely make the point that the Windows tarball is "different" to the linux tarball, in a way that is significant to the plugin.
Anyway, I will adapt....
Dave

```

leamas, 2020-02-04:
Hm... In my code there is a commit  "e81c463: oesenc_pi: fix binary helper lookup"  This adds the following sequence after code above  (+ the find_in_path() function):

    +    if (!wxFileExists(g_sencutil_bin)) {
    +        std::string path(find_in_path("oeserverd"));
    +        if (path == "") {
    +            wxLogWarning("Cannot locate oeserverd binary in $PATH");
    +        }
    +        else {
    +            g_sencutil_bin = wxString(path.c_str());
    +        }
    +    }

Bottom line: helper lookup, like data, needs to be fully relocatable. Which it becomes with this. Or?

leamas, 2020-02-04:
Sorry for absolutely insane, deleted, post. Back when I'm sober...

bdbcat, 2020-02-04:
Alec...
I think I have fixed this up in the core, for Windows.

extern DECL_EXP wxString GetPluginDataDir(const char* plugin_name);

was simply broken for Windows.  Fixed now.  Changes rippled elsewhere, causing other adjustments required.
Verified by me oernc_pi, my present focus.  Here the helper is found like:
```

      // Specify the location of the xxserverd helper.
#ifdef __WXMSW__
      g_server_bin = GetPluginDataDir("oernc_pi") + _T("\\oeaserverd.exe");

#else      

So I think we are OK now. We shall see how Rick's plugins perform.
Dave

```

rgleason, 2020-02-05:
Dave, I have built a new opencpn for windows.  Weather Routing is still screwed up. I will have more time tomorrow afternoon to look into this further.
 - Right now, %localappdata%/\pencpn\plugins\weatherrouting_pi.dll
 - C:\Program Files (x86)\OpenCPN 5.0.522\plugins\weather_routing_pi.dll

rgleason, 2020-02-05:
I've got it building as weather_routing_pi.dll  now, but need to push to cloudsmith and modify the meta-url  in rg-alpa.  

In the options/plugins  Plugins List "managed" still reads as "weatherrouting" and the "unmanaged" as "WeatherRouting".   I haven't figured out where that is fixed yet.

Also the "managed" does not seem to have an enable button, but that is possibly because the alpha catalog is looking at the old version & xml, yet I have installed the new version locally, but that is not seen.   So I need to push to cloudsmith and change the meta-url  ocpn-plugins.xml in rg-alpha branch with the new record for weather_routing.   ...Tomorrow afternoon.

bdbcat, 2020-02-05:
Rick...
I think you are going in circles.  If you will push to your github account something that will actually build, I'll take a look at it.  Tried to clone just now, but it will not build without errors.
Dave

bdbcat, 2020-02-05:
Rick...
Belay last.
For some reason it does not build on my Win7, but builds OK on W10.
Another problem, for another day....
I have forked your repo, and will work with it...
Dave

bdbcat, 2020-02-05:
Rick...
I am working with weather routing.  Some progress on loading, but now crashes on enable.
I should have a PR for you later this evening.
Please stand by...
Dave

bdbcat, 2020-02-06:
Rick...
There is a Pull Request available in your github weather routing repo.  All seems to be working OK now.  You will need to revert to your cloudsmith repo creds in CMakeLists.txt to publish.
btw, I see no need to publish a Legacy version of this plugin as a github tagged Release. Once a plugin is available as managed format, we will not be updating the Legacy plugin, except in very rare and individual circumstances.  To say another way:
"Once a Managed plugin is published, any related Legacy plugin is deprecated, and has reached "end-of-life".
Correspondingly, in your tests, please DO NOT USE a Legacy plugin built from this code.  It will not work correctly, especially the icons.  If you want to test upgrading from a Legacy plugin, the only Legacy plugin you should pre-install is the official OpenCPN Release plugin.  Any other test results are invalid.

Good Luck
Dave

rgleason, 2020-02-06:
Thanks!

Its building for the second time now (flatpak was having trouble), but everything deployed ok. Haven't added the xml to rg-alpha to be picked up by the meta-url yet. Traveling again tomorrow morning, but hope to get this done in the early morning.  ...Flatpak just built and deployed.

> Correspondingly, in your tests, please DO NOT USE a Legacy plugin built from this code. It will not work correctly, especially the icons. If you want to test upgrading from a Legacy plugin, the only [...]

I thought that originally one of the goals of the PI Manager was to be compatible with the older packaged system. Now moving to PI Manager appears to break the packaged system.  My guess is some old time users will howl, (if it means anything).   Related matter see: https://github.com/OpenCPN/OpenCPN/issues/1695

I'll try to follow your instructions about this, but old habits die slowly.

bdbcat, 2020-02-06:
Rick...
We are compatible with existing packaged plugins.
But if a dev is going to the trouble to build and deploy a managed plugin, I think they would be happy to drop maintenance of the packaged Legacy version, and move forward with the managed variety only.
Or, if they like, stay with the packaged version, only.
But I see no reason to have both, never mind the build problems she will face.

Really, I have no trouble telling users:
"If you need a new feature that has just been added to XXX_pi, you must upgrade to OCPN 5.2 (or whatever), and install the managed plugin."

You will remember my instructions when you find that the packaged version built from your repo does not display toobar icons correctly....
Dave

rgleason, 2020-02-06:
Sure I will try to remember(:-)...
.. indeed neither the downloaded legacy version nor the current packaged version treat icons properly, they both create black squares in the menu, yet they do work. The icons are an annoyance at this point and shrinking the toolbar fixes it. Someone will figure out how to fix that later.

I have installed the alpha version of weather_routing and it works well with no anomalies in icons or double slightly different versions.  It uninstalls and reinstalls and seems to behave with both the recent packaged legacy version and the legacy download (with the exception of icons).

Give me more time to mess with it however before we close this please.

[image]
 
Uninstalled
[image]
Legacy
[image]

Maybe the use of screen space could be better if the Website link was positioned on the left with the enable and disable buttons?  There is something about the left restriction on "Website" that uses up a lot of screen space!

[image]

rgleason, 2020-02-07:
This is working very nicely. The change in icon to green puzzle piece signals that you are using a packaged version. That's good!  I've tested it more and even when I activate "Master" catalog, I can reinstall and uninstall and install the managed plugin weather_routing_pi which is on the Alpha catalog using a meta-url to my repos plugins  branch rg-alpha.

This is working quite nicely now I think. Closing.

closed 2020-02-07 by rgleason