OpenCPN/OpenCPN #1696: Plugin Manager: Icons incorrect in Toolbar

rgleason, 2020-02-01:
The proper icons are located in the data directory.

[image]

[image]

[image]

bdbcat, 2020-02-01:
Rick...
Where is the source code repo and branch for this version of WeatherRouting?
I'll take a look.  Seems to me to be a problem with plugin using legacy style data directory access.
Dave

rgleason, 2020-02-01:
Jon Gough has been helping to configure this plugin for PI Installer. I used his cmake files for testplugin and he has been working back and forth on it to simplify it for Plugin Devs.

The repository & branch  is here https://github.com/rgleason/weather_routing_pi/commits/master
I am still learning about what to look for. It seems to be happening to testplugin too. This would be a big help. Thanks.

"Using legacy style data directory access..."
IE:Instead of %localappdata%\opencpn\plugins\plugins
it is using  %PROGRAMDATA%/opencpn/plugin?

Is this the path for the Toolbar Icon is incorrect somehow?
Because there is a weather_routing icon in the Plugin List

bdbcat, 2020-02-01:
Rick...
Add a line to "icons.cpp" to show exactly where the toolbar icon is assumed to be coming from:
At the bottom of icons.cpp, add the line after the comment:
"Add this line here"...
Then rebuild, and check the logfile for the message "Loading toolbar icon".

Then we shall see...

Dave

```
#ifdef PLUGIN_USE_SVG
    wxFileName fn;
    fn.SetPath(GetPluginDataDir("weather_routing_pi"));
    fn.AppendDir(_T("plugins"));
    fn.AppendDir(_T("weather_routing_pi"));
    fn.AppendDir(_T("data"));
    fn.SetFullName(_T("weather_routing_pi.svg"));
    _svg_weather_routing = fn.GetFullPath();
    // Add this line here.
    wxLogMessage(_T("Loading toolbar icon: ") + _svg_weather_routing;
    
    fn.SetFullName(_T("weather_routing_pi_rollover.svg"));
    _svg_weather_routing_rollover = fn.GetFullPath();
    fn.SetFullName(_T("weather_routing_pi_toggled.svg"));
    _svg_weather_routing_toggled = fn.GetFullPath();
#endif

```

rgleason, 2020-02-01:
I was wrong in the post above, there are svg icons in these locations typically,

- C:\Program Files (x86)\OpenCPN 5.0.0+9065270\plugins\weather_routing_pi\data\weather_routing_pi.svg
- C:\Program Files (x86)\OpenCPN 5.0.522\plugins\weather_routing_pi\data\weather_routing_pi.svg

The  second location is the one we want. It looks to me like the fn.AppendDir & setfullname should work.   Also, there are svg icons at
C:\Users\Rick\AppData\Local\opencpn\plugins\plugins\weather_routing_pi\data\weather_routing_pi.svg
or the same location  %localappdata%\opencpn\plugins\plugins
Can I use that ass the path?

bdbcat, 2020-02-01:
Rick...
Anything in Program Files/.... is just wrong.  No managed plugin parts are installed there.  Those are leftover from the legacy plugin.

What did the log say after rebuild?
Dave

rgleason, 2020-02-01:
Just a sec, I have to add 
C:\Users\Rick\AppData\Local\opencpn\plugins\plugins\weather_routing_pi\data\weather_routing_pi.svg
there, I'll be right back.

rgleason, 2020-02-01:
Tried with \ and got errors about syntax
Tried with / and got this

```
  icons.cpp
C:\Users\Rick\Documents\GitHub\weather_routing_pi\src\icons.cpp(33): error C2059: syntax error: '/'
 [C:\Users\Rick\Documents\GitHub\weather_routing_pi\build\weatherrouting_pi.vcxproj]
C:\Users\Rick\Documents\GitHub\weather_routing_pi\src\weather_routing_pi.cpp(187): error C2065: 'PL
UGIN_COMMON_NAME': undeclared identifier [C:\Users\Rick\Documents\GitHub\weather_routing_pi\build\w
eatherrouting_pi.vcxproj]
```

rgleason, 2020-02-01:
```
#ifdef PLUGIN_USE_SVG
    wxFileName fn;
    fn.SetPath(GetPluginDataDir("weather_routing_pi"));
    fn.AppendDir(_T("plugins"));
    fn.AppendDir(_T("weather_routing_pi"));
    fn.AppendDir(_T("data"));
    fn.SetFullName(_T("weather_routing_pi.svg"));
    _svg_weather_routing = fn.GetFullPath();
   C:/Users/Rick/AppData/Local/opencpn/plugins/plugins/weather_routing_pi/data/weather_routing_pi.svg
    fn.SetFullName(_T("weather_routing_pi_rollover.svg"));
    _svg_weather_routing_rollover = fn.GetFullPath();
    fn.SetFullName(_T("weather_routing_pi_toggled.svg"));
    _svg_weather_routing_toggled = fn.GetFullPath();
#endif

	return;
}
```

rgleason, 2020-02-01:
I has just a little while ago this evening asked jon gough about the missing identifier, because we made some commits to fix the "name" issue,   These are the changes I made in my weather_routing_pi local repository, So I am trying to fix that  PLUGIN_COMMON_NAME  missing definition and this icon issue at the same time.

bdbcat, 2020-02-01:
Rick...
What happened to the line:
    wxLogMessage(_T("Loading toolbar icon: ") + _svg_weather_routing;
  ???
That is the one to add.

rgleason, 2020-02-01:
Duh , sorry

```
  icons.cpp
C:\Users\Rick\Documents\GitHub\weather_routing_pi\src\icons.cpp(33): error C2143: syntax error: missing ')' before ';' [C:
\Users\Rick\Documents\GitHub\weather_routing_pi\build\weatherrouting_pi.vcxproj]
C:\Users\Rick\Documents\GitHub\weather_routing_pi\src\weather_routing_pi.cpp(187): error C2065: 'PLUGIN_COMMON_NAME': unde
clared identifier [C:\Users\Rick\Documents\GitHub\weather_routing_pi\build\weatherrouting_pi.vcxproj]

Code

```
#ifdef PLUGIN_USE_SVG
    wxFileName fn;
    fn.SetPath(GetPluginDataDir("weather_routing_pi"));
    fn.AppendDir(_T("plugins"));
    fn.AppendDir(_T("weather_routing_pi"));
    fn.AppendDir(_T("data"));
    fn.SetFullName(_T("weather_routing_pi.svg"));
    _svg_weather_routing = fn.GetFullPath();
   wxLogMessage(_T("Loading toolbar icon: ") + _svg_weather_routing;
    fn.SetFullName(_T("weather_routing_pi_rollover.svg"));
    _svg_weather_routing_rollover = fn.GetFullPath();
    fn.SetFullName(_T("weather_routing_pi_toggled.svg"));
    _svg_weather_routing_toggled = fn.GetFullPath();
#endif
```

rgleason, 2020-02-01:
BTW re the PLUGIN_COMMON_NAME error is not present in the Testplugin_pi because it compiles. I am going to have to look into that tomorrow morning.

bdbcat, 2020-02-01:
Rick...

syntax error: missing ')' before ';'
My mistake.  Correct is:
`wxLogMessage(_T("Loading toolbar icon: ") + _svg_weather_routing);
`

rgleason, 2020-02-01:
That fixed that error, now I have 

```
  icons.cpp
C:\Users\Rick\Documents\GitHub\weather_routing_pi\src\weather_routing_pi.cpp(187): error C2065: 'PLUGIN_COMMON_NAME': unde
clared identifier [C:\Users\Rick\Documents\GitHub\weather_routing_pi\build\weatherrouting_pi.vcxproj]
```

I'll have to look into this one and maybe Jon will have a solution, once I get it fixed, I can get back to you after running weather_routing.  OK?

rgleason, 2020-02-01:
Looking at https://github.com/jongough/testplugin_pi/commit/2ada410082300a1942f0be76c2abc8aed168f617

I have checked that I added this line in cmake/in-files/version.h.in

`#define PLUGIN_COMMON_NAME "${VERBOSE_NAME}"

and it is there.
`

rgleason, 2020-02-01:
Will have to leave this for morning. Sorry.

rgleason, 2020-02-01:
For some reason it compiles when pushed to my repos without the error above, although it still shows up when I build locally (have cleaned out build too).
https://github.com/rgleason/weather_routing_pi   Did a tag v1.13.3.1 push and the tarballs landed in cloudsmith here  https://cloudsmith.io/~rick-gleason/repos/opencpn-plugins-prod/packages/

Then downloaded the metadata files copied to my local opencpn/plugins  rg-alpha branch, built the new ocpn-plugins.xml with weather_routing and testplugin_pi and then made a new commit and pushed it to my repository   https://github.com/rgleason/plugins/tree/rg-alpha

I started Opencpn, current beta version, and weather_routing v1.13.3.1 shows but it would not install.
Yet the pathname in the xml works to browser to download.

Attached is part of the logfile
[logfile.txt](https://github.com/OpenCPN/OpenCPN/files/4142668/logfile.txt)

[image]
[image]
[image]

rgleason, 2020-02-01:
Correction!
Nope it is not found!

https://dl.cloudsmith.io/public/rick-gleason/opencpn-plugins-pkg/raw/names/weatherrouting-msvc-10.0.14393-tarball/versions/v1.13.3.1/weatherrouting_pi-1.13.3.1-ov50-1.16_msvc-10.0.14393.tar.gz

This is the path that is needed
https://dl.cloudsmith.io/public/rick-gleason/opencpn-plugins-prod/raw/names/weatherrouting-msvc-10.0.14393-tarball/versions/v1.13.3.1/weatherrouting_pi-1.13.3.1-ov50-1.16_msvc-10.0.14393.tar.gz

rgleason, 2020-02-01:
After putting it in a notepad++ compare the difference is

opencpn-plugins-prod
opencpn-plugins-pkg

I think it should say prod

rgleason, 2020-02-01:
Now it successfully installs, but the icons aren't right and there are two weather-routing icons and when I click on one it goes blank, but the menu comes up.

[image]

[logfile2.txt](https://github.com/OpenCPN/OpenCPN/files/4142698/logfile2.txt)

closed 2020-02-08 by rgleason