OpenCPN/OpenCPN #1698: Problems building OpenCPN Master

sbfreddie, 2020-02-03:
All:
I'm trying to build the master version of OpenCPN on my RockPro64 with 4GB of Main Memory.
I'm using Debian 9.11 Mate with Kernel 4.4.207(aarch64).
I had to to add to the CMakeLists.txt file to help Cmake find all the libs stashed away in various locations on this Distributions.  At line 259 I added:
set(_LIBDIRS "~/.local/lib:/bin:/usr/local/lib:/usr/lib/arm-linux-gnueabihf:/usr/lib:/lib/arm-linux-gnueabihf:/lib")

This got me thru the first hurdle.
After a successful configure with Cmake (Entire output file included below), it built all the way to the link section where it died tried to find  -lwx_gtk2u_webview-3.0.

However in the confirgure section (at the top of the output file) Cmake claims to have found all the  wx_gtk stuff:

-- Found wxWidgets: -L/usr/lib/arm-linux-gnueabihf;-pthread;;;-lwx_gtk2u_gl-3.0;-lwx_baseu_net-3.0;-lwx_baseu_xml-3.0;-lwx_gtk2u_html-3.0;-lwx_gtk2u_adv-3.0;-lwx_gtk2u_aui-3.0;-lwx_gtk2u_core-3.0;-lwx_baseu-3.0;-lwx_gtk2u_webview-3.0 (found version "3.0.2") 
-- Found wxWidgets webview add-on
-- Found wxWidgets...
--  wxWidgets Include: /usr/lib/arm-linux-gnueabihf/wx/include/gtk2-unicode-3.0;/usr/include/wx-3.0
--  wxWidgets Libraries: -L/usr/lib/arm-linux-gnueabihf;-pthread;;;-lwx_gtk2u_gl-3.0;-lwx_baseu_net-3.0;-lwx_baseu_xml-3.0;-lwx_gtk2u_html-3.0;-lwx_gtk2u_adv-3.0;-lwx_gtk2u_aui-3.0;-lwx_gtk2u_core-3.0;-lwx_baseu-3.0;-lwx_gtk2u_webview-3.0
--  Revised wxWidgets Libraries: -L/usr/lib/arm-linux-gnueabihf;-pthread;-lwx_gtk2u_gl-3.0;-lwx_baseu_net-3.0;-lwx_baseu_xml-3.0;-lwx_gtk2u_html-3.0;-lwx_gtk2u_adv-3.0;-lwx_gtk2u_aui-3.0;-lwx_gtk2u_core-3.0;-lwx_baseu-3.0;-lwx_gtk2u_webview-3.0

It appears that Cmake found these libs in /usr/lib/arm-linux-gnueabihf however did not set the variables in the CMakeCache.txt file:

//Flags used by the linker.
CMAKE_EXE_LINKER_FLAGS:STRING=

//Flags used by the linker during debug builds.
CMAKE_EXE_LINKER_FLAGS_DEBUG:STRING=

//Flags used by the linker during release minsize builds.
CMAKE_EXE_LINKER_FLAGS_MINSIZEREL:STRING=

//Flags used by the linker during release builds.
CMAKE_EXE_LINKER_FLAGS_RELEASE:STRING=

//Flags used by the linker during Release with Debug Info builds.
CMAKE_EXE_LINKER_FLAGS_RELWITHDEBINFO:STRING=

Anyone have an idea what path I should take to remedy this problem on this new ARMv8 system.
Thanks,
Freddie

P.S. attached are the pertinent files:
[CMakeCache.txt](https://github.com/OpenCPN/OpenCPN/files/4145844/CMakeCache.txt)
[CMakeLists.txt](https://github.com/OpenCPN/OpenCPN/files/4145845/CMakeLists.txt)
[Cmake Output.txt](https://github.com/OpenCPN/OpenCPN/files/4145847/Cmake.Output.txt)

leamas, 2020-02-03:
This is a known problem on some distros: cmake finds the headers  but not library for the webview component.

You should be able to walk around using -DOCPN_USE_WEBVIEW=OFF to cmake.

sbfreddie, 2020-02-04:
Leamas:
That took care of the problem.
What is Webview for?

Thanks,
Freddie

nohal, 2020-02-04:
`wxWebView` is a slightly better HTML viewer for the embedded help, if not available a very basic `wxHtmlWindow` is used instead. Nothing you could not live without.

sbfreddie, 2020-02-04:
Thanks, Pavel

closed 2020-02-04 by sbfreddie