OpenCPN/OpenCPN #1701: Win: VS 2017 Unix End of line

Hakansv, 2020-02-06:
Windows builds.
I've had trouble with my VS 2017 lately. It has put in line end CRLF instead of LF. Despite the somewhat hidden selection in tools.
After some search I found a method to add a config file where OpenCPN.sln resides, i.e. ~\build.
Would it be possible to include this in some make file to cmake for common use? An example attached. 
Håkan

[editorconfig.zip](https://github.com/OpenCPN/OpenCPN/files/4165664/editorconfig.zip)

transmitterdan, 2020-02-07:
@Hakansv 

This is fixed by git options.  We configure git to remove all the CR bytes from windows source files when pushing a commit.

Hakansv, 2020-02-07:
Dan..
Yes I know but for me that's a last rescue barrier. If the source for the changes done correct we can avoid seeing half a screen with warnings.

leamas, 2020-03-03:
See also #1416

rgleason, 2020-03-03:
When I and and commit, I get a message changing the eol to LF, isn't this the way this is changed normally? I believe it is a setting in Git https://opencpn.org/wiki/dokuwiki/doku.php?id=opencpn:developer_manual:developer_guide:compiling_windows#git

So why is this intermittant and isn't working correctly?

Hakansv, 2020-03-03:
No it's not a real problem since Git will prevent not Unix eol to pass through to master core. But for me it's annoying to see the Git messages about the change. There is no good way to set this in VS2017. For the last months I've used a editorconfig file in every repository I have and no correcting message from Git have been seen since.
So my advice is more for own and others comfort.

leamas, 2020-03-03:
If this was a VS-specific fix I would hesitate... but it's actually usable on all sorts of editors, as can be seen in #1416. So, why not?

Many things related to this is in a bad shape seen on the overall codebase, see #1118. But a editorconfig  would at least be helpful when it comes to new content.

In short. IMHO:  Make a PR so we can get this done.

rgleason, 2020-03-05:
File Found in vdr_pi  by SethDart

".gitattributes"
-------------
# Auto detect text files and perform LF normalization
* text=auto

# Custom for Visual Studio
*.cs     diff=csharp
*.sln    merge=union
*.csproj merge=union
*.vbproj merge=union
*.fsproj merge=union
*.dbproj merge=union

# Standard to msysgit
*.doc	 diff=astextplain
*.DOC	 diff=astextplain
*.docx diff=astextplain
*.DOCX diff=astextplain
*.dot  diff=astextplain
*.DOT  diff=astextplain
*.pdf  diff=astextplain
*.PDF	 diff=astextplain
*.rtf	 diff=astextplain
*.RTF	 diff=astextplain
---------------

Hakansv, 2020-03-29:
Mostly hesitations. And I don't know how to adapt CmakeList for this.

closed 2020-03-29 by Hakansv